XKNX/xknx #137: CouldNotParseTelegram percentage as binary value

itineric, 2018-08-30:
Hi,

I add some KNX device to XKNX and hass and I am having some odd error in the logs:
The group address 8/0/1 is the address to obtain the fan speed (its the speed state address). Its type is 5.001 (percentage).
I think xknx tries to parse it as a binary value.
Here is the error message:

`2018-08-30 09:40:40 ERROR (MainThread) [xknx.log] Error while processing telegram <CouldNotParseTelegram description="payload invalid" device_name="fan" group_address="8/0/1" payload="<DPTBinary value="0" />"/>`

Nota:
- the components works perfectly and the speed is correctly displayed in home assistant interface
- using ETS and the group monitor I am sure the data on the bus is 5.001 for address 8/0/1

I would like to get rid of this error before submitting the component.

itineric, 2018-08-30:
The configuration on home assistant is:
```
fan:
  - platform: knx
    name: fan
    speed_address: '8/0/31'
    speed_state_address: '8/0/1'
```

do you want the code of the device ?

moustic999, 2018-08-30:
Fan does not exist, so I suppose you create this device. 
I think it should work same as a dimming light.
probably the state is as the light state_address (binary) but it should be like brightness_state_address.

you can try to use knx light component with your fan, it should work, then adapt code of your device to match

Julius2342, 2018-08-30:
There is no fan knx component. 

"payload invalid"  is usually the case if e.g. xknx expects different type of payload (e.g. an array instead of a bit-field switch).

itineric, 2018-08-30:
Its a device I have developped.
I looked at cover component and how it manages the position state.

What I do not understand is the "<DPTBinary" that is provided in the error message. Where does it come from ? From the bus ? from the component ?

moustic999, 2018-08-30:
you should not use Cover as basis, but more light with dimming, variyng speed is much more like dimming light than moving a store.
check brightness_address, brightness_state_address

itineric, 2018-08-30:
Ok, looking into it, brightness is managed the same way as cover position.
I keep the bus monitor on in ETS to check if I can find some correlation.

Will submit the device soon.

itineric, 2018-08-30:
Does xknx (or home assistant) refresh the state of all devices once an hour ?
I have a massive read operation every hour on my bus (seen on group monitor in ETS).
It is during this operation than the error appears in the log.

itineric, 2018-08-30:
Ok, found it, but do not understand why...

I got:
```
        speed_processed = yield from self.speed.process(telegram)
        if speed_processed:
            yield from self.after_update()
```

Removed the "if" and the "after_update()", so now I just have 
`yield from self.speed.process(telegram)`
and it does not print an error.

Started the dev on a previous version of xknx/home-assistant, worked before without error, still works but with error.

Can close this, sorry...

closed 2019-03-16 by farmio