XKNX/xknx #142: Add DPT 9.020 (Voltage) a 2 byte float

magenbrot, 2018-09-11:
Hi,

My KNX power supply is able to collect some metrics like power, voltage and current. It seems that "voltage" isn't supported by xknx and HA yet, as it isn't mentioned in the home-assistant documentation. I've added it manually and it works. I'll send an PR later if anyone could confirm that voltage really isn't supported right now and I just did not see it ;)

best regards
magenbrot

Julius2342, 2018-09-11:
I think voltage is not yet included ... so a PR is highly appreciated!

(mhmmjm, Magenbrot :-) )

Julius2342, 2018-09-11:
I also have a (half-working) (refactoring)[https://github.com/XKNX/xknx/pull/127] of the DPT's to make things more flexible - which i started earlier this year, and somehow lost focus. Depending on how complex your change is, we might consider putting this inside this branch. And i could take the chance to start working on this again :-)

magenbrot, 2018-09-12:
the change is rather small, please review this PR #143

magenbrot, 2018-09-19:
new PR #146, replacing the other.

farmio, 2018-11-12:
It is merged now and 9.020 Voltage is in the HA documentation now I think this issue can be closed.

closed 2018-11-14 by Julius2342