XKNX/xknx #147: Error when reading group addresses with timers

stefanlaheij, 2018-09-17:
I have an extensive KNX installation with all kinds of actors, sensors, switches, etc.
My integration with Home Assistant works perfect, but I do get the following error in the logs when input is detected on an object linked to a timer:

`Error doing job: Exception in callback UDPTransport._on_read_ready
Traceback (most recent call last):
  File "uvloop/cbhandles.pyx", line 69, in uvloop.loop.Handle._run
  File "uvloop/handles/udp.pyx", line 64, in uvloop.loop.UDPTransport._on_read_ready
  File "/usr/local/lib/python3.6/site-packages/xknx/io/udp_client.py", line 55, in datagram_received
    self.data_received_callback(data)
  File "/usr/local/lib/python3.6/site-packages/xknx/io/udp_client.py", line 85, in data_received_callback
    knxipframe.from_knx(raw)
  File "/usr/local/lib/python3.6/site-packages/xknx/knxip/knxip.py", line 78, in from_knx
    pos += self.body.from_knx(data[pos:])
  File "/usr/local/lib/python3.6/site-packages/xknx/knxip/tunnelling_request.py", line 48, in from_knx
    pos += self.cemi.from_knx(raw[pos:])
  File "/usr/local/lib/python3.6/site-packages/xknx/knxip/cemi_frame.py", line 116, in from_knx
    return self.from_knx_data_link_layer(raw)
  File "/usr/local/lib/python3.6/site-packages/xknx/knxip/cemi_frame.py", line 147, in from_knx_data_link_layer
    self.cmd = APCICommand(tpci_apci & 0xFFC0)
  File "/usr/local/lib/python3.6/enum.py", line 291, in __call__
    return cls.__new__(cls, value)
  File "/usr/local/lib/python3.6/enum.py", line 533, in __new__
    return cls._missing_(value)
  File "/usr/local/lib/python3.6/enum.py", line 546, in _missing_
    raise ValueError("%r is not a valid %s" % (value, cls.__name__))
ValueError: 960 is not a valid APCICommand`

Two minutes after this error I then receive:
`Error: reading rading group address from KNX bus failed: ErrorCode.E_CONNECTION_ID`

Julius2342, 2018-09-18:
what do you mean with "linked to a timer" ?

Julius2342, 2018-09-18:
So it looks like the problem is here:

https://github.com/XKNX/xknx/blob/master/xknx/knxip/knxip_enum.py#L52

It looks like you receive cemi frames with "960" (== 0x03C0)  as APCICommand - which is not implemented. An Exception is raised and the tunnel connection gets out of sync. When trying the next read you get the E_CONNECTION_ID error.

I have to admit i never saw these type of packets in my system. Any idea what they are about?

stefanlaheij, 2018-09-18:
Hi Julis,

I am sorry. I mean a KNX timer object in ETS. This is an input I use that
accepts start/stop. (DPT 1.010 start/stop)
Use case: When my PIR motion sensor detects activity it sends a boolean to
the group address (only the rising edge, activation). A switch actuator
timer is activated or prolonged when it receives a 1 value again.
I also use this for my analog doorbell as input, with a switch actuator for
a couple of seconds to activate the chime.
Hopefully this helps.

Best regards, Stefan

stefanlaheij, 2018-09-18:
> So it looks like the problem is here:
>
> https://github.com/XKNX/xknx/blob/master/xknx/knxip/knxip_enum.py#L52
>
> It looks like you receive cemi frames with "960" (== 0x03C0) as APCICommand - which is [...]

Thanks for looking into this!
Yes, I already found that piece of code too  :)
When I convert the decimal 960 to hex I get 3C0. When I looked into that I found this code that included it: https://github.com/christoph2/knx/blob/master/knxReTk/busaccess/APCI.py
But.. I had no idea if I first had to convert it to hex :)

Julius2342, 2018-09-18:
Thanks god, KNX is such a well documented standard ...

stefanlaheij, 2018-09-18:
I found one other (more meaningful) reference to it via Google: 
`#define COUPLER_SPECIFIC_SERVICE 0x3C0`
Source: https://github.com/wireshark/wireshark/blob/master/epan/dissectors/packet-knxnetip.c

stefanlaheij, 2018-09-20:
> So it looks like the problem is here:
>
> https://github.com/XKNX/xknx/blob/master/xknx/knxip/knxip_enum.py#L52
>
> It looks like you receive cemi frames with "960" (== 0x03C0) as APCICommand - which is [...]

By monitoring the KNX bus I see a broadcasted telegram (GA 0/0/0) of type NetworkParameterResponse. After that I get the issues.
Maybe this helps?

stefanlaheij, 2018-09-26:
Hi @Julius2342 ,

I see this user has exactly the same issue with another product. 
https://knx-user-forum.de/forum/projektforen/edomi/1123307-networkparameterresponse-stopt-bus-anbindung-von-edomi-telegramm-of-death

Maybe this helps?

Julius2342, 2018-09-26:
May I ask you to catch the TCP packets with TCPdump? Then we could implement NetworkParameterResponses - and directly write a unit test for it :)

stefanlaheij, 2018-09-26:
Hi @Julius2342,

I think the packet is in here:
[capture.zip](https://github.com/XKNX/xknx/files/2421646/capture.zip)

stefanlaheij, 2018-10-16:
Hi @Julius2342 ,
Did you find anything?

Julius2342, 2018-10-16:
Not yet. I was pretty busy with my dayjob recent weeks so i did not have time to look into it.

farmio, 2019-05-27:
I have found this documented in the KNX specifications at 3.2.6 . 
I wonder if we could just ignore all broadcast telegrams (CEMI payload with destination GA 0/0/0) to avoid these errors? Do we actually need any of these for the current functions of xknx?

farmio, 2020-03-21:
I think I have found the root of the CEMI errors that cause xknx to restart (eg. "CEMI too small" or "APCI not supported"). These unsupported telegrams are `TunnellingRequest` telegrams. Since we raise `CouldNotParseKNXIP` in cemi_frame.py our udp_client.py catches this and never calls `self.handle_knxipframe(knxipframe)`. See
https://github.com/XKNX/xknx/blob/9790cdb696285b5700ec06c571193a7feeb96a39/xknx/io/udp_client.py#L90
Therefore the registered callback is never called - and in this case this would be sending the `TunnellingAck` the KNX-Interface is waiting for. When it doesn't receive the Ack it sends a DisconnectRequest (that we seem to ignore) and quits the connection - our Connection_ID is lost and the next heartbeat will fail and we attempt a reconnect.

This is reproduceable by doing a Line Scan from ETS like suggested in https://github.com/home-assistant/core/issues/32394 .

PR #271 (2020-03-25): Ignore unknown CEMI Messages
If we don't understand the CEMI Message ignore it and move on (eg. send TunnellingAck).
Previously we would raise CuldNotParseKNXIP and loose the connection because no Ack was sent.

> From 03_06_03 EMI_IMI v01.03.03 AS:
> 4.1.3.3 Exception handling: unknown messages
> General behaviour after reception of an unknown or unsupported message (this is, the cEMI Server does not know the received cEMI message code): the received message shall be ignored, this is, the cEMI Server shall generate no confirmation message.

closes #147 
fixes home-assistant/core#32394
fixes # [... cut]
closed 2020-03-25 by farmio