XKNX/xknx #148: Startup Warnings  with Climate component on HA

resoai, 2018-09-21:
It look  like every time HA starts, xknx raises a warning for each climate entity as follows:

```
Could not read value of <Climate name="Master Bathroom" temperature="GroupAddress("4/6/10")/None/<DPTArray value="[0xd,0x8]" />/25.76"  target_temperature="GroupAddress("4/5/10")/None/<DPTArray value="[0xc,0x1a]" />/21.0"  setpoint_shift="GroupAddress("3/3/10")/GroupAddress("4/3/10")/<DPTArray value="[0x0]" />/0" setpoint_shift_step="0.5" setpoint_shift_max="6" setpoint_shift_min="-6" group_address_operation_mode="GroupAddress("3/1/10")" group_address_operation_mode_state="None" group_address_controller_status="None" group_address_controller_status_state="GroupAddress("4/0/10")" /> 3/1/10
```

It looks like we need to skip checking operation_mode_state if controller_status_state is defined.

resoai, 2018-09-23:
Sadly I can’t check it since I’m running Hassio

resoai, 2018-09-30:
Would you be able to create a PR bumping the version?

Julius2342, 2018-09-30:
I bumped it last week, and I got some weird warnings. i want to investigae these before moving on :-)

rsterrenburg, 2018-10-12:
@Julius2342 any plans on bumping the version or still getting the weird warnings?

Julius2342, 2018-10-23:
I tried it, and already found some bugs.

At the moment the syncing does not work. Will have a look at it asap. But it is progressing (slowly).

resoai, 2018-12-05:
I'm afraid I'm still getting warning messages.

Dudu67, 2018-12-11:
Same here :
```
Log Details (WARNING)
Tue Dec 11 2018 08:22:16 GMT+0100 (heure normale d’Europe centrale)

Could not read value of <Climate name="Sensor.clim_ch3.climate" temperature="None/GroupAddress("3/1/7")/<DPTArray value="[0x7,0xbc]" />/19.8"  target_temperature="GroupAddress("3/2/7")/None/None/None"  setpoint_shift="None/None/None/None" setpoint_shift_step="0.5"
setpoint_shift_max="6" setpoint_shift_min="-6"
group_address_operation_mode="GroupAddress("3/4/10")" group_address_operation_mode_state="GroupAddress("3/4/10")" group_address_controller_status="None" group_address_controller_status_state="None" group_address_controller_mode="None" group_address_controller_mode_state="None" group_address_on_off="None/None/None/None" /> 3/4/10
```

And this append cyclically every hour.
By looking at this error I see that there is a "group_adress_on_off" paramater that is not documented in the climate component (and by the way not implement in the TileBoard @resoai).
For the moment i map a switch on this group to be able to switch on/off the HVAC unit, but it will be better to be able to call the climate.turn_on or climate.turn_off from home assistant.

Olivier

farmio, 2019-01-13:
Hm 🤔. Seems related to #120

resoai, 2019-03-20:
So, with HA 0.90 I'm afraid I'm still getting warning messages, although I've now started to receive separate ones for "mode":

```
Could not read value of <Climate name="AV room" temperature="None/GroupAddress("4/6/2")/<DPTArray value="[0xc,0x4e]" />/22.04"  target_temperature="None/GroupAddress("4/5/2")/<DPTArray value="[0xc,0x1a]" />/21.0"  setpoint_shift="GroupAddress("3/3/2")/GroupAddress("4/3/2")/<DPTArray value="[0x0]" />/0" setpoint_shift_step="0.5" setpoint_shift_max="6" setpoint_shift_min="-6" group_address_on_off="None/None/None/None" /> 3/1/2

```

```
Could not read value of <ClimateMode name="AV room Mode" group_address_operation_mode="GroupAddress("3/1/2")" group_address_operation_mode_state="None" group_address_controller_status="None" group_address_controller_status_state="GroupAddress("4/0/2")" group_address_controller_mode="None" group_address_controller_mode_state="None" /> 3/1/2
```

The config I'm using for climate is as follows:

```
- platform: knx
  name: AV room
  temperature_address: 4/6/2
  target_temperature_state_address: 4/5/2
  setpoint_shift_address: 3/3/2
  setpoint_shift_state_address: 4/3/2
  operation_mode_address: 3/1/2
  controller_status_state_address: 4/0/2
```

Please note that I have to use `controller_status_state_address` because I have Jung room controllers and they handle operation modes differently. It was only thanks to @Julius2342 that we have discovered it last year. Do you guys know what could be the source of those warnings? I'm afraid setting `rate_limit` did not help.

moustic999, 2019-05-10:
You are missing the : operation_mode_state_address ??

resoai, 2019-05-10:
> You are missing the : operation_mode_state_address ??

I have to use `controller_status_state_address` instead of `operation_mode_state_address` because I have Jung room controller which reports status in different format (AN 097/07).

Julius2342, 2019-05-12:
@resoai : yes, on your devices reading and writing is done via different "types" of addresses. For whatever reason.

resoai, 2019-05-12:
Could this be the reason of the warnings?

Julius2342, 2019-05-12:
@resoai : may you log telegrams and past which exact read-telegram xknx is sending out directly before you see this message?

farmio, 2019-05-27:
There are even more issues with controller_status and controller_mode actuators. See https://github.com/home-assistant/home-assistant/issues/21737

PR #190 (2019-06-10): Climate setpoint shift fix
- Climate().target_temperature is now a property. Its implementation abstracts the usage of a writeable _target_temperature RemoteValueTemp() or a read only one with setpoint_shift.
- ~_base_temperature RemoteValueTemp() is introduced to reflect an immutable base temperature used together with setpoint_shift. It can be read from the bus if the actor supports it.~ (removed - no  known actor supports this)
- Climate().base_temperature property reflects the default temperature (setpoint-shift=0) for the active climate mode.
 
This PR should 
- resolve #183
- resolve #14 [... cut]
commit 56fffe22 (2019-06-10): read state addresses only for climat modes

this should resolve #148
closed 2019-06-10 by marvin-w