XKNX/xknx #150: Restore state of switch after a HA restart

jncanches, 2018-09-27:
Hi,

I tried moving to 0.9.0 recent release which seems to have a lot of good addition.
But this point in the changelog makes me ask a question : 

> Breaking Change: Only read explicit state addresses #140 

Is it because of this that now, each time I restart Home assistant, my switches lost their state and are "off" ? (They are configured as usual, with only an address property, no state_address).
Is there a way with 0.9 release to keep switch state restore ?

Thank you for your work !

JN

mag2bue, 2018-10-05:
To read to status, use  „state_address“  in your config.

moustic999, 2018-11-21:
I think there is a misunderstanding how KNX works.
A switch has no State! a light has a state ( on off .... ) 
The main issue of XKNX is regarding the use of the state address group.
@Julius2342, can we have a discussion regarding this ? I really think the way state is used must be changed.

TinusK, 2018-11-21:
Well, the fakt is. A switch has a state. Using this state the switch 'knows' what to send if switched. In a 'simple' knx setup a switch switches a single light/socket on or off. Usually a knx switch has a internal state of a external state object.

If the 'internal' state is used then every time the switch is used the state is updated: if switched on, then the state of the switch is on. The next time the switch is used a 'off' is send (and viseversa). This all is independent of the state of the light/socket that the switch switches.

Now, the following situation: you have a physical switch somewhere in your building and you have a second one (physical of software: hass). If the physical switch uses the internal state then it will send a on of off message to the light/socker actuator depending of the internal state. Lets say the light is of. If switched on using the physical switch then the state of the switch is on and the state of that light is on. If I now switch off that light using HASS then the physical switch still 'thinks' the light is on. If you now press the physical switch, nothing happens because the switch will send a off message (internal state of that switch is still on).

To solve this, every knx switch I came across sofar has the option to enable an external state object. This object can be put in the same groupaddress that switches the light/socket on or off. This way the physical switch state is set the same as the state of the light/socket.

An other GA should be created with the state object of the actuator that controls the light/socket. This GA is set in HASS so that the switch in HASS also knows what the state is of that light/socket.

If you have two or more lights controlled by one switch then you should use a logic function that defines the state of a switch to be on or off.

The way xknx uses the state is, in my opnion, correct.

Hope this helps,
Tinus

Julius2342, 2018-11-21:
I think the right approach would be to configure the state address. (and set the state address to the same address as the normal address.)

If this does not work it is an easy to fix bug.

moustic999, 2018-11-21:
@TinusK "Well, the fakt is. A switch has a state. Using this state the switch 'knows' what to send if switched. In a 'simple' knx setup a switch switches a single light/socket on or off. Usually a knx switch has a internal state or a external state object."
You are right, a switch has a state, but you cannot use the state communication object to query it !
the state communication object is used to tell to the switch what should be it's new internal status !
this is sent by the actuator to the switch

farmio, 2019-05-27:
> I can imagine the purpose of a binary_sensor if receiving a message on a group address and triggering another group address, but the same, I can't understand.

You are right, this is a flaw in the documentation. It will be fixed with the next release.

For future reference to the main topic of this Issue: Just use the same `state_address` as the `address` if your installation doesn't provide explicit ones.

closed 2019-05-27 by farmio