XKNX/xknx #170: Busmonitor: Example from homepage not working - NoneType can't be used

svenihoney, 2019-01-30:
Hello,
I am trying to run the "Callbacks" example from the homepage https://xknx.io/xknx. But I get the following exception and the event loop is stopped:
```
sven@beely:~$ virtualenv  xknx -p python3
...
(xknx) sven@beely:~$ cd xknx/
(xknx) sven@beely:~/xknx$ pip install xknx
...
Successfully installed netifaces-0.10.9 pyyaml-3.13 xknx-0.9.4
(xknx) sven@beely:~/xknx$ python callbacks.py 
Press Ctrl+C to stop
Telegram received: <Telegram group_address="GroupAddress("0/0/3")", payload="<DPTBinary value="0" />" telegramtype="TelegramType.GROUP_WRITE" direction="TelegramDirection.INCOMING" />
Task exception was never retrieved
future: <Task finished coro=<TelegramQueue.run() done, defined at /home/sven/xknx/lib/python3.6/site-packages/xknx/core/telegram_queue.py:58> exception=TypeError("object NoneType can't be used in 'await' expression",)>
Traceback (most recent call last):
  File "/home/sven/xknx/lib/python3.6/site-packages/xknx/core/telegram_queue.py", line 67, in run
    await self.process_telegram(telegram)
  File "/home/sven/xknx/lib/python3.6/site-packages/xknx/core/telegram_queue.py", line 95, in process_telegram
    await self.process_telegram_incoming(telegram)
  File "/home/sven/xknx/lib/python3.6/site-packages/xknx/core/telegram_queue.py", line 113, in process_telegram_incoming
    ret = await telegram_received_cb.callback(telegram)
TypeError: object NoneType can't be used in 'await' expression
```
What am I doing wrong here? I would expect to have all telegrams printed without an exception occurring.

svenihoney, 2019-01-30:
The `async` keyword did the trick, thanks! Perhaps update the homepage with this.

Julius2342, 2019-01-30:
done :)

closed 2019-01-30 by Julius2342

svenihoney, 2019-01-30:
I assume the same applies for the callback `device_updated_cb`, perhaps you should update this, too.