XKNX/xknx #181: how to get the telegram sender

marcor4, 2019-02-20:
I am currently working on a code for telegram monitoring. I started from the available example [https://github.com/XKNX/xknx/blob/master/examples/example_telegram_monitor.py](url)

The telegrams returned by the code are including information such as the group address, telegram type, direction, payload. Is there a way of getting also the physical address of the sender?

Julius2342, 2019-02-21:
Not with the telegram callabacks. You have to listen to the raw data stream and wait for CEMIFrame. In there it should be within the attribute [`self.src_addr`](https://github.com/XKNX/xknx/blob/master/xknx/knxip/cemi_frame.py#L128).

To get these frames, you have to directly register a callback to the `UDPClient`  (either a member of the Tunnel or the Routing object, which themselves are members of the `KNXIPInterface()` which is a member of the `XKNX()` object.

But all this is at your own risk :-)

marcor4, 2019-02-21:
This is helpful, I'll try it. Thank you.

What about including this information within the telegram object in future releases?

marcor4, 2019-02-21:
Super easy: I simple added a line in cemi_frame.py [`def telegram(self)`](https://github.com/XKNX/xknx/blob/master/xknx/knxip/cemi_frame.py#39):
```python
    def telegram(self):
        """Return telegram."""
        telegram = Telegram()
        telegram.payload = self.payload
        telegram.group_address = self.dst_addr
        telegram.source_address = self.src_addr
        …
```

Thank you. It would have been quite complicated without your references.

Julius2342, 2019-02-22:
uh dirty :) But if it works!

Im not a big fan of adding this to the `Telegram` object, bc the main intention behind this object is to keep it as lean as possible.

I will close the issue as it seems to work now for you!

closed 2019-02-22 by Julius2342