XKNX/xknx #185: Binary sensor question

resoai, 2019-02-23:
My KNX installer has left me with a physical button winch sends a 1 byte value which is supposed to open to gate. I'm trying to intercept it with binary sensor so I can send an HTTP request to the my Mobotix doorentry. Unfortunately I have no access to ETS and if I understand correctly, binary sensor would not really work OK with such values?

https://imgur.com/sTdFySp

So far I've managed to only find a very dirty workaround (which I don't like):
```
  fire_event: true
  fire_event_filter: ["7/7/1"]
````

Any chance I could still make it work with a binary sensor?

farmio, 2019-02-23:
`fire_event_filter` is normally used when you want to get an event for a KNX telegram, even if the value of the group address hadn't changed since the last telegram. This could be the case if your gate gets triggered to open and closes automatically after some time, but doesn't report its state to the KNX bus. Depending on your installation `fire_event_filter` could be perfectly fine.

You can monitor your bus even with the demo version of ETS (you will not have named group addresses and DPTs since you will probably not be able to open your project) or even with xknx (set logging level to debug so you can see CEMI Frames directly). You could then find out more about the sent values.

Alternatively ask your installer for the project file (always good to have) - there are tools that can extract DPTs from that - or you can ask someone with an ets license to look it up for you.

resoai, 2019-03-09:
Thank you. I know that the value on the group address changes to "100" but I presume I would not be able to simply use binary_sensor to catch it?

farmio, 2019-03-09:
Binary_sensor can only handle binary (0/1) values. You could use a `sensor:` and map it inside HA with an automation.

closed 2019-03-16 by farmio