XKNX/xknx #187: Sensor with value_type 'pulse' does not read state on startup

elratt0r, 2019-03-01:
When adding a sensor to HomeAssistant with type 'pulse', the state is not read from the KNX bus.

Type 'pulse' is handled special in [xknx/devices/sensor.py](https://github.com/XKNX/xknx/blob/master/xknx/devices/sensor.py), has no group_address_state.

Is there any reason for this? I can't spot one.

farmio, 2019-03-01:
Have you set `group_address_state:` ?
Please provide the relevant parts of your configuration file. 
My second guess would be a missing "Read" flag in your devices Knx communication object parameters.

elratt0r, 2019-03-01:
sensors.yaml
```
- platform: knx
  name: 'fan_eg_raw'
  address: '5/1/8'
  type: 'pulse'
```

As said, there is no GroupValueRead Telegram coming from HA/XKNX. That's when I started to investigate. Setting the type to percent and the Read is there (but can't use the data, because I need to access some bits of the raw value in HA).

I could quick and dirty work around the issue by the following change:
```
--- xknx/devices/remote_value_dpt_value_1_ucount.py  2019-03-01 05:47:07.164401609 +0000
+++ xknx/devices/remote_value_dpt_value_1_ucount.py 2019-03-01 00:46:04.819341413 +0000
@@ -19,7 +19,7 @@
         """Initialize remote value of KNX DPT 5.010."""
         # pylint: disable=too-many-arguments
         super(RemoteValueDptValue1Ucount, self).__init__(
-            xknx, group_address, None,
+            xknx, group_address, group_address,
             device_name=device_name, after_update_cb=after_update_cb)

     def payload_valid(self, payload):
```

Nearly all other sensors initialize the master class (RemoteValue) with group_address_state=group_address (and group_address=None, default value).

That was why I was asking why pulse is handled that special way in sensor.py

farmio, 2019-03-01:
Now this is some information we can work with, thank you! 

Sensors are treated as read only devices, therefore they have only one GA. 
We changed the way devices are initialized back in 0.9.0 - maybe the pulse sensor slipped through. I'll look into it. 🧐

farmio, 2019-03-03:
This should be resolved as soon as the next xknx release ( > 0.10.0) is integrated in HA. Be aware that this could take several weeks. Alternatively you can always use [xknx custom component](https://xknx.io/home_assistant).
You will be able to use 'pulse', 'DPT-5' or '1byte_unsigned' in your configuration then.

closed 2019-03-03 by farmio