XKNX/xknx #192: Send telegrams while monitoring

marcor4, 2019-03-11:
I am working on a code in which the main engine is based on [example_telegram_monitor.py](https://github.com/XKNX/xknx/blob/master/examples/example_telegram_monitor.py). During the monitoring of the bus, I would like to send other telegrams and I modified the code as follows:

```python
async def telegrams_monitor():
        global message_received
        xknx = XKNX()
        xknx.telegram_queue.register_telegram_received_cb(telegram_received_cb)

        connection_config=ConnectionConfig(
                connection_type=ConnectionType.TUNNELING,
                gateway_ip="192.168.1.110",
                gateway_port=3671,
                local_ip="192.168.1.103")

        await xknx.start(daemon_mode=False,connection_config=connection_config)
        remote_value=RemoteValueSwitch(xknx,group_address=GroupAddress("10/1/1"))
        while True:
                await remote_value.on()
                await asyncio.sleep(10)
                await remote_value.off()
                await asyncio.sleep(10)
```

This is working great, except the fact that it does not seem the right way to do it. In particular, I would also like to send telegrams with arrays, but when I use the more generic `RemoteValue` (capable of processing both `DPTBinary` and `DPTArray`) I got the log `payload_valid not implemented for RemoteValue`.

How can I send telegrams within the while loop reported above?

farmio, 2019-03-17:
You should use one of the subclasses of RemoteValue() like `RemoteValue1Count()`. Pick a fitting one for the DPT of the corresponding group address.
The usage of a generic RemoteValue would result in sending different telegram sizes and encodings to the same group address. Your knx actors would not be happy about it.

marcor4, 2019-03-18:
I am currently trying with `xknx.telegrams.put(telegram_to_send)` and it works (and I realized that `RemoteValue*` is using the same function). For some reasons my TCP/IP interface was crashing when sending arrays (but not booleans). I had a blackout last week and things seems to work now,

> The usage of a generic RemoteValue would result in sending different telegram sizes and encodings to the same group address. Your knx actors would not be happy about it.

Of course I will do my best to match the telegram size with the one required by the group addresses. I know it will be mutually beneficial for our happiness :)

closed 2019-05-27 by farmio