XKNX/xknx #193: Errors in home-assistant.log at start

jncanches, 2019-03-21:
Hi,

First of all, thanks for this great library.

I have a very specific case that makes home assistant to log an error at start.

I compute using HA, an average temperature which I then expose to KNX (for the use of an HVAC unit, but it is not the point here). BUT, on startup (before its default value is affected), this computed temperature have a state of "unknown". The expose to KNX process first try to convert this temperature and it fails (and log into HA.log file).

`2019-03-21 01:21:25 ERROR (MainThread) [homeassistant.core] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/lib/python3.5/asyncio/tasks.py", line 239, in _step
    result = coro.send(None)
  File "/srv/homeassistant/lib/python3.5/site-packages/homeassistant/components/knx/__init__.py", line 338, in _async_entity_changed
    await self.device.set(float(new_state.state))
ValueError: could not convert string to float: 'unknown'`

Generally, I edit the "homeassistant/components/knx/__init__.py" file and insert this at line 338 as an additional check :

`    if new_state.state == "unknown":
         return
` (sorry return don't want to go to next line in github editor)

I wonder if it's possible to add this check in the standard code ?

Thanks a lot !

farmio, 2019-03-21:
Thank you for the tip! 👍
Yes it is possible. I'm currently reworking some bits of the expose sensor code and stumbled upon the same issue.

jncanches, 2019-03-21:
Fantastic :-) thanks !

closed 2019-04-18 by marvin-w

jncanches, 2019-07-18:
Hi,

this issue is not fixed in 0.11.1 :(
I still have those errors in my home assistant log : 

> 2019-07-18 21:36:46 ERROR (MainThread) [homeassistant.core] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/srv/homeassistant/lib/python3.7/site-packages/homeassistant/components/knx/__init__.py", line 336, in _async_entity_changed
    await self.device.set(float(new_state.state))
ValueError: could not convert string to float: 'unknown'
2019-07-18 21:36:46 ERROR (MainThread) [homeassistant.core] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/srv/homeassistant/lib/python3.7/site-packages/homeassistant/components/knx/__init__.py", line 336, in _async_entity_changed
    await self.device.set(float(new_state.state))
ValueError: could not convert string to float: 'unknown'
2019-07-18 21:36:46 ERROR (MainThread) [homeassistant.core] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/srv/homeassistant/lib/python3.7/site-packages/homeassistant/components/knx/__init__.py", line 336, in _async_entity_changed
    await self.device.set(float(new_state.state))
ValueError: could not convert string to float: 'unknown'
2019-07-18 21:36:46 ERROR (MainThread) [homeassistant.core] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/srv/homeassistant/lib/python3.7/site-packages/homeassistant/components/knx/__init__.py", line 336, in _async_entity_changed
    await self.device.set(float(new_state.state))
ValueError: could not convert string to float: 'unknown'
2019-07-18 21:36:46 ERROR (MainThread) [homeassistant.core] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/srv/homeassistant/lib/python3.7/site-packages/homeassistant/components/knx/__init__.py", line 336, in _async_entity_changed
    await self.device.set(float(new_state.state))
ValueError: could not convert string to float: 'unknown'
2019-07-18 21:36:46 ERROR (MainThread) [homeassistant.core] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/srv/homeassistant/lib/python3.7/site-packages/homeassistant/components/knx/__init__.py", line 336, in _async_entity_changed
    await self.device.set(float(new_state.state))
ValueError: could not convert string to float: 'unknown'

farmio, 2019-07-18:
Hi! 0.11.1 is not yet merged to HA. Have you installed it manually or are you using the dev or beta build?

jncanches, 2019-07-18:
@farmio Sorry for the confusion. I mean this issue is still alive in the current version of HA.
In fact, that was not fixed in #189

farmio, 2019-07-19:
I think it was, it's just that #189 is part of 0.11 and this is not yet in HA. It will come with HA 0.97 so you can wait another 2 weeks or switch to the dev branch now to test it.

jncanches, 2019-07-19:
Fantastic. Thank you. Of course I will wait :-)