XKNX/xknx #202: Cant parse DPT2ByteFloat -320.0

spali, 2019-05-11:
**Description of problem:**

```
2019-05-11 19:03:44 ERROR (MainThread) [xknx.log] Error while syncing device: <ConversionError description="Cant parse DPT2ByteFloat" value="-320.0"/>
```
In my case it's and edge case with low prio.
Because it's a new weather station, that I didn't fully setup yet.
When the station has some error like no additional power source etc. it reports -320 °C as temperature. But maybe it leads to a bug in parsing floats in general?
The strange thing is, the error displays the value correct... so why the error... is it out of a specification for temperature?

- [ ] using xknx standalone
- [X] using Home-Assistant knx plugin

**Version information:**
- xknx / Home-Assistant release with the issue:
0.92.2
- last working xknx / Home-Assistant release (if known):
unknown

**KNX installation:**

**Product:**
Griesser EMX-8
**DPT:**
9.001 Temperatur (°C)
**Monitor export:** (first telegram is the read request from homeassistant, second is the answer)
```xml
<CommunicationLog xmlns="http://knx.org/xml/telegrams/01">
  <Telegram Timestamp="2019-05-11T17:03:44.7197551Z" Service="L_Data.ind" FrameFormat="CommonEmi" RawData="2900BCE0110A2C02030040A030" />
  <Telegram Timestamp="2019-05-11T17:03:44.6892171Z" Service="L_Data.ind" FrameFormat="CommonEmi" RawData="2900BCE010FA2C02010000" />
</CommunicationLog>
```

**Problem-relevant `xknx.yaml` or `configuration.yaml` entries (fill out even if it seems unimportant):**
```
knx:
  tunneling:
    host: !secret knx_gateway_ip
    port: 3671
    local_ip: !secret hass_instance_ip
sensor:
- platform: knx
  name: aussen_temperatur
  type: temperature
  address: 5/4/2
```

**Traceback (if applicable):**
not available

farmio, 2019-05-11:
Hello!
You are exactly right. -320°C is out of the KNX specification of DPT 9.001 which would be: `-273 °C ... 670 760 °C` (and its also out of specification of laws of physics 🤓)
If you can't disable this behaviour in the device application, you could bypass this error by using a general 2-byte float datatype
```
sensor:
- platform: knx
  name: aussen_temperatur
  type: 'DPT-9'
  address: '5/4/2'
  device_class: 'temperature'
```
I haven't tested setting `device_class` in the config, but it should work.

spali, 2019-05-11:
> Hello!
> You are exactly right. -320°C is out of the KNX specification of DPT 9.001 which would be: `-273 °C ... 670 760 °C` (and its also out of specification of laws of physics 🤓)

didn't even thought about that... you are absolutely right :see_no_evil:

> If you can't disable this behaviour in the device application, you could bypass this error by using a general 2-byte float datatype
> 
> ```
> sensor:
> - platform: knx
>   name: aussen_temperatur
>   type: 'DPT-9'
>   address: '5/4/2'
>   device_class: 'temperature'
> ```
> 
> I haven't tested setting `device_class` in the config, but it should work. This is for the little temperature sign in the visualization and may be needed for some components like HomeKit.

`device_class` is not allowed there, but with customization it's possible to set the `device_class: 'temperature'`

For anyone else that has this 1:1000000000 problem, the following results in a working (**physical not possible**) solution that looks exactly the same in home assistant 🦄 

```yaml
sensor:
- platform: knx
  name: aussen_temperatur
  type: 'DPT-9'
  address: 5/4/2

homeassistant:
  customize:
    sensor.aussen_temperatur:
      friendly_name: 'Aussen Temperatur'
      device_class: 'temperature'
      unit_of_measurement: '°C'
```

closed 2019-05-11 by spali