XKNX/xknx #215: Can't connect to Knx gateway behind NAT on VM

guntherper, 2019-06-29:
I am trying to connect to a knx ip gateway from a homeassistant vm but get a connection error. The vm is behind a NAT on the host.

KNX Gateway IP: 192.168.2.191
Host IP: 192.168.2.249
VM IP: 192.168.1.11

My homeassistant configuration
 <pre>Tunneling:
    Host: '192.168.2.191'
    Port: 3671
    Local_ip: '192.168.1.11'</pre>

Iptables configuration on the host
<pre> echo 1 > /proc/sys/net/ipv4/ip_forward
iptables -t nat -A PREROUTING -s '192.168.2.191' -j DNAT --to-destination '192.168.1.11' 
iptables -t nat -A POSTROUTING -d '192.168.2.191' -j SNAT --to-source '192.168.2.249' </pre>

On tcpdump I see an UDP packet going to Knx gateway but there is no reply packet.

farmio, 2019-06-29:
NAT and proxies aren't supported by xknx. I had the same issue when trying to use docker without `--network host`.
If you insist to use NAT you could try this fork (its still 0.94 or something): https://github.com/farmio/xknx/tree/docker-support . Have a look at the last 3 commits. It will not be configurable via home-assistant so you'd have to adapt the [custom component](https://github.com/farmio/xknx/blob/docker-support/home-assistant-plugin/custom_components/xknx/__init__.py).

guntherper, 2019-06-30:
Yes this is indeed what I am looking for :)
I followed the instrucions on de xknx homeassistant to make the symlink to the custom_components but it always installs the default  xknx 0.9.4 lib as a dep. 
And I can't seem to figure out how to export the pythonpath en start de hass command from within hassio.
Any advice?

farmio, 2019-06-30:
Oh I did not know you were running Hassio in a VM (it is specialized for RaspberryPi). As far as I know Hassio has no support for custom components. But I may be wrong / outdated. 
That said there is now a manifest.json where the version has to be commented out. I'm not sure if the old `Requirement = ...` in the __init__.py file is still supported. But if it is still there remove it. 
In HA configuration you have to change all `knx:` and `platform: Knx` to `xknx` and `platform: xknx`

You could also hack the changes into the hassio site-packages and component just to see if it worked. But then it would be overwritten by an update.

guntherper, 2019-06-30:
Thnx, I have put the changes in the hassio site-packages and now it works :)
Any chance this will end up in the official XKNX package?

farmio, 2019-07-01:
Glad to hear that it works.
I'd like to have it merged to xknx too; maybe in the next release. I'll have to rebase it and integrate to the custom component but dayjob keeps me busy right now...

farmio, 2019-07-04:
@guntherper I have updated the branch to xknx 0.11 - could you test it? If it works I'll file a pull request to the official xknx repo. Thanks!
https://github.com/farmio/xknx/tree/docker-support

Julius2342, 2019-07-05:
@farmio : Do you think it would make sense to integrate this into main xknx?

guntherper, 2019-07-05:
@farmio yes normally I have some time tomorrow to test this 😊

farmio, 2019-07-05:
@Julius2342 I think it would be nice for xknx, but not necessarily for HA (except for guntherper ;) , but thats not a big step to take then.
Its not a big, complicated change and brings possibilities like dockered bus monitoring systems and maybe even remote bus diagnostics.

guntherper, 2019-07-06:
Tested and it works :)

retoo, 2020-05-10:
Any news on this? It would be awesome if you could merge this :D

farmio, 2020-05-10:
Recently I tried to change the config name from "proxy_ip" to something different (I think "knx_client_ip") but fell into a black hole of refactoring because I didn't want the IP to be stored in the udp_client. 
Long story short: It's on my list but time is limited.

spacegaier, 2020-06-07:
I would also be quite interested in that since then I could use my Ubuntu VM that I use for HA development also for the KNX part rather than having to revert back to directly dev/testing on my RPI.

PR #544 (2021-02-17): Route back

## Description

Quote from KNX specification:

> For special communication scenarios it is not suitable to put information about IP Host Protocol addresses into the payload of KNXnet/IP Frames.
If e.g. the Frame shall travel across IP routers performing network address translation (NAT), the IP Host Protocol address information that the KNXnet/IP Server needs to address to send KNXnet/IP Frames back to the KNXnet/IP Client may not be known to the KNXnet/IP Client.
For these scenarios a special encoding of the HPAI with the IP address and IP port number all set to zero shall be used [... cut]
closed 2021-02-17 by farmio