XKNX/xknx #218: Climate does no longer work in HA 0.96

kmplngj, 2019-07-18:
**Climate Component in HA does not work.**

- [ ] using xknx standalone
- [x] using Home-Assistant knx plugin

**Version information:**
- xknx / Home-Assistant release with the issue: HA 0.96
- last working xknx / Home-Assistant release (if known): HA 0.95.4

**KNX installation:**
ETS5
Weinzierl Linemaster 762
MDT AKH-0800.02

**Problem-relevant `xknx.yaml` or `configuration.yaml` entries (fill out even if it seems unimportant):**

```
  - platform: knx
    name: Essen
    temperature_address: '3/1/2'
    setpoint_shift_address: '3/3/14'
    target_temperature_state_address: '3/2/2'
    operation_mode_address: '3/4/2'
```

**Traceback (if applicable):**

```
homeassistant          | 2019-07-18 11:19:41 ERROR (MainThread) [homeassistant.core] Error doing job: Task exception was never retrieved
homeassistant          | Traceback (most recent call last):
homeassistant          |   File "/usr/local/lib/python3.7/site-packages/xknx/core/telegram_queue.py", line 67, in run
homeassistant          |     await self.process_telegram(telegram)
homeassistant          |   File "/usr/local/lib/python3.7/site-packages/xknx/core/telegram_queue.py", line 95, in process_telegram
homeassistant          |     await self.process_telegram_incoming(telegram)
homeassistant          |   File "/usr/local/lib/python3.7/site-packages/xknx/core/telegram_queue.py", line 120, in process_telegram_incoming
homeassistant          |     await device.process(telegram)
homeassistant          |   File "/usr/local/lib/python3.7/site-packages/xknx/devices/device.py", line 73, in process
homeassistant          |     await self.process_group_write(telegram)
homeassistant          |   File "/usr/local/lib/python3.7/site-packages/xknx/devices/climate.py", line 221, in process_group_write
homeassistant          |     await self.temperature.process(telegram)
homeassistant          |   File "/usr/local/lib/python3.7/site-packages/xknx/devices/remote_value.py", line 79, in process
homeassistant          |     await self.after_update_cb()
homeassistant          |   File "/usr/local/lib/python3.7/site-packages/xknx/devices/device.py", line 33, in after_update
homeassistant          |     await device_updated_cb(self)
homeassistant          |   File "/usr/src/app/homeassistant/components/knx/climate.py", line 191, in after_update_callback
homeassistant          |     await self.async_update_ha_state()
homeassistant          |   File "/usr/src/app/homeassistant/helpers/entity.py", line 226, in async_update_ha_state
homeassistant          |     self._async_write_ha_state()
homeassistant          |   File "/usr/src/app/homeassistant/helpers/entity.py", line 256, in _async_write_ha_state
homeassistant          |     attr.update(self.state_attributes or {})
homeassistant          |   File "/usr/src/app/homeassistant/components/climate/__init__.py", line 181, in state_attributes
homeassistant          |     if self.target_temperature_step:
homeassistant          |   File "/usr/src/app/homeassistant/components/knx/climate.py", line 222, in target_temperature_step
homeassistant          |     return self.device.temperature_step
homeassistant          | AttributeError: 'Climate' object has no attribute 'temperature_step'
homeassistant          | 2019-07-18 11:19:45 WARNING (MainThread) [xknx.log] Could not read value of <ClimateMode name="Essen Mode" group_address_operation_mode="GroupAddress("3/4/2")" group_address_operation_mode_state="None" group_address_controller_status="None" group_address_controller_status_state="None" group_address_controller_mode="None" group_address_controller_mode_state="None" /> 3/4/2
homeassistant          | 2019-07-18 11:19:46 WARNING (MainThread) [homeassistant.components.switch] Setup of platform broadlink is taking over 10 seconds.
homeassistant          | 2019-07-18 11:19:46 WARNING (MainThread) [xknx.log] Could not read value of <Climate name="Essen" temperature="None/GroupAddress("3/1/2")/None/None"  target_temperature="None/GroupAddress("3/2/2")/None/None"  setpoint_shift="GroupAddress("3/3/14")/GroupAddress("3/3/14")/None/None" setpoint_shift_step="0.5" setpoint_shift_max="6" setpoint_shift_min="-6" group_address_on_off="None/None/None/None" /> 3/1/2
homeassistant          | 2019-07-18 11:19:47 WARNING (MainThread) [xknx.log] Could not read value of <Climate name="Essen" temperature="None/GroupAddress("3/1/2")/None/None"  target_temperature="None/GroupAddress("3/2/2")/None/None"  setpoint_shift="GroupAddress("3/3/14")/GroupAddress("3/3/14")/None/None" setpoint_shift_step="0.5" setpoint_shift_max="6" setpoint_shift_min="-6" group_address_on_off="None/None/None/None" /> 3/2/2
homeassistant          | 2019-07-18 11:19:48 WARNING (MainThread) [xknx.log] Could not read value of <Climate name="Essen" temperature="None/GroupAddress("3/1/2")/None/None"  target_temperature="None/GroupAddress("3/2/2")/None/None"  setpoint_shift="GroupAddress("3/3/14")/GroupAddress("3/3/14")/None/None" setpoint_shift_step="0.5" setpoint_shift_max="6" setpoint_shift_min="-6" group_address_on_off="None/None/None/None" /> 3/3/14
homeassistant          | 2019-07-18 11:19:49 WARNING (MainThread) [xknx.log] Could not read value of <Climate name="Essen" temperature="None/GroupAddress("3/1/2")/None/None"  target_temperature="None/GroupAddress("3/2/2")/None/None"  setpoint_shift="GroupAddress("3/3/14")/GroupAddress("3/3/14")/None/None" setpoint_shift_step="0.5" setpoint_shift_max="6" setpoint_shift_min="-6" group_address_on_off="None/None/None/None" /> 3/4/2
```

marvin-w, 2019-07-18:
@farmio Was your PR with the 0.10 release merged into 0.96?

I think it contained a rename of the `temperature_step_size` to `temperature_step` commit.

farmio, 2019-07-18:
0.10 is in HA since 0.90 - see https://github.com/home-assistant/home-assistant/pull/21541
since then only the new climate stuff from you and @monte-monte was added.

0.11.1 is now in dev, so it will launch with HA 0.97 I think.

farmio, 2019-07-31:
resolved with HA 0.96.3

closed 2019-07-31 by farmio