XKNX/xknx #220: Error when reading group addresses with temperature data

auredor, 2019-08-02:
I have a KNX installation with all kinds of actors, sensors, switches, etc.
My integration with Home Assistant works perfect, but I do get the following error in the logs when datat is detected on an temperature object. I had similar problems with lux values. I get read of the erro messages with good DPT specifications. But in this case, tI use the good DPT (9.001) but still get the error.

Thanks.

- [ ] using xknx standalone
- [x ] using Home-Assistant knx plugin

**Version information:**
- xknx / Home-Assistant release with the issue: 96.5
- last working xknx / Home-Assistant release (if known): never

**KNX installation:**

Different sensors from differents manufacturers : Thermostat (Merten), PIR sensor (BEG) but all with a temperature sensor object : 
- DPT setup  : 9.001
- Flags : Communication & Transmission

**Problem-relevant `xknx.yaml` or `configuration.yaml` entries (fill out even if it seems unimportant):**

- platform: knx
    name: Température Parents
    address: '0/1/12'
    type: 'temperature'
  - platform: knx
    name: Température SDB
    address: '0/1/13'
    type: 'temperature'
  - platform: knx
    name: Température Bureau
    address: '0/1/21'
    type: 'temperature'
  ....

**Traceback (if applicable):**

Could not read value of <Light name="Chambre Parent - Lit Gauche" switch="GroupAddress("0/0/12")/GroupAddress("0/0/33")/<DPTBinary value="0" />/Value.OFF" brightness="GroupAddress("0/0/28")/GroupAddress("0/0/29")/<DPTArray value="[0x0]" />/0" /> 0/0/33
17:54 __main__.py (WARNING)
Could not read value of <Sensor name="Température SDD" sensor="None/GroupAddress("0/1/37")/<DPTArray value="[0xc,0xfb]" />/25.5" value="25.5" unit="°C"/> 0/1/37
17:54 __main__.py (WARNING) - message first occured at 14:53 and shows up 2 times
Could not read value of <Sensor name="Température Chambre Victor" sensor="None/GroupAddress("0/1/36")/<DPTArray value="[0xc,0xd7]" />/24.78" value="24.78" unit="°C"/> 0/1/36
17:54 __main__.py (WARNING)
Could not read value of <Sensor name="Température Chambre Charlotte" sensor="None/GroupAddress("0/1/35")/<DPTArray value="[0xc,0xbf]" />/24.3" value="24.3" unit="°C"/> 0/1/35
17:54 __main__.py (WARNING)
Could not read value of <Sensor name="Température Salle de Jeux" sensor="None/GroupAddress("0/1/34")/<DPTArray value="[0xc,0x8c]" />/23.28" value="23.28" unit="°C"/> 0/1/34
17:54 __main__.py (WARNING) - message first occured at 14:53 and shows up 2 times
Could not read value of <Sensor name="Température Bureau" sensor="None/GroupAddress("0/1/21")/<DPTArray value="[0xc,0xfd]" />/25.54" value="25.54" unit="°C"/> 0/1/21
17:54 __main__.py (WARNING)
Could not read value of <Sensor name="Température SDB" sensor="None/GroupAddress("0/1/13")/<DPTArray value="[0xc,0xd8]" />/24.8" value="24.8" unit="°C"/> 0/1/13
17:54 __main__.py (WARNING)
Could not read value of <Sensor name="Température Parents" sensor="None/GroupAddress("0/1/12")/<DPTArray value="[0xc,0xd7]" />/24.78" value="24.78" unit="°C"/> 0/1/12
17:54 __main__.py (WARNING)
Could not read value of <Sensor name="Température Terrasse" sensor="None/GroupAddress("0/1/38")/<DPTArray value="[0xc,0x1f]" />/21.1" value="21.1" unit="°C"/> 0/1/38
16:54 __main__.py (WARNING)
Could not read value of <Sensor name="Température Chambre Charlotte" sensor="None/GroupAddress("0/1/35")/<DPTArray value="[0xc,0xbe]" />/24.28" value="24.28" unit="°C"/> 0/1/35
16:54 __main__.py (WARNING)
Could not read value of <Sensor name="Température Salle de Jeux" sensor="None/GroupAddress("0/1/34")/<DPTArray value="[0xc,0x98]" />/23.52" value="23.52" unit="°C"/> 0/1/34
16:54 __main__.py (WARNING)
Could not read value of <Sensor name="Température Bureau" sensor="None/GroupAddress("0/1/21")/<DPTArray value="[0xd,0x1]" />/25.62" value="25.62" unit="°C"/> 0/1/21
16:54 __main__.py (WARNING) - message first occured at 15:53 and shows up 2 times
Could not read value of <Sensor name="Température Parents" sensor="None/GroupAddress("0/1/12")/<DPTArray value="[0xc,0xdd]" />/24.9" value="24.9" unit="°C"/> 0/1/12
16:54 __main__.py (WARNING)
Error doing job: SSL error in data received
16:37 /srv/homeassistant/lib/python3.7/site-packages/homeassistant/core.py (ERROR) - message first occured at 15:55 and shows up 2 times
Error doing job: SSL handshake failed
16:37 /srv/homeassistant/lib/python3.7/site-packages/homeassistant/core.py (ERROR) - message first occured at 15:55 and shows up 2 times
Could not read value of <Sensor name="Température Chambre Victor" sensor="None/GroupAddress("0/1/36")/<DPTArray value="[0xc,0xd5]" />/24.74" value="24.74" unit="°C"/> 0/1/36
15:53 __main__.py (WARNING)
Could not read value of <Sensor name="Température Chambre Charlotte" sensor="None/GroupAddress("0/1/35")/<DPTArray value="[0xc,0xc1]" />/24.34" value="24.34" unit="°C"/> 0/1/35
15:53 __main__.py (WARNING)
Could not read value of <Sensor name="Température Salle de Jeux" sensor="None/GroupAddress("0/1/34")/<DPTArray value="[0xc,0x9a]" />/23.56" value="23.56" unit="°C"/> 0/1/34
15:53 __main__.py (WARNING)
Could not read value of <Sensor name="Température SDB" sensor="None/GroupAddress("0/1/13")/<DPTArray value="[0xc,0xe0]" />/24.96" value="24.96" unit="°C"/> 0/1/13
15:53 __main__.py (WARNING)
Could not read value of <Sensor name="Température Parents" sensor="None/GroupAddress("0/1/12")/<DPTArray value="[0xc,0xdf]" />/24.94" value="24.94" unit="°C"/> 0/1/12
15:53 __main__.py (WARNING)
Could not read value of <Sensor name="Température Terrasse" sensor="None/GroupAddress("0/1/38")/<DPTArray value="[0x7,0x76]" />/19.1" value="19.1" unit="°C"/> 0/1/38
14:53 __main__.py (WARNING)
Could not read value of <Sensor name="Température Chambre Victor" sensor="None/GroupAddress("0/1/36")/<DPTArray value="[0xc,0xd6]" />/24.76" value="24.76" unit="°C"/> 0/1/36
14:53 __main__.py (WARNING)
Could not read value of <Sensor name="Température Chambre Charlotte" sensor="None/GroupAddress("0/1/35")/<DPTArray value="[0xc,0xc4]" />/24.4" value="24.4" unit="°C"/> 0/1/35
14:53 __main__.py (WARNING)
Could not read value of <Sensor name="Température Bureau" sensor="None/GroupAddress("0/1/21")/<DPTArray value="[0xd,0x2]" />/25.64" value="25.64" unit="°C"/> 0/1/21
14:53 __main__.py (WARNING)
Could not read value of <Sensor name="Température SDB" sensor="None/GroupAddress("0/1/13")/<DPTArray value="[0xc,0xe1]" />/24.98" value="24.98" unit="°C"/> 0/1/13
14:53 __main__.py (WARNING)

farmio, 2019-08-02:
Xknx sends a GroupValueRead request to temperature group addresses, but yours don't have the `read` flag set. This is done once an hour so the error is repeated hourly for each group address it can't read.
I'd suggest to set a read flag on those communication objects (as always only one per GA).

Alternatively - starting from home-assistant 0.97, which should launch next week - you can disable the recurring reading of a `sensor: ` by adding `sync_state: False` to its configuration.

closed 2019-08-02 by farmio

auredor, 2019-08-02:
Thanks.

I will give a try with the Read Flag.

Thanks