XKNX/xknx #222: HA KNX actions without intervention

fernasneri, 2019-08-09:
I have a symptom issue of a light coming on, without any intervention or automation, the strange thing is that when this light turns on the others on turn off.
The light is in group address '1/0/3' and in the log I don't see this event related to this group address.
If you turn off the home assistant server the symptom does not happen, I assume it is some information exchanged with the misinterpreted bus.
I have a KNX interface IP from Hagger, any idea?

Log of the period in which it issue occurred:
2019-08-09 11:57:33 WARNING (MainThread) [xknx.log] Error: KNX bus did not respond in time to GroupValueRead request for: 5/0/1
2019-08-09 11:57:33 WARNING (MainThread) [xknx.log] Could not sync group address '5/0/1' from <BinarySensor group_address_state="GroupAddress("5/0/1")" name="Sensor Hall IO" state="BinarySensorState.OFF"/>
2019-08-09 11:57:34 WARNING (MainThread) [xknx.log] Error: KNX bus did not respond in time to GroupValueRead request for: 5/0/2
2019-08-09 11:57:34 WARNING (MainThread) [xknx.log] Could not sync group address '5/0/2' from <BinarySensor group_address_state="GroupAddress("5/0/2")" name="Sensor Corredor IO" state="BinarySensorState.OFF"/>
2019-08-09 11:57:35 ERROR (MainThread) [xknx.log] Error while processing telegram <CouldNotParseTelegram description="invalid payload" device_name="Cozinha e Sala Mode" payload="<DPTBinary value="0" />"/>
2019-08-09 11:57:35 WARNING (MainThread) [xknx.log] Error: KNX bus did not respond in time to GroupValueRead request for: 4/1/4
2019-08-09 11:57:35 WARNING (MainThread) [xknx.log] Could not sync group address '4/1/4' from <Climate name="Suite" temperature="None/GroupAddress("4/1/4")/None/None" target_temperature="None/GroupAddress("4/3/4")/None/None" setpoint_shift="None/None/None/None" setpoint_shift_step="0.5" setpoint_shift_max="6" setpoint_shift_min="-6" group_address_on_off="None/None/None/None" />
2019-08-09 11:57:36 WARNING (MainThread) [xknx.log] Error: KNX bus did not respond in time to GroupValueRead request for: 4/3/4
2019-08-09 11:57:36 WARNING (MainThread) [xknx.log] Could not sync group address '4/3/4' from <Climate name="Suite" temperature="None/GroupAddress("4/1/4")/None/None" target_temperature="None/GroupAddress("4/3/4")/None/None" setpoint_shift="None/None/None/None" setpoint_shift_step="0.5" setpoint_shift_max="6" setpoint_shift_min="-6" group_address_on_off="None/None/None/None" />
2019-08-09 11:57:37 WARNING (MainThread) [xknx.log] Error: KNX bus did not respond in time to GroupValueRead request for: 2/2/6
2019-08-09 11:57:37 WARNING (MainThread) [xknx.log] Could not sync group address '2/2/6' from <Cover name="Suite Varanda" updown="GroupAddress("2/0/6")/None/None/None" step="GroupAddress("2/1/6")/None/None/None" position="GroupAddress("2/3/6")/GroupAddress("2/2/6")/None/None" angle="None/None/None/None" travel_time_down="26" travel_time_up="29" />
2019-08-09 11:57:38 ERROR (MainThread) [xknx.log] Error while processing telegram <CouldNotParseTelegram description="invalid payload" device_name="Cozinha e Sala Mode" payload="<DPTBinary value="0" />"/>
2019-08-09 11:57:38 WARNING (MainThread) [xknx.log] Error: KNX bus did not respond in time to GroupValueRead request for: 2/2/10
2019-08-09 11:57:38 WARNING (MainThread) [xknx.log] Could not sync group address '2/2/10' from <Cover name="WC Serviço" updown="GroupAddress("2/0/10")/None/None/None" step="GroupAddress("2/1/10")/None/None/None" position="GroupAddress("2/3/10")/GroupAddress("2/2/10")/None/None" angle="None/None/None/None" travel_time_down="26" travel_time_up="29" />
2019-08-09 11:57:39 WARNING (MainThread) [xknx.log] Error: KNX bus did not respond in time to GroupValueRead request for: 2/2/1
2019-08-09 11:57:39 WARNING (MainThread) [xknx.log] Could not sync group address '2/2/1' from <Cover name="Despensa" updown="GroupAddress("2/0/1")/None/None/None" step="GroupAddress("2/1/1")/None/None/None" position="GroupAddress("2/3/1")/GroupAddress("2/2/1")/None/None" angle="None/None/None/None" travel_time_down="26" travel_time_up="29" />
2019-08-09 11:57:40 WARNING (MainThread) [xknx.log] Error: KNX bus did not respond in time to GroupValueRead request for: 2/2/7
2019-08-09 11:57:40 WARNING (MainThread) [xknx.log] Could not sync group address '2/2/7' from <Cover name="Suite Jardim" updown="GroupAddress("2/0/7")/None/None/None" step="GroupAddress("2/1/7")/None/None/None" position="GroupAddress("2/3/7")/GroupAddress("2/2/7")/None/None" angle="None/None/None/None" travel_time_down="26" travel_time_up="29" />
2019-08-09 11:57:41 WARNING (MainThread) [xknx.log] Error: KNX bus did not respond in time to GroupValueRead request for: 2/2/4
2019-08-09 11:57:41 WARNING (MainThread) [xknx.log] Could not sync group address '2/2/4' from <Cover name="Quarto Rodrigo" updown="GroupAddress("2/0/4")/None/None/None" step="GroupAddress("2/1/4")/None/None/None" position="GroupAddress("2/3/4")/GroupAddress("2/2/4")/None/None" angle="None/None/None/None" travel_time_down="26" travel_time_up="29" />
2019-08-09 11:57:42 WARNING (MainThread) [xknx.log] Error: KNX bus did not respond in time to GroupValueRead request for: 2/2/8
2019-08-09 11:57:42 WARNING (MainThread) [xknx.log] Could not sync group address '2/2/8' from <Cover name="Biblioteca" updown="GroupAddress("2/0/8")/None/None/None" step="GroupAddress("2/1/8")/None/None/None" position="GroupAddress("2/3/8")/GroupAddress("2/2/8")/<DPTArray value="[0xe6]" />/90" angle="None/None/None/None" travel_time_down="26" travel_time_up="29" />
2019-08-09 11:57:43 WARNING (MainThread) [xknx.log] Error: KNX bus did not respond in time to GroupValueRead request for: 2/2/3
2019-08-09 11:57:43 WARNING (MainThread) [xknx.log] Could not sync group address '2/2/3' from <Cover name="Sala" updown="GroupAddress("2/0/3")/None/None/None" step="GroupAddress("2/1/3")/None/None/None" position="GroupAddress("2/3/3")/GroupAddress("2/2/3")/None/None" angle="None/None/None/None" travel_time_down="26" travel_time_up="29" />
2019-08-09 11:57:44 WARNING (MainThread) [xknx.log] Error: KNX bus did not respond in time to GroupValueRead request for: 2/2/5
2019-08-09 11:57:44 WARNING (MainThread) [xknx.log] Could not sync group address '2/2/5' from <Cover name="Quarto Maria" updown="GroupAddress("2/0/5")/None/None/None" step="GroupAddress("2/1/5")/None/None/None" position="GroupAddress("2/3/5")/GroupAddress("2/2/5")/None/None" angle="None/None/None/None" travel_time_down="26" travel_time_up="29" />
2019-08-09 11:57:45 WARNING (MainThread) [xknx.log] Error: KNX bus did not respond in time to GroupValueRead request for: 2/2/9
2019-08-09 11:57:45 WARNING (MainThread) [xknx.log] Could not sync group address '2/2/9' from <Cover name="WC Interior" updown="GroupAddress("2/0/9")/None/None/None" step="GroupAddress("2/1/9")/None/None/None" position="GroupAddress("2/3/9")/GroupAddress("2/2/9")/None/None" angle="None/None/None/None" travel_time_down="26" travel_time_up="29" />
2019-08-09 11:57:46 WARNING (MainThread) [xknx.log] Error: KNX bus did not respond in time to GroupValueRead request for: 2/2/2
2019-08-09 11:57:46 WARNING (MainThread) [xknx.log] Could not sync group address '2/2/2' from <Cover name="Cozinha" updown="GroupAddress("2/0/2")/None/None/None" step="GroupAddress("2/1/2")/None/None/None" position="GroupAddress("2/3/2")/GroupAddress("2/2/2")/<DPTArray value="[0xef]" />/94" angle="None/None/None/None" travel_time_down="26" travel_time_up="29" />
2019-08-09 11:57:48 WARNING (MainThread) [xknx.log] Error: KNX bus did not respond in time to GroupValueRead request for: 1/1/9
2019-08-09 11:57:48 WARNING (MainThread) [xknx.log] Could not sync group address '1/1/9' from <Light name="Sanca Recuperador" switch="GroupAddress("1/0/9")/GroupAddress("1/1/9")/<DPTBinary value="0" />/False" />
2019-08-09 11:57:49 WARNING (MainThread) [xknx.log] Error: KNX bus did not respond in time to GroupValueRead request for: 1/1/10
2019-08-09 11:57:49 WARNING (MainThread) [xknx.log] Could not sync group address '1/1/10' from <Light name="Quarto Rodrigo" switch="GroupAddress("1/0/10")/GroupAddress("1/1/10")/<DPTBinary value="0" />/False" />
2019-08-09 11:57:50 WARNING (MainThread) [xknx.log] Error: KNX bus did not respond in time to GroupValueRead request for: 1/1/11
2019-08-09 11:57:50 WARNING (MainThread) [xknx.log] Could not sync group address '1/1/11' from <Light name="Sanca Quarto Rodrigo" switch="GroupAddress("1/0/11")/GroupAddress("1/1/11")/<DPTBinary value="0" />/False" />
[... cut ...]
    local_ip: '192.168.1.75'
    
light:
#ILUMINAÇÃO_COZINHA
  - platform: knx
    name: Despensa
    address: '1/0/1'
    state_address: '1/1/1'
    
  - platform: knx
    name: Iha Cozinha
    address: '1/0/2'
    state_address: '1/1/2'
    
  - platform: knx
    name: Cozinha
    address: '1/0/3'
    state_address: '1/1/3'

  - platform: knx
    name: Exaustor Cozinha
    address: '1/0/4'
    state_address: '1/1/4'

  - platform: knx
    name: Sanca Cozinha
    address: '1/0/5'
    state_address: '1/1/5'

DirkMu, 2019-08-09:
Hi, 

I am facing the same issue since I upgraded to 0.97.0 (xknx 0.11.1).
I see  switches turning off. Sometimes also lights turn on. 

Two things I saw. 
The following warning is thrown when states are read form the bus, which seem to happen hourly.
`Could not sync group address '7/1/65' from <BinarySensor group_address_state="GroupAddress("7/1/65")" name="flur_fb1_sensor_motion" state="BinarySensorState.OFF"/>
19:07 __main__.py (WARNING) - message first occured at 8:01 and shows up 12 times
Error: KNX bus did not respond in time to GroupValueRead request for: 7/1/65
19:07 __main__.py (WARNING) - message first occured at 8:01 and shows up 12 times`

It seems to be related to the L flag in ETS because in my case the binary sensor does not have the read flag.  

In addition a switch is turned off occasionally at the same time:
1/2/132 is the state address of the switch whereas 1/2/131 is the address.
1.1.18 the actor.
1.1.250 is the KNX IP Interface of Jung (IPS200REG)

`2019-08-09 19:07:41 DEBUG (MainThread) [xknx.log] Sync gaebad_light_aktor
2019-08-09 19:07:41 DEBUG (MainThread) [xknx.telegram] <Telegram group_address="GroupAddress("1/2/132")", payload="None" telegramtype="TelegramType.GROUP_READ" direction="TelegramDirection.OUTGOING" />
2019-08-09 19:07:41 DEBUG (MainThread) [xknx.knx] Sending: <KNXIPFrame <KNXIPHeader HeaderLength="6" ProtocolVersion="16" KNXIPServiceType="KNXIPServiceType.TUNNELLING_REQUEST" Reserve="0" TotalLength="21" />
 body="<TunnellingRequest communication_channel_id="81" sequence_counter="180" cemi="<CEMIFrame SourceAddress="PhysicalAddress("1.1.250")" DestinationAddress="GroupAddress("1/2/132")" Flags="1011110011100000" Command="APCICommand.GROUP_READ" payload="None" />" />" />
2019-08-09 19:07:41 DEBUG (MainThread) [xknx.knx] Received: <KNXIPFrame <KNXIPHeader HeaderLength="6" ProtocolVersion="16" KNXIPServiceType="KNXIPServiceType.TUNNELLING_ACK" Reserve="0" TotalLength="10" />
 body="<TunnellingAck communication_channel_id="81" sequence_counter="180" status_code="ErrorCode.E_NO_ERROR" />" />
2019-08-09 19:07:41 DEBUG (MainThread) [xknx.log] Success: received correct answer from KNX bus: ErrorCode.E_NO_ERROR
2019-08-09 19:07:41 DEBUG (MainThread) [xknx.knx] Received: <KNXIPFrame <KNXIPHeader HeaderLength="6" ProtocolVersion="16" KNXIPServiceType="KNXIPServiceType.TUNNELLING_REQUEST" Reserve="0" TotalLength="21" />
 body="<TunnellingRequest communication_channel_id="81" sequence_counter="96" cemi="<CEMIFrame SourceAddress="PhysicalAddress("1.1.250")" DestinationAddress="GroupAddress("1/2/132")" Flags="1011110011100000" Command="APCICommand.GROUP_READ" payload="<DPTBinary value="0" />" />" />" />
2019-08-09 19:07:41 DEBUG (MainThread) [xknx.knx] Sending: <KNXIPFrame <KNXIPHeader HeaderLength="6" ProtocolVersion="16" KNXIPServiceType="KNXIPServiceType.TUNNELLING_ACK" Reserve="0" TotalLength="10" />
 body="<TunnellingAck communication_channel_id="81" sequence_counter="96" status_code="ErrorCode.E_NO_ERROR" />" />
2019-08-09 19:07:41 DEBUG (MainThread) [xknx.telegram] <Telegram group_address="GroupAddress("1/2/132")", payload="<DPTBinary value="0" />" telegramtype="TelegramType.GROUP_READ" direction="TelegramDirection.INCOMING" />
2019-08-09 19:07:41 DEBUG (MainThread) [xknx.knx] Received: <KNXIPFrame <KNXIPHeader HeaderLength="6" ProtocolVersion="16" KNXIPServiceType="KNXIPServiceType.TUNNELLING_REQUEST" Reserve="0" TotalLength="22" />
 body="<TunnellingRequest communication_channel_id="81" sequence_counter="97" cemi="<CEMIFrame SourceAddress="PhysicalAddress("1.1.89")" DestinationAddress="GroupAddress("4/1/6")" Flags="1011110011100000" Command="APCICommand.GROUP_WRITE" payload="<DPTArray value="[0x0]" />" />" />" />
2019-08-09 19:07:41 DEBUG (MainThread) [xknx.knx] Sending: <KNXIPFrame <KNXIPHeader HeaderLength="6" ProtocolVersion="16" KNXIPServiceType="KNXIPServiceType.TUNNELLING_ACK" Reserve="0" TotalLength="10" />
 body="<TunnellingAck communication_channel_id="81" sequence_counter="97" status_code="ErrorCode.E_NO_ERROR" />" />
2019-08-09 19:07:41 DEBUG (MainThread) [xknx.telegram] <Telegram group_address="GroupAddress("4/1/6")", payload="<DPTArray value="[0x0]" />" telegramtype="TelegramType.GROUP_WRITE" direction="TelegramDirection.INCOMING" />
2019-08-09 19:07:41 DEBUG (MainThread) [xknx.knx] Received: <KNXIPFrame <KNXIPHeader HeaderLength="6" ProtocolVersion="16" KNXIPServiceType="KNXIPServiceType.TUNNELLING_REQUEST" Reserve="0" TotalLength="21" />
 body="<TunnellingRequest communication_channel_id="81" sequence_counter="98" cemi="<CEMIFrame SourceAddress="PhysicalAddress("1.1.18")" DestinationAddress="GroupAddress("1/2/132")" Flags="1011110011010000" Command="APCICommand.GROUP_RESPONSE" payload="<DPTBinary value="0" />" />" />" />
2019-08-09 19:07:41 DEBUG (MainThread) [xknx.knx] Sending: <KNXIPFrame <KNXIPHeader HeaderLength="6" ProtocolVersion="16" KNXIPServiceType="KNXIPServiceType.TUNNELLING_ACK" Reserve="0" TotalLength="10" />
 body="<TunnellingAck communication_channel_id="81" sequence_counter="98" status_code="ErrorCode.E_NO_ERROR" />" />
2019-08-09 19:07:41 DEBUG (MainThread) [xknx.telegram] <Telegram group_address="GroupAddress("1/2/132")", payload="<DPTBinary value="0" />" telegramtype="TelegramType.GROUP_RESPONSE" direction="TelegramDirection.INCOMING" />
`

Currently, I am not able to trace down the reason for that. Everything worked fine with HA 0.96.5

farmio, 2019-08-09:
@DirkMu If your binary sensor doesn't support Read-Flags (or you can't change it) you can configure it with `sync_state: False` to disable hourly reading. 
@fernasneri Your "Cozinha e Sala" mode doesn't seem to be supported, but it should not be related to your problem. 

For the other issues I have no clue. I think one would have to have detailed knowledge about your installation and an ETS log to find the source of your random switching. 
The only explanation I have would be that some devices react to the answer of the hourly group readings of your configured binary sensors (other sensors were read hourly before 0.97)

DirkMu, 2019-08-09:
@farmio Thank you. I already solved the binary sensor issue (Could not sync group address / KNX bus did not respond in time to GroupValueRead request for) by either adding flag L (read) in ETS or by adding `sync_state: false` to the configuration if L is not set in ETS. 

The issue with the switch seemed to be due to an incorrect setting in ETS --> solved

I still don't know why lights and sometimes climates cannot be read in time --> still monitoring this

farmio, 2019-08-09:
@DirkMu do you have multiple Knx Lines or areas? If so, have you added a dummy device for HA and assigned GAs in ETS to get correct filter tables for line couplers?

DirkMu, 2019-08-10:
Yes, I have multiple lines but rather due to the number of devices and the required power supply. There is no filter (yet). And yes it worked before. 

I got one warning this night which was related to a light that did not respond in time. It is a dimming actor. The GAs are readable.

fernasneri, 2019-08-12:
@farmio Thank you. I already solved the issue adding in "configuration.yaml" the line "state_updater: false".
Since then I have not had symptoms of lights turn on and off without action.

farmio, 2019-08-12:
That's good to hear. 
I still guess there's something wrong with your binary sensors. Maybe you are integrating a binary scene or something that triggers on  reading. You could try to find it by adding `sync_state: False` to some and see if the problem persists (with state_updater enabled then).

marvin-w, 2019-08-12:
I also had to add `state_updater: False` but thats due to the fact that my devices are old as hell :-D.

closed 2020-01-21 by farmio