XKNX/xknx #226: ValueError: 448 is not a valid APCICommand

elupus, 2019-08-19:
**Description of problem:**
Exception during processing of messages, likely while ETS is communicating with new Secure IP Interface.

- [ ] using xknx standalone
- [X] using Home-Assistant knx plugin

**Version information:**
- xknx / Home-Assistant release with the issue: 0.98
- last working xknx / Home-Assistant release (if known): None

**KNX installation:**

ETS 5.7.2

**Problem-relevant `xknx.yaml` or `configuration.yaml` entries (fill out even if it seems unimportant):**

**Traceback (if applicable):**
```
Aug 19 19:09:24 hass hass[32967]: Traceback (most recent call last):
Aug 19 19:09:24 hass hass[32967]:   File "/usr/lib/python3.6/asyncio/events.py", line 127, in _run
Aug 19 19:09:24 hass hass[32967]:     self._callback(*self._args)
Aug 19 19:09:24 hass hass[32967]:   File "/usr/lib/python3.6/asyncio/selector_events.py", line 1078, in _read_ready
Aug 19 19:09:24 hass hass[32967]:     self._protocol.datagram_received(data, addr)
Aug 19 19:09:24 hass hass[32967]:   File "/home/homeassistant/env/lib/python3.6/site-packages/xknx/io/udp_client.py", line 55, in datagram_received
Aug 19 19:09:24 hass hass[32967]:     self.data_received_callback(data)
Aug 19 19:09:24 hass hass[32967]:   File "/home/homeassistant/env/lib/python3.6/site-packages/xknx/io/udp_client.py", line 87, in data_received_callback
Aug 19 19:09:24 hass hass[32967]:     knxipframe.from_knx(raw)
Aug 19 19:09:24 hass hass[32967]:   File "/home/homeassistant/env/lib/python3.6/site-packages/xknx/knxip/knxip.py", line 78, in from_knx
Aug 19 19:09:24 hass hass[32967]:     pos += self.body.from_knx(data[pos:])
Aug 19 19:09:24 hass hass[32967]:   File "/home/homeassistant/env/lib/python3.6/site-packages/xknx/knxip/tunnelling_request.py", line 48, in from_knx
Aug 19 19:09:24 hass hass[32967]:     pos += self.cemi.from_knx(raw[pos:])
Aug 19 19:09:24 hass hass[32967]:   File "/home/homeassistant/env/lib/python3.6/site-packages/xknx/knxip/cemi_frame.py", line 113, in from_knx
Aug 19 19:09:24 hass hass[32967]:     return self.from_knx_data_link_layer(raw)
Aug 19 19:09:24 hass hass[32967]:   File "/home/homeassistant/env/lib/python3.6/site-packages/xknx/knxip/cemi_frame.py", line 143, in from_knx_data_link_layer
Aug 19 19:09:24 hass hass[32967]:     self.cmd = APCICommand(tpci_apci & 0xFFC0)
Aug 19 19:09:24 hass hass[32967]:   File "/usr/lib/python3.6/enum.py", line 291, in __call__
Aug 19 19:09:24 hass hass[32967]:     return cls.__new__(cls, value)
Aug 19 19:09:24 hass hass[32967]:   File "/usr/lib/python3.6/enum.py", line 533, in __new__
Aug 19 19:09:24 hass hass[32967]:     return cls._missing_(value)
Aug 19 19:09:24 hass hass[32967]:   File "/usr/lib/python3.6/enum.py", line 546, in _missing_
Aug 19 19:09:24 hass hass[32967]:     raise ValueError("%r is not a valid %s" % (value, cls.__name__))
Aug 19 19:09:24 hass hass[32967]: ValueError: 448 is not a valid APCICommand
```

elupus, 2019-08-19:
I have the docs, but also found the list of APCI commands here:
https://github.com/knxd/PyKNyX/blob/e7568fecf28953dc79aa7185b8514de0a7983422/pyknyx/stack/layer7/apci.py

So i likely hit a ADC_RES.

elupus, 2019-08-22:
This seem to crash home assistant since this is not handled by the background task.

Julius2342, 2019-08-22:
Hi @elupus,

there are two problems. The first one is that this APCICommand should be handled correctly.

The second one is that the Exception is not "hashable". The Hash function was removed from @farmio with commit 9cfbc5f25d88c65278e64dc3dd40a07c7673f6a5 . Now we know again why it is needed :-) ( #228 )

farmio, 2019-09-08:
The specs call this "A_ADC_Response-PDU" that shall only be implementd in "point-to-point connection-orioented" communication modes described in 3/3/7 Application Layer §3.5.3

From knx specs 3/3/7 Application Layer §2 APDU I'd read we shall ignore unsupported commands:

> The complete PDU for each service primitive is shown in the description of every service. Not defined and not supported Application Layer services shall be ignored by the Application Layer.

Shall we change
https://github.com/XKNX/xknx/blob/4b71c28db4cde694695da3dfad0072dd5383717f/xknx/knxip/cemi_frame.py#L143
to something like
```python
try: 
    self.cmd = APCICommand(tpci_apci & 0xFFC0)
except ValueError:
    raise CouldNotParseKNXIP("APCI not supported")
```
or even just `self.xknx.knx_logger.debug("APCI not supported: %s", (tpci_apci & 0xFFC0))`?

elupus, 2019-09-10:
Yea that should absolutely be done to avoid the ValueError leaking out. Isn't there a point in supporting to parse all known values thou? We can ignore them even if we parse them.

PR #235 (2019-09-22): Avoid leaking ValueError exception on unknown APCI command
This avoid crashing homeassistant on these types of messages.

Fixes #226
PR #271 (2020-03-25): Ignore unknown CEMI Messages
If we don't understand the CEMI Message ignore it and move on (eg. send TunnellingAck).
Previously we would raise CuldNotParseKNXIP and loose the connection because no Ack was sent.

> From 03_06_03 EMI_IMI v01.03.03 AS:
> 4.1.3.3 Exception handling: unknown messages
> General behaviour after reception of an unknown or unsupported message (this is, the cEMI Server does not know the received cEMI message code): the received message shall be ignored, this is, the cEMI Server shall generate no confirmation message.

closes #147 
fixes home-assistant/core#32394
fixes # [... cut]
closed 2019-09-22 by farmio