XKNX/xknx #228: XKNXException is an unhashable exception

elupus, 2019-08-22:
**Description of problem:**
Seem the XKNXException exception is not hashable causing problems with python traceback module: https://bugs.python.org/issue28603. Oddly that aught to have been solved already.

- [ ] using xknx standalone
- [X] using Home-Assistant knx plugin

**Version information:**
- xknx / Home-Assistant release with the issue:
- last working xknx / Home-Assistant release (if known): 0.98

**KNX installation:**

**Problem-relevant `xknx.yaml` or `configuration.yaml` entries (fill out even if it seems unimportant):**

**Traceback (if applicable):**
Aug 22 08:18:24 hass hass[35975]: TypeError: unhashable type: 'XKNXException'
Aug 22 08:18:24 hass hass[35975]:     _seen.add(exc_value)
Aug 22 08:18:24 hass hass[35975]:   File "/usr/lib/python3.6/traceback.py", line 462, in __init__
Aug 22 08:18:24 hass hass[35975]:     type(value), value, tb, limit=limit).format(chain=chain):
Aug 22 08:18:24 hass hass[35975]:   File "/usr/lib/python3.6/traceback.py", line 100, in print_exception
Aug 22 08:18:24 hass hass[35975]:     traceback.print_exception(ei[0], ei[1], tb, None, sio)
Aug 22 08:18:24 hass hass[35975]:   File "/usr/lib/python3.6/logging/__init__.py", line 533, in formatException
Aug 22 08:18:24 hass hass[35975]:     record.exc_text = self.formatException(record.exc_info)
Aug 22 08:18:24 hass hass[35975]:   File "/usr/lib/python3.6/logging/__init__.py", line 583, in format
Aug 22 08:18:24 hass hass[35975]:     return fmt.format(record)
Aug 22 08:18:24 hass hass[35975]:   File "/usr/lib/python3.6/logging/__init__.py", line 838, in format
Aug 22 08:18:24 hass hass[35975]:     msg = self.format(record)
Aug 22 08:18:24 hass hass[35975]:   File "/usr/lib/python3.6/logging/__init__.py", line 992, in emit

Julius2342, 2019-08-22:
The hash function was removed with 9cfbc5f25d88c65278e64dc3dd40a07c7673f6a5 bc nobody remembered why this was needed.

elupus, 2019-08-22:
The removed hash function seem quite invalid anyway :)

Julius2342, 2019-08-22:
It at least mitigated this error :-)

farmio, 2019-08-22:
stability through ignorance ;)

PR #234 (2019-09-22): Fix exceptions
Restore hash ability of exceptions

Also make sure we override base repr function since base repr will
just use the init value of base class

Fixes #228
commit cd676350 (2019-09-22): Fix exceptions (#234)

* Restore hash ability of exceptions

Also make sure we override base repr function since base repr will
just use the init value of base class

Fixes #228

* No point in passing message to constructor __str__ is overriden

* Add some tests for exceptions

* Add some missing docs
closed 2019-09-22 by farmio