XKNX/xknx #230: Lights with multiple status addresses?

lukaszolek, 2019-08-28:
Hi,
I have a KNX configuration with following group addresses:
- 0/0/1 - whole home
- 0/0/2 - whole ground floor
- 0/0/3 - whole upper floor
- 0/0/10 - all lights in the living room
- 1/1/15 - living room ceiling
- 1/1/16 - living room floor lights

For the living room ceiling I have configured: 1/1/15 as the main group address, and 0/0/1, 0/0/2, 0/0/10 as state group addresses. 
In Openhab I could easily configure this providing string: "1/1/15+<0/0/10+<0/0/2+<0/0/1"
How can I specify this installation in XKNX? I can see only one group_address_switch_state parameter for each item.

Thanks!

marvin-w, 2019-08-28:
Hi, 

can you share your ETS config? From what you explain I guess you're using this other addresses to get some kind of central function working.

If you turn off the light (no matter how) will the status feedback not always reflect the current value of your light?

Other than that I'm afraid, I don't think adding multiple state addresses works right now - at least not that I'm aware of.

Cheers,
pointer

lukaszolek, 2019-08-30:
The ability to specify many group addresses per item seems to be very basic and useful in KNX.
You can use it to define groups of items (e.g. lights) and manage them using group addresses too.
That's why you need to specify multiple group addresses that no matter how you turn the light on/off (using individual or "group" group address) the status will reflect the value of my light.

This is an example how my one of my switch port and wall button are defined for particular light:
[image]

And this shows how my wall button is defined for the whole floor:
[image]

So when I press the whole floor button and send "0" to 0/0/16, the switch listens to this and turns the light off (the same for all the lights on this floor), and the wall switch button also listens to this address and updates the LED status.

This is how OpenHab allows to define KNX items: https://www.openhab.org/addons/bindings/knx/#group-address-notation

lukaszolek, 2019-08-30:
Would it be difficult to change "group_address_switch_state" to accept a list of group addresses? This should solve the problem.

farmio, 2019-08-30:
Does your light actuator provide status objects?

If you turn on your floor and then manually turn off say the kitchen light then the status of floor would be on whereas kitchen switch group address is off. 
With actuator channel state objects you can always be sure the state represents the light it is controlling.

lukaszolek, 2019-08-30:
@farmio, I'm not sure. I have https://katalog.gira.de/en/datenblatt.html?ean=4010337042204
How can I check this?
Thanks!

farmio, 2019-08-30:
You have. 👍
In English it's called "Switching status feedback (switching operation)" you can find it in the technical documentation on page 29. 
Whenever the "Switching output (switching mode)" communication object is triggered (by any group address) the new state will be sent to the feedback object (we like to call it state or status around here). So when you add a group address for the feedback and use its address for xknx *_state you will always have the correct state represented. 
Be sure to set the Read flag on the feedback object if you want xknx to be able to query it - eg after a restart of HA.

lukaszolek, 2019-08-30:
Thanks a lot! I'll try to change my configuration today to check it.

closed 2020-01-21 by farmio