XKNX/xknx #238: Configuration via HA user interface

farmio, 2019-09-20:
👋 Do you want to improve the xknx home-assistant integration? Nice 😍!

Enable configuration via home-assistant ui by defining a config flow.

https://developers.home-assistant.io/docs/en/config_entries_config_flow_handler.html

marvin-w, 2020-08-31:
@farmio I personally don't feel like configuring each entity via the UI would provide a huge benefit for the user, as it's probably more painful to configure group addresses in the UI than in HA.

A very welcomed addition would nevertheless be to have the connection part in the config flow.

Optionally, we could think about completely dropping the YAML configuration inside home assistant and force the user to use our xknx.yaml.

When implemented correctly, we could setup some kind of reload check like:

* Check if the defined yaml has changed within the last x minutes
** If yes, drop all devices from `self.xknx.devices` and readd those from the YAML
*** TODO: Check if we would need to do additional checks or remove something
*** We probably need to restart the `StateUpdater` as well

What do you think? I personally don't like the way our xknx.yaml is structured... But I'd also like to avoid another breaking change... so I'm unsure.

marvin-w, 2020-08-31:
Btw. I tested the GatewayScanner. At least on my network it did not find my router. KNX IP BAOS 777

TinusK, 2020-08-31:
I think it is a Great idea!

farmio, 2020-09-01:
>  I personally don't feel like configuring each entity via the UI would provide a huge benefit for the user, as it's probably more painful to configure group addresses in the UI than in a yaml file where you can just copy paste an entire block.

The addresses themselves have to be edited anyway. If we provide a lot of named text fields for every device this would be just the same. I think UI config would allow for new users to discover all possibilities of this integration. 

> A very welcomed addition would nevertheless be to have the connection part in the config flow.

👍

> Optionally, we could think about completely dropping the YAML configuration inside home assistant and force the user to use our xknx.yaml.

I think reloading yaml config from HA is possible too.

> * Check if the defined yaml has changed within the last x minutes

I think it's not unappropriate to require triggering a reload manually after editing the xknx.yaml . After all it could be edited, but still in a draft state.

> * We probably need to restart the `StateUpdater` as well

If we properly `del` the old/unused devices we don't need to. Except for when the connection block was updated. We can't know if the states had changed while we reconnected, but this should be implemented in xknx and shall be triggered in any case when reconnecting.

> What do you think? I personally don't like the way our xknx.yaml is structured... But I'd also like to avoid another breaking change... so I'm unsure.

I like the HA syntax better too, its just shorter.

-------------

That said, I think it will be a huge task to implement this for standard devices and even more for BinarySensors that provide automation hooks.

marvin-w, 2020-09-01:
Here is the PR for the hard part (binary sensors): https://github.com/XKNX/xknx/pull/365

Tested and working locally.

da-anda, 2020-09-28:
I suppose an import feature for ETS group exports (XML, CSV) as part of the config flow is not feasible since it would be hard to create/map actual entities (covers, lights, etc) from all the groups? And I guess it is not possible to provide some sort of drag'n'drop editor inside the config flow so that users could build entities manually by wiring the imported GAs?
Nevertheless, working with imported GA names even if they can be only be provided as dropdown might simplify things a bit for less tech savvy users.
Just an idea.

marvin-w, 2020-09-28:
Hm. I'll have a look at an export later. I'm pretty sure we cannot do this in HA, but maybe we can do this with our own config.

farmio, 2020-09-28:
Just for reference https://github.com/JensVanhooydonck/xknx-configuration-importer

da-anda, 2020-09-28:
Ah right, I now recall that I think OpenHab had an importer for the ETS project file (which basically is just a ZIP). Inside there is one file that contains the topo and all GAs in a readable format.

marvin-w, 2022-01-09:
The first version of config flow is live now. I will close this here. Maybe we will continue this whole topic in the scope of another feature request.

closed 2022-01-09 by marvin-w