XKNX/xknx #250: @Enhancement Finer control what expose does

nianj, 2019-11-03:
Currently I'm trying to get the values of my 20+ 1Wire temperature sensors to the KNX bus via the expose functionality.
As every change in temperature is broadcasted on the bus, it would be nice to have control over that to reduce the number of messages written on the bus.

Additionally, a "force write" would be useful to use in HA automations. For example I want to send a sensor value every 5 or 10 minutes to the bus.
Right now, I have to use the knx send service, which is not really comfortable to use.

farmio, 2019-11-03:
👍 I like the idea of a write limit and periodic update for ExposeSensor.

farmio, 2020-08-30:
As a workaround, the knx.send service is now much more comfortable to use. It can be called with a `type` attribute just that takes values just like the config of an ExposeSensor.

farmio, 2021-04-03:
This behavior can be achieved with exposing a [HA filter](https://www.home-assistant.io/integrations/filter/) or using `knx.send`.

closed 2021-04-03 by farmio