XKNX/xknx #257: is_open / is_closed not correctly reported if position is inverted

SoftXperience, 2020-01-24:
**Description of problem:**
The cover device directs the is_open and is_closed methods to the TravelCalculator. The problem is, that the TravelCalculator doesn't know about inverted position. So, is_open returns false if the position is 0. But with inverted position 0 is open.

I'm not sure where to fix it... should we change the TravelCalculator to correctly report open/closed? Or should we change the cover device which could check itself if current_position == open_position (which is correctly stored in the device).

Please give an advice and I could submit a PR

- [x] using xknx standalone
- [x] using Home-Assistant knx plugin

**Version information:**
- xknx / Home-Assistant release with the issue: 0.104.2
- last working xknx / Home-Assistant release (if known):

**KNX installation:**
Using a cover actuator from Jung.

**Problem-relevant `xknx.yaml` or `configuration.yaml` entries (fill out even if it seems unimportant):**
- platform: knx
  name: bath
  move_long_address: '8/2/1'
  move_short_address: '8/1/1'
  position_address: '8/3/1'
  position_state_address: '8/4/1'
  invert_position: true

**Traceback (if applicable):**

farmio, 2020-01-26:
Thanks for pointing this out.
I'd just 
```python
def is_open(self):
        """Return if cover is open."""
        return self.travelcalculator.is_open() ^ self.invert_position
```
and same for is_closed in cover.py and leave travelcalculator as is (so we don't have to mess with movement directions and keep the calculator simple).

SoftXperience, 2020-01-26:
I tested it in my home assistant environment. The xknx cover device reports the correct percent value. So it reports 0 if the cover is fully open, 100 if it is fully closed and 20 if it is closed only a small part. So yes, the value is correctly scaled and calculated. Also the slider in the UI works correct to set a new position.

But the TravelCalculator doesn't know about the inverted open/close values. It receives only the travel times. In the init method it is hardcoded to position_closed = 0. So it reports is_closed = true for the value 0. Although it is inverted in my environment.

So I guess your first suggestion would be the simplest one?

farmio, 2020-01-26:
🤔 shouln't 0=open and 100%=closed be standard in KNX?
I just checked for ABB, Gira and MDT actuators.

SoftXperience, 2020-01-26:
Hmm, I don't know. I only have Jung actuators for blinds. Oh and I have an MDT actuator for my marquee and there it is the same, 100% is closed.

farmio, 2020-01-27:
@SoftXperience could you test this branch: https://github.com/farmio/xknx/tree/cover-fix-is-open
I have no blind actuator to test it in the field.

SoftXperience, 2020-01-27:
Hmm... I'm new to the python ecosystem... I have a development environment for homeassistant, bit I believe xknx is not included, isn't it? So is there an easy way to integrate your branch in my test environment? Is it easily explained? Otherwise I have to google around a bit to get it working...

farmio, 2020-01-27:
Yes. See https://xknx.io/home_assistant
It doesn't work with Hassio but with normal home-assistant installation.

SoftXperience, 2020-01-28:
Ok, I tested your branch. I still have to use the "invert_position=true" option?!

If I do, fully open is reported as position 0 and open. Fully closed is reported as 100 and close. All positions inbetween are reported as open (which I think is correct).

But now the buttons in the lovelace UI are still wrong. The up-button is closing the cover and vice versa.
Also in fully open state only the up-button is enabled.

farmio, 2020-01-28:
Yes. Unfortunately [HA defines it](https://github.com/home-assistant/home-assistant/blob/8ceef728535005cb424a44b65a41b722d5b1ba8a/homeassistant/components/cover/__init__.py#L160) contrary to knx so invert_position is still needed.
When you say it reports 0 as open do you mean in ETS?

Can you post your configuration?
Did you get this commit https://github.com/farmio/xknx/commit/3cd87f4d249a2489537f6c655e036fd906188ec5 ?

SoftXperience, 2020-01-28:
I just pulled and now the buttons have the correct direction (down-arrow closes, up-arrow opens). Sorry, didn't notice that you made further commits meanwhile.

But still one problem: in closed state, only the down-arrow is enabled (and vice versa).

`- platform: xknx
  name: work_gable
  move_long_address: '8/2/11'
  move_short_address: '8/1/11'
  position_address: '8/3/11'
  position_state_address: '8/4/11'
  invert_position: true`

farmio, 2020-01-28:
Ok. could you try this commit without invert_position: https://github.com/farmio/xknx/commit/e48c3afeb20afba760a1aa5f7e55e95fda9ac335 
I think it should be good now. 
I am constantly confused by the HA-ui slider where left is closed and right is open 🙃

SoftXperience, 2020-01-28:
The buttons (enable/disable) react correct.

But now the position is inverted again. Fully open is correctly reported as "open", but the position is reported as 100 (which leads to the confusing slider). Before the position was correctly shown as 0.

farmio, 2020-01-29:
With `invert_position` to true or false?

SoftXperience, 2020-01-29:
Still same config with inverted = true

farmio, 2020-01-29:
Good. With invert set to false it should work correct now. 
There are still some mixups regarding inversion in the xknx code. I'll try to correct them later today.

SoftXperience, 2020-01-29:
No, I'm sorry.

With inverted=false, fully open blind is in HA:
- open (correct)
- position 100 (wrong)
- down-button enabled (correct)
- down-button closes (correct)
- going to position 25 moves blind to 3/4 closed (wrong)

With inverted = true, fully open blind is in HA:
- closed (wrong)
- position 0 (correct)
- down-button disabled (wrong)
- down-button closes (correct)
- going to position 25 moves blind to 1/4 closed (correct)

So still messed up... Still think the base version with just inverted close/open states may work correctly?

Another question: I'm not sure if we really need the TravelCalculator... at least my two blinds actuators (Jung, MDT) have group addresses to set the new position and the acuator calculates and controls the travel times. So maybe we just need to write to the position group address (at least if it is set?) and ignore the calculator...

farmio, 2020-01-29:
Not all actuators support position states. 

Position 100 in HA is meant to be fully open. 
If you set position 25 in HA it should be 3/4 closed. The first try with inverted True seems ok to me.

SoftXperience, 2020-02-02:
Hmmm, this is strange... How can I think a cover with 100% does cover nothing? For me this is not logical. Are you sure that it is meant this way? Is it documented somewhere?

So maybe I have to retest your changes because I didn't thought that way.

And I'm not sure how to go on, now because I don't like it that way and maybe have my own fork with that fixed. Which would be annoying in docker environment where my productive system will run...

farmio, 2020-02-02:
Be sure to check out the fork again. I made some more commits since. 

I don't know if it is defined in the documentation, but in the code. I have already linked it here: https://github.com/XKNX/xknx/issues/257#issuecomment-579474537

I'd stay on normal HA and xknx builds to get updates. If you don't like the slider direction you can always customize a Lovelace card where it is reversed. 
The numbers the backendes deal with don't matter imho as long as everything work as supposed.

SoftXperience, 2020-02-02:
Oh sorry, I didn't see the link. Ok, feels still very strange. I used fhem and (currently) openHAB before and none of them use it that way.

Customizing the card is a good idea, but I guess I also have to invert all my scripts and automations.

I'll try to test it again this evening. Your latest changes as well as the current implementation in HA, maybe everything works as intended (although I don't like it)...

SoftXperience, 2020-02-02:
So, to be honest, I'm really confused now... I just retested the current implementation in HA.
[image]
This is the state in HA when the blind is fully closed, so that it is dark in the room.#

The buttons of the NOT inverted blink are working correctly (in MY understanding, as MY blinds are moving from top to down to close). So the down button closes the blind. So the down button is disabled correctly, as it is closed. Also the closed state is correclty reported as you can see at the icon.
Only problem is that the slider (and so the position) is reported as 0. So if I set it to position 25, the blind is 3/4 closed which is opposed to what my actuator reports and my understanding. But if this is correct from the viewpoint of HA, it is working as supposed.
And it is confusing, as the integration docs state that if the actuator reports 100% as fully closed, I have to set inverted=true

If I use inverted=true, the buttons and the open/closed state is swapped. Although the position is correct.

So, this was for the current implementation. Now I will test the same with your current branch.

SoftXperience, 2020-02-02:
So with your current branch, the inverted=false mode looks very similiar. But also with wrong (in my opinion) position.
The inverted=true is not working correctly, as I can press the down button in closed state (which will lead to no movement because the actuator knows it is already closed), but the travelcalculator interpolates the time to let the position move from 100 to 0.

But before we further test the implementation, I think we have to clarify what intended behaviour is...
Which position should be open? And if I use invert, what should be inverted? Just the position or also the direction? Also I would like an option to not use the TravelCalculator as my actuators do that on their own and I don't want to confuse the system with additional travel times and needless stop telegrams. 🤔

closed 2020-08-14 by marvin-w

Mululu, 2024-11-30:
Is there a solution to this?

The display shows 100% at the top but should be 0% and 0% at the bottom but should be 100%!?

[image]

farmio, 2024-11-30:
@Mululu there is no solution. HA defines 100% as open - this is not configurable. See https://community.home-assistant.io/t/why-the-heck-cant-we-freely-configure-a-cover-to-interpret-its-percentage-value-as-percentage-open-or-percentage-closed/225453
and https://community.home-assistant.io/t/cover-entities-that-are-0-open-and-100-closed/581979

Mululu, 2024-12-09:
OK, and why not work on a solution or a way to decide the behavior yourself?

farmio, 2024-12-09:
@Mululu feel free to do so, if you find a way. You're welcome to contribute. 
As you have probably read in the links I posted above, this would need to be done in HA Core, not in xknx.

PR #322 (2020-08-13): Cover update - up/down from bus; stop and movement direction
this is derived from #259 so 
- in xknx 0% is open and 100% is closed now
- adds is_opening and is_closing
- react to up/down from bus and start travel calculator
- add group_address_stop for actuators that have it (or don't support step)

resolves #314 
fixes #257 (I hope)
fixes #105 (without implementing the DPT1 state address - just listening for the normal group_address so initial status may be off)
closes #25
PR #259 (2020-08-14): Change Cover open/close % to Knx default values
- Change Cover and TravelCalculator to
  0% = open
  100% = closed
- Implemented `is_opening` and `is_closing` for Cover() and HA-integration
- remove inversion for 1bit telegrams
  1 bit telegrams have been inverted if `invert_position` was true.
  These are specified in knx as 0="up/open" and 1="down/close".
  I think invert_position shall only be applied to relative position.
- Removed TravelCalculator.time_set_from_outside in favor of unittest.mock.patch

fixes #257 

I've looked through the handbooks of the current blind actuators from ABB,  [... cut]