XKNX/xknx #270: DPT4ByteFloat raises TypeError

gchris, 2020-03-20:
**DPT4ByteFloat raises TypeError**

- Using Version **0.11.2**

Following example raises an exception:

```python
import asyncio
from xknx import XKNX
from xknx.io import GatewayScanner,Tunnel
from xknx.knx import PhysicalAddress,DPT2ByteUnsigned,DPT4ByteFloat,Telegram,GroupAddress,DPTBinary

async def main():

    xknx = XKNX()
    gatewayscanner = GatewayScanner(xknx)
    gateways = await gatewayscanner.scan()
    gateway = gateways[0]

    print("Found KNX gateway at {}:{}".format(gateway.ip_addr, gateway.port))

    tunnel = Tunnel(
            xknx,
            PhysicalAddress("15.15.249"),
            local_ip=gateway.local_ip,
            gateway_ip=gateway.ip_addr,
            gateway_port=gateway.port)

    await tunnel.connect_udp()
    await tunnel.connect()

    await tunnel.send_telegram(
            Telegram(
                GroupAddress('9/0/0'), 
                payload=DPT4ByteFloat.to_knx(666)))

    await tunnel.disconnect()

loop = asyncio.get_event_loop()
loop.run_until_complete(main())
loop.close()
```

If I swap `DPT4BytesFloat.to_knx(666)` into `DPTBinary(0)`, then I can see the value appear in the ETS bus monitor. What am I doing wrong?

**Traceback:**

```txt
Found KNX gateway at 192.168.1.132:3671
Traceback (most recent call last):
  File "test.py", line 43, in <module>
    loop.run_until_complete(main())
  File "/usr/lib/python3.7/asyncio/base_events.py", line 584, in run_until_complete
    return future.result()
  File "test.py", line 34, in main
    payload=DPT4ByteFloat.to_knx(666)))
  File "/usr/local/lib/python3.7/dist-packages/xknx/io/tunnel.py", line 129, in send_telegram
    success = await self._send_telegram_impl(telegram)
  File "/usr/local/lib/python3.7/dist-packages/xknx/io/tunnel.py", line 150, in _send_telegram_impl
    await tunnelling.start()
  File "/usr/local/lib/python3.7/dist-packages/xknx/io/request_response.py", line 34, in start
    await self.send_request()
  File "/usr/local/lib/python3.7/dist-packages/xknx/io/request_response.py", line 43, in send_request
    knxipframe.normalize()
  File "/usr/local/lib/python3.7/dist-packages/xknx/knxip/knxip.py", line 87, in normalize
    self.header.set_length(self.body)
  File "/usr/local/lib/python3.7/dist-packages/xknx/knxip/header.py", line 44, in set_length
    body.calculated_length()
  File "/usr/local/lib/python3.7/dist-packages/xknx/knxip/tunnelling_request.py", line 34, in calculated_length
    self.cemi.calculated_length()
  File "/usr/local/lib/python3.7/dist-packages/xknx/knxip/cemi_frame.py", line 104, in calculated_length
    raise TypeError()
TypeError
```

farmio, 2020-03-20:
The interface for the DPT Classes isn't perfectly consistent yet. 
I'd suggest to define a Device (eg. ExposeSensor) or maybe a fitting RemoteValue Class and use it's methods instead of the DPT4ByteFloat directly. 

If you really want to use it directly try `DPTArray(DPT4ByteFloat.to_knx(666))` as payload.

gchris, 2020-03-21:
Thank you for the quick answer.

Now, I tried two things:

1. Using `DPTArray(DPT4ByteFloat.to_knx(666))` which resulted in a 2-Byte telegram appearing in ETS?!
2. Changing to `ExposeSensor` which resulted in _1.001 Switch_ to appear in ETS.

Here is my code for point 2:

```python
import asyncio
from xknx import XKNX
from xknx.io import GatewayScanner,Tunnel
from xknx.devices import *
from xknx.knx import PhysicalAddress

async def main():

    xknx = XKNX()
    gatewayscanner = GatewayScanner(xknx)

    gateways = await gatewayscanner.scan()
    gateway = gateways[0]

    print("Found KNX gateway at {}:{}".format(gateway.ip_addr, gateway.port))

    tunnel = Tunnel(
            xknx,
            PhysicalAddress("15.15.249"),
            local_ip=gateway.local_ip,
            gateway_ip=gateway.ip_addr,
            gateway_port=gateway.port)

    await tunnel.connect_udp()
    await tunnel.connect()

    powerSensor = ExposeSensor(
            xknx,
            'Wattage.PV.Sensor',
            '9/0/0',
            value_type='power')

    await powerSensor.sync()

    wattage = 0.0
    while True:
        await powerSensor.set(wattage)
        print( wattage )
        await asyncio.sleep(1)
        wattage = wattage + 1

    await tunnel.disconnect()

loop = asyncio.get_event_loop()
loop.run_until_complete(main())
loop.close()
```

farmio, 2020-03-21:
Your example didn't work for me. I didn't get anything on my bus. 
I have adapted it so it works on my system with a tunneling connection (this is the default in xknx). It sends 4-byte floats to the bus. 

```python
import asyncio
import logging
from xknx import XKNX
from xknx.devices import ExposeSensor

async def main():
    logging.basicConfig(level=logging.DEBUG)

    xknx = XKNX()
    await xknx.start()

    powerSensor = ExposeSensor(
        xknx,
        'Wattage.PV.Sensor',
        '9/0/0',
        value_type='power')

    await powerSensor.sync()

    wattage = 0.0
    while True:
        await powerSensor.set(wattage)
        print(wattage)
        await asyncio.sleep(1)
        wattage = wattage + 1

    await xknx.stop()

loop = asyncio.get_event_loop()
loop.run_until_complete(main())
loop.close()
```
Sync wasn't sent because of ExposeSensor not having `group_address_state`.

gchris, 2020-03-22:
Thank you, your example worked!
Good support ;-)

closed 2020-03-22 by gchris