XKNX/xknx #279: ValueError: 523 is not a valid KNXIPServiceType

bluecmd, 2020-04-25:
**Description of problem:**

Exception thrown when programming KNX application and running HA.
```
Apr 25 15:49:10 homeassistant hass[3550]: 2020-04-25 15:49:10 ERROR (MainThread) [homeassistant.core] Error doing job: Exception in callback _SelectorDatagramTransport._read_ready()
Apr 25 15:49:10 homeassistant hass[3550]: ValueError: 523 is not a valid KNXIPServiceType
```

- [ ] using xknx standalone
- [x] using Home-Assistant knx plugin

**Version information:**
- xknx / Home-Assistant release with the issue: `HA: version | 0.108.3`
- last working xknx / Home-Assistant release (if known):

**KNX installation:**

Using a LineMaster 762 IP gateway, HA using IP Router with the same Layer-2 as the LineMaster.

**Problem-relevant `xknx.yaml` or `configuration.yaml` entries (fill out even if it seems unimportant):**

```
general:
  own_address: '9.1.1'
connection:
  routing:
```

**Traceback (if applicable):**

```
Apr 25 15:49:10 homeassistant hass[3550]: 2020-04-25 15:49:10 ERROR (MainThread) [homeassistant.core] Error doing job: Exception in callback _SelectorDatagramTransport._read_ready()
Apr 25 15:49:10 homeassistant hass[3550]: ValueError: 523 is not a valid KNXIPServiceType
Apr 25 15:49:10 homeassistant hass[3550]: During handling of the above exception, another exception occurred:
Apr 25 15:49:10 homeassistant hass[3550]: Traceback (most recent call last):
Apr 25 15:49:10 homeassistant hass[3550]:   File "/usr/lib/python3.7/asyncio/events.py", line 88, in _run
Apr 25 15:49:10 homeassistant hass[3550]:     self._context.run(self._callback, *self._args)
Apr 25 15:49:10 homeassistant hass[3550]:   File "/usr/lib/python3.7/asyncio/selector_events.py", line 965, in _read_ready
Apr 25 15:49:10 homeassistant hass[3550]:     self._protocol.datagram_received(data, addr)
Apr 25 15:49:10 homeassistant hass[3550]:   File "/home/homeassistant/app/lib/python3.7/site-packages/xknx/io/udp_client.py", line 55, in datagram_received
Apr 25 15:49:10 homeassistant hass[3550]:     self.data_received_callback(data)
Apr 25 15:49:10 homeassistant hass[3550]:   File "/home/homeassistant/app/lib/python3.7/site-packages/xknx/io/udp_client.py", line 87, in data_received_callback
Apr 25 15:49:10 homeassistant hass[3550]:     knxipframe.from_knx(raw)
Apr 25 15:49:10 homeassistant hass[3550]:   File "/home/homeassistant/app/lib/python3.7/site-packages/xknx/knxip/knxip.py", line 75, in from_knx
Apr 25 15:49:10 homeassistant hass[3550]:     pos = self.header.from_knx(data)
Apr 25 15:49:10 homeassistant hass[3550]:   File "/home/homeassistant/app/lib/python3.7/site-packages/xknx/knxip/header.py", line 34, in from_knx
Apr 25 15:49:10 homeassistant hass[3550]:     self.service_type_ident = KNXIPServiceType(data[2] * 256 + data[3])
Apr 25 15:49:10 homeassistant hass[3550]:   File "/usr/lib/python3.7/enum.py", line 310, in __call__
Apr 25 15:49:10 homeassistant hass[3550]:     return cls.__new__(cls, value)
Apr 25 15:49:10 homeassistant hass[3550]:   File "/usr/lib/python3.7/enum.py", line 564, in __new__
Apr 25 15:49:10 homeassistant hass[3550]:     raise exc
Apr 25 15:49:10 homeassistant hass[3550]:   File "/usr/lib/python3.7/enum.py", line 548, in __new__
Apr 25 15:49:10 homeassistant hass[3550]:     result = cls._missing_(value)
Apr 25 15:49:10 homeassistant hass[3550]:   File "/usr/lib/python3.7/enum.py", line 577, in _missing_
Apr 25 15:49:10 homeassistant hass[3550]:     raise ValueError("%r is not a valid %s" % (value, cls.__name__))
Apr 25 15:49:10 homeassistant hass[3550]: ValueError: 523 is not a valid KNXIPServiceType
```

farmio, 2020-04-25:
Hi! 
Thank you for reporting. As you said it should only occur when programming devices (523 is A_Memory_Read-service). 
The error should be fixed by #271 - which will be included in the next release.

bluecmd, 2020-04-25:
Perfect! :-) Let's close this and I can re-open it if the referenced fix for some reason doesn't work for this issue.

Thanks for the quick reply and really cool library!

farmio, 2020-12-21:
So this is back again when using routing. It appears when I choose an Interface in ETS (open dropdown menu).

PR #541 (2020-12-21): Unsupported knxip service types

## Description

- Handle exceptions from received unsupported or not implemented KNXIP Service Type identifiers
- move Unhandled and Unsupported KNXIPFrame logs from xknx.log to xknx.knx logger
- added some more KNXIPServiceType codes to the Enum (found at the [wireshark repo](https://github.com/wireshark/wireshark/blob/22e7ddb63789ff603641be116ee24834ca7631f9/epan/dissectors/packet-knxip.c)
- 
Fixes #279 for real this time 😉

## Type of change

- [x] Bug fix (non-breaking change which fixes an issue)
- [ ] New feature (non-breaking change which adds functional [... cut]
closed 2020-12-21 by farmio