XKNX/xknx #283: Support exposing further DPTs

sla89, 2020-05-07:
Hello together,

I have a home assistant installation running and I want to expose some data to my bus which is of type DPT 1.001 (switch). 

My currently used workaround is to define a switch in my configuration and toggle it but when devices query the state of the group address nobody answers. 

I want to extend to code of xknx to support this and later also extend home assistant (I have already found the place in the source code there).

Am I correct when I do something similar to `expose_sensor.py` called `expore_switch.py` which then can support the different DPTs used for switches.

Thank you,
Stefan

farmio, 2020-05-07:
This is already possible. 
In HA just use 'type: "binary"` in your expose.

closed 2020-05-09 by farmio

sla89, 2020-05-09:
Does this also work if the "sensor" is not of type binary like the position of the sun? Do you know that?

farmio, 2020-05-09:
Type binary only works for binary values. There are plenty types for you to choose. 
What is the `state` of the position of the sun entity? What DPT do you want to get on the bus?

sla89, 2020-05-11:
Well normally the DPT for day/night would be 1.024 but 1.002 (bool) would do it too.
The state can be 'above_horizon' and 'below_horizon': https://www.home-assistant.io/docs/scripts/conditions/#sun-condition.

To have it as a binary I would need a template. Maybe this would be easier than implementing support for this in xknx and HA.

I found a solution here: https://community.home-assistant.io/t/exposing-a-boolean-value-to-a-knx-group-address-possible/61514/7

I will go for this approach. Thank you for your support.