XKNX/xknx #284: ConnectionStateResponse cannot be parsed (Unkown data/unexpected trailing data)

henneaux, 2020-05-16:
**Description of problem:**
In the process of trying xknx to talk with a new KNX installation, which I have not access to in terms of ETS (also lack experience with KNX in general). But, I first tried 'example_gatewayscanner.py': 

```
> Gateway found: XXTER / 20.0.1.10:3671
> Device supports tunneling
```
Next, I tried 'example_tunnel.py' which resulted in the following error/traceback which implies that it does not like the response of tunnel.connectionstate(). Below a TCPDUMP/PCAP of all relevant packets (port 3671), including the response with 'unknown/unexpected trailing data'.
```
Connecting to 20.0.1.10:3671 from 20.0.1.11
<CouldNotParseKNXIP description="KNXIP data has wrong length" />
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/xknx/io/udp_client.py", line 88, in data_received_callback
    knxipframe.from_knx(raw)
  File "/usr/local/lib/python3.7/site-packages/xknx/knxip/knxip.py", line 81, in from_knx
    raise CouldNotParseKNXIP("KNXIP data has wrong length")
xknx.exceptions.exception.CouldNotParseKNXIP: <CouldNotParseKNXIP description="KNXIP data has wrong length" />
Error: KNX bus did not respond in time to request of type 'ConnectionState'
Task was destroyed but it is pending!
task: <Task pending coro=<Tunnel.do_heartbeat() done, defined at /usr/local/lib/python3.7/site-packages/xknx/io/tunnel.py:223> wait_for=<Future pending cb=[<TaskWakeupMethWrapper object at 0x76308810>()]>>
```
Not sure how to troubleshoot further, as my experience with KNX is limited. Hope someone can steer me in the right direction, thanks!

**Version information:**
using xknx standalone (xknx-0.11.3) with Python 3.7.7 on Raspberry Pi

**KNX installation:**
Xxter (https://www.xxter.com/en/productinfo/)
Only access to the Xxter module (webUI) to enable KNX tunnel access. Don't have access to ETS, the installation is managed by an implementation partner.

**Problem-relevant:**
TCPDUMP/PCAP of all relevant packets (port 3671) when running 'example_tunnel.py'. 
Unknown data in the ConnectionStateResponse (all other requests/responses seem to be OK). 

> 22	4.043780	20.0.1.10	20.0.1.11	KNXnet/IP	60	? ConnStateResp 40 OK

Attached the full TCPDUMP/PCAP file [xxter-3671.pcap.zip](https://github.com/XKNX/xknx/files/4639116/xxter-3671.pcap.zip)

**Traceback (if applicable):**
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/xknx/io/udp_client.py", line 88, in data_received_callback
    knxipframe.from_knx(raw)
  File "/usr/local/lib/python3.7/site-packages/xknx/knxip/knxip.py", line 81, in from_knx
    raise CouldNotParseKNXIP("KNXIP data has wrong length")

farmio, 2020-05-17:
Hi!

Is this always the case (is it easily reproduceable)? It doesn't look conform to the knx standard for me. I'd check with the manufacturer of your interface - maybe they even already have some kind of firmware update.

You could try to do "unsafe mode" by commenting out https://github.com/XKNX/xknx/blob/a6d0e551168884a822826d7b7f33bf9e07497817/xknx/knxip/knxip.py#L80 and see if every ConnectionStateResponse (this is used periodically as a heartbeat in tunnelling connections) is malformed this way.

You can also check if a connection from ETS and gets closed after such a ConnectionStateResponse. (ETS Demo version is free after registration and has fully functional Busmonitor).

PS: You seem to use a non-private IP address range in your network. This is probably not the cause of this problem, but it may cause networking problems in the future (20.0.0.1/8 belongs to Microsoft).

henneaux, 2020-05-17:
Hi, thanks! 

Yes, the ConnectionStateResponse is always malformed with the Unknown data/Unexpe'**x**'ted trailing data (even the error message is malformed, consistency!). I noticed that there is a small firmware update for the interface, will check. Will also check the ETS Demo version when I have physical access to the installation.

Thanks for the heads-up on the IP range, didn't know.. kind of went for a subsequent VLAN numbering schema. 

Will close the issue, as it is likely not xknx related. Thanks for your help, appreciated!

closed 2020-05-17 by henneaux