XKNX/xknx #285: Controlling HDL DALI Gateway

xBeAsTx, 2020-05-21:
**Description of problem:**
I have HDL DALI gateway which controlling RGBW lights. But instead of DPT 251.600 addresses it uses separate DPT 1.001/DPT 5.001 for each color R/G/B/W. So how I can integrate it with your library? Can we make some abstract class which control each color by separate address and throw it up to Home-Assistant?

https://b2b.hdl-automation.cz/cz/ke-stazeni/produkty/manualy/knx/komunikacni-rozhrani/appl-program-information-m-dali-1_(v1-1)

- [ ] using xknx standalone
- [x] using Home-Assistant knx plugin

**Version information:**
- xknx / Home-Assistant release with the issue:
0.109.6

- last working xknx / Home-Assistant release (if known):
0.109.6

**KNX installation:**
KNX IP BAOS 777

[dali_groups.txt](https://github.com/XKNX/xknx/files/4664238/dali_groups.txt)
(it's xml export)

Lennart99, 2020-12-15:
I have created a way to control RGBW lights with individual addresses for the colors, you can try it if you would want to. The code for the home assistant plugin is in the PR too.

closed 2020-12-21 by farmio

xBeAsTx, 2020-12-22:
Wow! Great, thanks a lot. I will try later...