XKNX/xknx #293: KNX integration stops working after TypeError

FredericMa, 2020-05-23:
**Description of problem:**

- [ ] using xknx standalone
- [x] using Home-Assistant knx plugin

I'm migrating my OpenHAB installation to Home Assistant and both KNX components are running at the same time. It is known that the KNX binding from OpenHAB sends rubish every now and then to the KNX bus. This was generating error logs in HA but everything kept on going so I didn't care.
Now, since release 0.110.1 with XKNX 0.11.3, no sending or receiving seems to work after an faulty message is received. Even with DEBUG logging enabled I don't see any entries in the log if I switch a light for example.

This is the junk (at least to me) OpenHAB sends on the KNX bus:
[image]

These are the two different tracebacks I'm seeing in HA:
```
2020-05-23 08:27:20 ERROR (MainThread) [homeassistant.core] Error doing job: Exception in callback _ProactorDatagramTransport._loop_reading(<_OverlappedF....102', 3671))>)
Traceback (most recent call last):
  File "c:\python3\Lib\asyncio\events.py", line 81, in _run
    self._context.run(self._callback, *self._args)
  File "c:\python3\Lib\asyncio\proactor_events.py", line 578, in _loop_reading
    self._protocol.datagram_received(data, addr)
  File "d:\homeassistant\lib\site-packages\xknx\io\udp_client.py", line 56, in datagram_received
    self.data_received_callback(data)
  File "d:\homeassistant\lib\site-packages\xknx\io\udp_client.py", line 90, in data_received_callback
    self.handle_knxipframe(knxipframe)
  File "d:\homeassistant\lib\site-packages\xknx\io\udp_client.py", line 99, in handle_knxipframe
    callback.callback(knxipframe, self)
  File "d:\homeassistant\lib\site-packages\xknx\io\routing.py", line 34, in response_rec_callback
    if knxipframe.body.src_addr == self.xknx.own_address:
  File "d:\homeassistant\lib\site-packages\xknx\telegram\address.py", line 62, in __eq__
    raise TypeError()
TypeError
```

```
2020-05-23 12:14:47 ERROR (MainThread) [homeassistant.core] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "d:\homeassistant\lib\site-packages\xknx\core\telegram_queue.py", line 68, in run
    await self.process_telegram(telegram)
  File "d:\homeassistant\lib\site-packages\xknx\core\telegram_queue.py", line 96, in process_telegram
    await self.process_telegram_incoming(telegram)
  File "d:\homeassistant\lib\site-packages\xknx\core\telegram_queue.py", line 121, in process_telegram_incoming
    for device in self.xknx.devices.devices_by_group_address(
  File "d:\homeassistant\lib\site-packages\xknx\devices\devices.py", line 32, in devices_by_group_address
    if device.has_group_address(group_address):
  File "d:\homeassistant\lib\site-packages\xknx\devices\datetime.py", line 58, in has_group_address
    return self.group_address == group_address
  File "d:\homeassistant\lib\site-packages\xknx\telegram\address.py", line 62, in __eq__
    raise TypeError()
TypeError
```

Both are caused by a TypeError at "xknx\telegram\address.py", line 62.

I solved it for me by catching the errors in XKNX:
- In udp_client.py:
```python
        def datagram_received(self, data, addr):
            """Call assigned callback. Callback for datagram received."""
            if self.data_received_callback is not None:
                try:
                    self.data_received_callback(data)
                except Exception as e:
                    self.xknx.logger.error("Error while processing telegram: %s" % e)
```
- In telegram_queue.py:
```python
    async def process_telegram(self, telegram):
        """Process telegram."""
        self.xknx.telegram_logger.debug(telegram)
        try:
            if telegram.direction == TelegramDirection.INCOMING:
                await self.process_telegram_incoming(telegram)
            elif telegram.direction == TelegramDirection.OUTGOING:
                await self.process_telegram_outgoing(telegram)
        except XKNXException as ex:
            self.xknx.logger.error("Error while processing telegram %s", ex)
        except Exception as ex:
            self.xknx.logger.error("Error while processing telegram: %s", ex)
```

This is just a duct tape solution for me. Maybe it is cleaner to make the TypeError also a XKNXException and catch it also in the udp_client as it is not catched in 0.11.3?
It looks like these are the 2 occurrences where the __eq__ of an address is called.

**Version information:**
- xknx / Home-Assistant release with the issue: 0.11.3 / 0.110.1
- last working xknx / Home-Assistant release (if known): 0.11.2 / 0.109.3

**KNX installation:**

MDT IP Router

**Problem-relevant `xknx.yaml` or `configuration.yaml` entries (fill out even if it seems unimportant):**
```yaml
knx: 
  routing:
    local_ip: 192.168.xxx.xxx
```

**Traceback (if applicable):**

farmio, 2020-05-23:
Are HA and OpenHab running on the same device (IP address)?

FredericMa, 2020-05-24:
Yes they are running on the same machine. I just tested it running it on another machine and I see the same error messages.
My IP-router only supports one tunneling connection so I only configured HA to use tunneling. I can confirm that in that case the error doesn't appear in HA and everything keeps on working.

I do recall that the OpenHAB binding is configured to ping the devices on the bus every 120 seconds. Probably that's where the messages coming from.

The error messages I was seeing in the log on HA 0.109.3 (and lwoer) with XKNX 0.11.2 looked like this:
```
2020-05-23 14:37:00 ERROR (MainThread) [xknx.log] <CouldNotParseKNXIP description="CEMI too small" />
Traceback (most recent call last):
  File "C:\Projects\ha_core\lib\site-packages\xknx\io\udp_client.py", line 89, in data_received_callback
    knxipframe.from_knx(raw)
  File "C:\Projects\ha_core\lib\site-packages\xknx\knxip\knxip.py", line 78, in from_knx
    pos += self.body.from_knx(data[pos:])
  File "C:\Projects\ha_core\lib\site-packages\xknx\knxip\cemi_frame.py", line 116, in from_knx
    return self.from_knx_data_link_layer(raw)
  File "C:\Projects\ha_core\lib\site-packages\xknx\knxip\cemi_frame.py", line 122, in from_knx_data_link_layer
    raise CouldNotParseKNXIP("CEMI too small")
xknx.exceptions.exception.CouldNotParseKNXIP: <CouldNotParseKNXIP description="CEMI too small" />
```
and this
```
2020-05-23 14:37:00 ERROR (MainThread) [xknx.log] <CouldNotParseKNXIP description="APCI not supported: 0b100001100000000" />
ValueError: 17152 is not a valid APCICommand

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Projects\ha_core\lib\site-packages\xknx\knxip\cemi_frame.py", line 147, in from_knx_data_link_layer
    self.cmd = APCICommand(tpci_apci & 0xFFC0)
  File "c:\Python38\lib\enum.py", line 304, in __call__
    return cls.__new__(cls, value)
  File "c:\Python38\lib\enum.py", line 595, in __new__
    raise exc
  File "c:\Python38\lib\enum.py", line 579, in __new__
    result = cls._missing_(value)
  File "c:\Python38\lib\enum.py", line 608, in _missing_
    raise ValueError("%r is not a valid %s" % (value, cls.__name__))
ValueError: 17152 is not a valid APCICommand

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Projects\ha_core\lib\site-packages\xknx\io\udp_client.py", line 89, in data_received_callback
    knxipframe.from_knx(raw)
  File "C:\Projects\ha_core\lib\site-packages\xknx\knxip\knxip.py", line 78, in from_knx
    pos += self.body.from_knx(data[pos:])
  File "C:\Projects\ha_core\lib\site-packages\xknx\knxip\cemi_frame.py", line 116, in from_knx
    return self.from_knx_data_link_layer(raw)
  File "C:\Projects\ha_core\lib\site-packages\xknx\knxip\cemi_frame.py", line 149, in from_knx_data_link_layer
    raise CouldNotParseKNXIP(
xknx.exceptions.exception.CouldNotParseKNXIP: <CouldNotParseKNXIP description="APCI not supported: 0b100001100000000" />
```

Maybe this is some valuable information.

It looks like running HA in routing mode and programming a device via ETS in routing mode reproduces the issue. I toggled a write flag on one of my devices as a test, programmed it and as a result I saw the error messages mentioned in my initial message and the integration stopped working.

PR #295 (2020-07-11): Multicast on macOS, routing bugfix
- This should enable multicast on macOS and fix a bug where unknown cemi frames raise a TypeError on routing connections.
- Removed `bind_to_multicast` option as it seemed to only be useful on macOS. (see conversation in #256 )
- adds RoutingIndication class
- RoutingIndication and TunnellingRequest class catch UnsupportedCEMIError and set their self.cemi to None. When unpacking the telegram in Tunnel() or Routing() this ich checked for None to handle it correctly (Routing: drop / Tunneling: send ACK, drop)

fixes #293
closed 2020-08-18 by farmio