XKNX/xknx #294: German Umlaute cause ValueError

josiasmontag, 2020-05-24:
**Description of problem:**

I am getting the following exception if I try to expose a string with *Umlaute* to KNX.

```
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/xknx/dpt/dpt_string.py", line 33, in to_knx
    raise ValueError
ValueError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/knx/__init__.py", line 385, in _async_entity_changed
    await self._async_set_knx_value(new_state.state)
  File "/usr/src/homeassistant/homeassistant/components/knx/__init__.py", line 400, in _async_set_knx_value
    await self.device.set(value)
  File "/usr/local/lib/python3.7/site-packages/xknx/devices/expose_sensor.py", line 74, in set
    await self.sensor_value.set(value)
  File "/usr/local/lib/python3.7/site-packages/xknx/remote_value/remote_value.py", line 120, in set
    payload = self.to_knx(value)  # pylint: disable=assignment-from-no-return
  File "/usr/local/lib/python3.7/site-packages/xknx/remote_value/remote_value_sensor.py", line 230, in to_knx
    return DPTArray(self.DPTMAP[self.value_type].to_knx(value))
  File "/usr/local/lib/python3.7/site-packages/xknx/dpt/dpt_string.py", line 40, in to_knx
    raise ConversionError("Cant serialize %s" % cls.__name__, value=value)
xknx.exceptions.exception.ConversionError: <ConversionError description="Cant serialize DPTString" value="Überwiegend bewölkt"/>
```

- [ ] using xknx standalone
- [x] using Home-Assistant knx plugin

**Version information:**
- xknx / Home-Assistant release with the issue: 0.110.1
- last working xknx / Home-Assistant release (if known): -

**KNX installation:**

**Problem-relevant `xknx.yaml` or `configuration.yaml` entries (fill out even if it seems unimportant):**

```
    - type: 'string'
      entity_id: 'sensor.owm_wetter_forecast'
      address: '0/3/1'
```

farmio, 2020-05-24:
🤔 it is also > 14 Characters. Have you tried to set up a notification instead of an expose sensor? This should sanitize the input. https://www.home-assistant.io/integrations/notify.knx

josiasmontag, 2020-05-24:
It seems to be the 14 characters limit. Sorry, I was not aware of that.

closed 2020-05-24 by josiasmontag

farmio, 2020-05-24:
So how did you avoid it? We should probably write a hint in the documentation.

josiasmontag, 2020-05-24:
I am just using strings <= 14 chars. Maybe just cutting the string after 14 chars would be better than throwing an exception.

farmio, 2020-05-24:
Yes. We do this in the Notification device. ExposeSensor tries to parse the raw value. I'll have a look into changing that.