XKNX/xknx #296: Name of KNX devices not definable via config file

spacegaier, 2020-05-27:
**Description of problem:** 
I am using xknx in conjunction with Home-Assistant (HA). In HA each KNX entity has a unique "Entity ID" and a "Name". I would like for the IDs to be technical (e.g. "light.office_ceiling1") and the name to be user friendly (e.g. "Office Main Lamp"). However, currently the config logic of xknx is always using the YAML structure entry name whereas if you were to use xknx standalone and create Python objects you can actually pass a user friendly name (see first example [here](https://xknx.io/light)).

I would be willing to prepare a pull-request that contains the following, but wanted to get some feedback first.

1. Add a new member "id" to class "[Device](https://github.com/XKNX/xknx/blob/a6d0e551168884a822826d7b7f33bf9e07497817/xknx/devices/device.py#L10)"
2. Add support for a new optional array element "name" to all device configs, e.g.:
> groups:
     light:
      office.ceiling1:
      {
        name: "Office Main Lamp",
        group_address_switch: "1/1/29",
        group_address_switch_state: "1/4/28"
      }

3. Adjust the "[def from_config(cls, xknx, name, config)](https://github.com/XKNX/xknx/blob/a6d0e551168884a822826d7b7f33bf9e07497817/xknx/core/config.py#L86)" logic to do the following:
3.1 Set the new member "id" with the YAML element name (as "technical key").
3.2. Check if the new element "name" is present in config: 
Yes => Use it as the name of the device
No => Fallback to current behavior of taking the YAML element name as device name (to remain backwards-compatible)
4. For the non config approach with direct Pythons device instances: Automatically set the new "id" to the "name" if no explicit value for "id" is provided (backwards compatible).

With those changes the xknx devices would then contain a technical "id" and a user friendly "name" that could then be used by HA (separate change in HA required to actually read the new info, that I would also take care of).

- [X] using xknx standalone
- [X] using Home-Assistant knx plugin

**Version information:**
- xknx / Home-Assistant release with the issue: latest (0.11.3 / 0.110.3)

Julius2342, 2020-05-27:
If that solves the "unique-id" problem within hass it sounds like a reasonable approach. @farmio : What do you think?

Have you confirmed, that "unique-id" within HASS has to be unique in the system or does it have to be unique world-wide. (like e.g. a MAC-Address?)

Should we be more strict with the IDS, e.g. by slugifying them? I guess that would also have side effects, with existing automations.

spacegaier, 2020-05-27:
@Julius2342 The entity ID has to be unique within a HA instance (more specifically its internal entity repository). See: https://developers.home-assistant.io/docs/entity_registry_index/

My proposal however is not trying to solve a uniqueness problem. The key/ID would still remain the same value as before from the xknx config files (the actual YAML element names) to not break any existing HA/xknx users setups, hence I would not slugify/etc. them. The user still has to ensure that those remain unique, but we then offer the possibility to assign optional meaningful (= user-friendly) names to the devices, so that the unique IDs can be treated as pure technical keys.

farmio, 2020-05-27:
I'm not sure about this being a uniqe id. From https://developers.home-assistant.io/docs/entity_registry_index:
> It is therefore very important that the unique ID is unique! It is also important that it is not possible for the user to change the unique ID, because that means it will lose all its settings related to it.

What purpose does it serve to have different enitity_id (not unique_id) and name? Is overriding the friendly_name with [customize](https://www.home-assistant.io/docs/configuration/customizing-devices/) not an option?

spacegaier, 2020-05-27:
From the HA [entity registry coding](https://github.com/home-assistant/core/blob/c7e97f0cf85fd86ea6da0cfac301b3a4842d31a5/homeassistant/helpers/entity_registry.py#L3):
> Entities are uniquely identified by their domain, platform and a unique id provided by that platform.
In the context of the KNX integration the whole entity ID (consisting of the various sub elements including the unique ID) is provided (indirectly) by the xknx config file.

Sticking with the example of my original post, this is what HA currently gets from xknx:
Name: office.ceiling1
Entity ID: light.office_ceiling1

[image]

So you can see "light" is here the domain and "office_ceiling1" is the unique ID (HA automatically converts here the dot from the xknx config into an underscore). In the dashboard setup everything is tied to the entity ID, but the name "office.ceiling1" is then shown at runtime in the dashboard which of course is not user friendly. 

It's true that there is the option to separately specify a name overwrite (friendly_name) in a separate configuration file in HA. But from my point of view, that is a last ditch effort for HA integrations that cannot provide something more meaningful. But since for the KNX integration we anyway have to already list everything in a xknx file to integrate, it would be much more streamlined and efficient to have it in one place and have the KNX integration provide a good starting point in terms of names. You then still always have the option to overwrite the names via the customizing with friendly_names if required. But that's then another step at runtime that we could otherwise skip.

farmio, 2020-05-27:
You don't have to use the xknx config file. You can always configure your devices from HA configuration.yaml (and imports) and set up meaningful device names like "Bathroom Mirror Left".

spacegaier, 2020-06-06:
That seems to be valid, although upon the first try I got an error reg. invalid entity_id.

I made some more complete tests (hence the delayed response) both with HASS vanilla configuration and with XKNX configuration.

[image]

In both approaches the entity_id is currently directly tied to the name, which I think is not ideal, but that looks more to be a general HASS topic. Neither approach automatically adds the platform into the entity_id apart from test scenario 1.
I still think that having the option to specify the ID within XKNX would be great (if the HASS integration then makes use of it), but I will probably create a discussion in HASS first to understand if there is a specific reason why you can only specify names in the vanilla configuration and if there is anything that speaks against integrations/addons having a more flexible configuration option in general.

farmio, 2020-06-07:
I'm not sure I understand you correctly. What is your desired configuration pattern? 
Do you want the entity_id to be like `light.knx.office_1` and the name (friendly_name) to be `Office 1`? How should it be configured? 

Remember that changing this would break all existing HA automations and templates using knx entities, so I wouldn't do it without good reason.

spacegaier, 2020-06-07:
My goal in general is just to have entity_id and name independent, so that I can just switch out the name (e.g. translate later or shorten it to fit the UI) without it breaking anything.

So in my ideal world I should be able to end up with this (example from very first post):

> entity_id: light.office_ceiling1
> name: LED Panel

```YAML
groups:
  light:
    office.ceiling1:
    {
      name: "LED Panel",
      group_address_switch: "1/1/29",
      group_address_switch_state: "1/4/28"
     }
```

My proposal I made on top would not break any existing HA installations, since as long as the new attribute "name" is not used, the behavior of how the entity_id and name are derived and passed on to HA would be as is now. Only once the new attribute "name" would be used in the configuration would the behavior change. At that point that would break indeed existing setups.

Reg. "knx" in the entity_id. I think your example with a second dot in the entity_id is not allowed. At least that was my understanding from the HA documentation and I have not seen that in HA so far.

farmio, 2020-06-07:
I like it. 
If "name" only affects the friendly_name I don't think any logic would break (only strings in UI possibly - but that's the whole point). 

Maybe there should even be a "friendly_name" attribute for this. Currently device.name is used to identify the device in our log messages - I think these should hold the entity_id rather than the friendly_name.

spacegaier, 2020-06-14:
> Maybe there should even be a "friendly_name" attribute for this.

I am not sure what exactly you mean here in this context. Would you like to rename the new "name" configuration parameter to "friendly_name" to be more in line with HA? If that is the argument, I would still go for "name" since more generic and therefore suited for non-HA applications.

Log messages: Why not both, something like: "<name> (<entity ID>)"?

marvin-w, 2020-09-10:
I'd like to incorporate this change in https://github.com/XKNX/xknx/issues/379. It would then be included in the new xknx.yaml version 2 as `friendly_name` for each device type. We will adjust the `Device` class and all it's subclasses to include the new attribute and also make sure that HA uses the attribute accordingly.

Additionally, we could think of improving some log messages to also log the `friendly_name` and not only the `name`. However, the `friendly_name` attribute will be optional so I'm unsure whether this is a good idea.

farmio, 2020-09-10:
Im not sure a user-configurable string is what HA would see as a qualified unique-id. 
What options exactly depend on it? I think at one time it was required for config flow but this is not the case anymore. 
Why not just use the entity-name (name)? One wouldn't have to change it anyway when there is also friendly_name.

spacegaier, 2020-09-10:
I need to double-check but I think setting a custom name (not talking about friendly_name here) and even custom entity ID is only possible when you have a unique_id defined. I agree ideally the unique_id is directly coming from a device serial number or MAC address, but we do not have that with KNX.

Those `unique_id` changes were not purely focused on HA, but instead I think it makes sense for XKNX objects in general to have an ID and a separate name/description.

farmio, 2020-09-10:
😂 how many names does one device need? name, friendly_name, entity_id, custom_name, unique_id...

In xknx you can always use the device directly eg. assign a device (object reference) to a variable.

marvin-w, 2020-09-10:
I recently did something very similar but then throw it away cause it is really not needed. In XKNX the name is the entity_id. With the additional friendly_name we simply have a way to not use the customization.yaml.

> We already have a lot of HA specific attributes and constants in our code - I'd like this not to grow too much. Instead these things should be handled from the integrations code whenever possible.

Exactly the opposite is true for HA. They want us to have as much code as possible in the lib - which makes sense from their stand point.

Concerning the `unique_id` @spacegaier: Having it in the config doesn't make much sense to me.

> [..] It is also important that it is not possible for the user to change the unique ID, because that means it will lose all its settings related to it.

Source: https://developers.home-assistant.io/docs/entity_registry_index/#unique-id-requirements

But I see your point.

farmio, 2020-09-10:
The serial number of a 20-fold switch actuator isn't going to be much help here. Using just the IA would yield the exact same results. 
If we combined IA (user configured?) and all GAs a device uses we could have uniqueness - until something in the device is changed (eg a GA is added) which is the opposite of what we want. 
There are even xknx-devices that are not representing one, but a couple of knx physical devices. Think of climate - a room controller, temperature sensor and maybe switch actuators. Or a logic engine that provides room controllers for every room in a building...

marvin-w, 2020-09-10:
Well then we just don't have real uniqueness available and should just not implement `unique_id`.

farmio, 2020-09-10:
Thats my point. We can, and maybe should, however make sure that `name` is only used once (at least for every device class).

marvin-w, 2020-09-11:
I think the name should be unique between all device classes. Otherwise devices will get overridden if the name is duplicated in xknx.devices.

da-anda, 2020-09-25:
sorry for kinda hijacking this issue, but since it is kinda related to the unique ID discussion. In HA 0.115.x and XKNX 0.13 there is now the requirement to use XKNX config files and no longer use the `platform: knx` way. Due to this change, it is no longer possible to use `entity_namespace` ( see https://www.home-assistant.io/docs/configuration/platform_options/#entity-namespace ) and I had to rename ALL of my entities (which kinda sucked). I used namespaces to differentiate between apartments (I have a two apartment home) while keeping the `name` unique to the apartment. Example:
```
- name="Livingroom"
  entity_namespace="Home"
  address: 0/0/0
- name="Livingroom"
  entity_namespace="Parents"
  address: 1/0/0
```
the entity namespace was supposed to generate IDs like `home_livingroom` while the name/friendly_name reported in HA was just supposed to be `Livingroom`.

Getting support for those namespaces back in when you refactor the entity_id vs name handling would be nice. Thanks.

marvin-w, 2020-09-25:
@da-anda: From the docs: "These options are being phased out and are only available for single platform integrations."

Why didn't you just rename the `name` with the namespace that you had before. This would have had exactly the same result. I don't understand.

We will not implement something like this in the library if it's already deprecated in HA.

da-anda, 2020-09-25:
@marvin-w argh, sorry, kinda missed that this is deprecated now - so sorry for the noise. I have had renamed all my entities and prefixed them with the namespace in order to keep everything working, but it was a huge PITA to do and I actually preferred the namespace concept.

And yes, I have already set a friendly_name in customization.yaml (out of necessity), but this should not have been necessary in the first place when the ID would differ from the name like proposed in this issue.

spacegaier, 2020-10-23:
I want to pick this topic up again, after I have been neck-deep in about 40 HA frontend PRs over the last weeks. 

My understanding following the various threads / issues here is (and please correct me if I got something wrong):
1. `xknx.yaml` is the way to go for the future
2. There is no plan to make use of `unique_id` so all the options that brings inside HA such as easy entity renaming and icon overrides (considered too risky as we do not have a real serial number?). Using the GA for that and living with the small chance that a user will change those at some point and then has to clean up a few things inside HA is not an option?
3. There is no plan or support for adding a separate name attribute inside `xknx.yaml`?

So that means that if I want to have nicer / easy-to-read entity names in HA (not talking about the actual front end labels since here apparently the HA customize option is the only way to go then), is to not specify "technical" keys in the YAML but directly the names.

So instead of something like this (like the docs would propose):
```
  light:
    Livingroom.Corner_Lamp_1:
      {
        group_address_switch: "1/1/29",
        ...
```
Directly go with this:
```
  light:
    Livingroom Corner Lamp 1:
      {
        group_address_switch: "1/1/29",
        ...
```

marvin-w, 2020-10-24:
Hey,

1. Yes, but the new version of the config will be. The old config looks messy. I started with this but it's not yet usable.
2. There are plans to support opting in for `unique_id` usage. However, you will need to actively opt in as there are chances that it might not work due to missing unique identifiers.
3. We will add a `friendly_name` attribute which will then be shown in the HA frontend. `name` will still be the `entity_id`.

For the new config check `xknx_v2.yaml`. Would that suit your needs?

farmio, 2021-04-03:
With unique_id you can now edit entity_id.

closed 2021-04-03 by farmio