XKNX/xknx #297: Connection details in xknx.yaml file not used

Tommatheussen, 2020-05-29:
**Description of problem:**
I was about to move my configuration from Home Assistant to a config file (xknx.yaml). When I enter my connection details, it seems they are not being used correctly. I'm getting problems with my instance reporting that no gateway was found.

I did some short debugging yesterday evening and found that the HA plugin seems to call the `knxip_interface.start` method with it's defined connection_config. However, since I used the config file to set up the system, it can't actually find the details in the `config[DOMAIN]` dictionary, as it only holds the config file location entry.
- [ ] using xknx standalone
- [x] using Home-Assistant knx plugin

**Version information:**
- xknx / Home-Assistant release with the issue: 0.110.3
- last working xknx / Home-Assistant release (if known): Never tried config file before

**KNX installation:**

Gateway used is MDT SCN-IP000.02. Nothing else configured in the system yet (using direct settings in HA config works so I don't think that's the problem)

**Problem-relevant `xknx.yaml` or `configuration.yaml` entries (fill out even if it seems unimportant):**
```
-- xknx.yaml --
connection:
  tunneling:
    gateway_ip: 192.168.150.200
    local_ip: 192.168.255.254
```

```
-- configuration.yaml --
knx:
  config_file: 'xknx.yaml'
```
**Traceback (if applicable):**
`2020-05-29 09:54:36 WARNING (MainThread) [custom_components.knx] OrderedDict([('config_file', 'xknx.yaml'), ('rate_limit', 20), ('state_updater', True)])`
I've overridden the build in component and added this logging at line: https://github.com/XKNX/xknx/blob/a6d0e551168884a822826d7b7f33bf9e07497817/home-assistant-plugin/custom_components/xknx/__init__.py#L202
There you can see that there actually is no connection info in the config for the domain, and thus it falls back to the auto config.

farmio, 2020-06-02:
The xknx.yaml is parsed from xknx directly. The connection details are not sent back to HA, so they won't be in `config[DOMAIN]`.
It seems that the integration is overwriting the parsed connection_config. Can you try to set https://github.com/XKNX/xknx/blob/a6d0e551168884a822826d7b7f33bf9e07497817/home-assistant-plugin/custom_components/xknx/__init__.py#L206 to `return None`? It should work then.

You can also omit the local_ip. Xknx should be able to figure out itself.

Gateway not found could hint to a networking problem. Xknx seems to be unable to communicate via multicast to your interface. If you turn on debug logging you should be able to see it searching for gateways (multicast) - and your MDT interface should answer in unicast.

Tommatheussen, 2020-06-03:
Yep indeed, setting it to None works. As workaround I have now also defined the connection details in HA itself, that way they are actually picked up correctly.

Perhaps a check should be built in to check if `xknx.yaml` file is set, if so then the component should not take any connection settings from HA itself?

I'll have to do some more testing with XKNX being able to figure out the local_ip itself, it had some issues where the connection couldn't be made until I added it. Might be due to my network setup, I've got multiple VLANs and HA and KNX are both in different ones. HA also runs in proxmox with a shared NIC and then docker adds it's own IP addresses as well :)

farmio, 2020-06-03:
It is not said that a connection is configured in xknx.yaml even if the path is set. It can also just hold device configuration. 

Have you still not configured devices? If docker adds IP address you seem to not run it in networking mode "host". This will probably not work. The interface needs to have a direct path (no NAT) to the xknx instance (it gets its return address not from the IP layer but from the Knx connection header - and xknx doesn't know about your NATing).

PR #300 (2020-08-02): HA: use connection configuration from xknx.yaml if given
or fall back to default.
Previously xknx.yaml was overridden by the default ConnectionConfig passed in to xknx.start()

closes #297
closed 2020-08-18 by farmio