XKNX/xknx #301: Percentage-Sensor not read correctly

georgbachmann, 2020-06-05:
**Description of problem:**

I have a light-switch from Lingg&Janke that also has a temperature and a humidity sensor build in. I expose those values on my KNX bus and would like to monitor those in home-assistant. So I added them in my xknx.yaml file like so:

```
groups:
    sensor:
        humidity.knx.buero: {group_address_state: '6/1/0', value_type: 'percent'}
```
When checking the values in Home-Assistant, I only saw NaN values. So I double checked using the ETS Group-Monitor and saw correct value there:

[image]

But I noticed that the value is not between 0 and 1, but rather between 0 an 100. So I trial-and-errored a bit and found out, that when changing my sensors type to `temperature` it works. Then it reads the value correctly. I then did a bit of `customize.yaml` magic and not it's working fine.
Like so:
```
xknx.yaml

groups:
    sensor:
        humidity.knx.buero: {group_address_state: '6/1/0', value_type: 'temperature'}

customize.yaml:

sensor.humidity_knx_buero:
  friendly_name: Luftfeuchtigkeit Büro
  device_class: humidity
  unit_of_measurement: '%'
```

Still I guess?!? (not sure) this is a bug in xknx?
I am totally new to xknx and knx in general... so not sure... I can imagine that it's cause the value is not 0...1 but rather 0...100?!? But this is just a wild guess...

- [ ] using xknx standalone
- [x] using Home-Assistant knx plugin

**Version information:**
- xknx / Home-Assistant release with the issue:
0.110.4

**Problem-relevant `xknx.yaml` or `configuration.yaml` entries (fill out even if it seems unimportant):**
Posted above

farmio, 2020-06-05:
You tried to decode a 2byte value with a 1byte decoder. Check the DPT number of your sensor. 9.*** is a 2byte float so you have to use a suitable value_type in your configuration. See here for a complete list of implemented value_types for sensors: https://www.home-assistant.io/integrations/sensor.knx

closed 2020-06-05 by farmio

georgbachmann, 2020-06-08:
Awesome... I was just looking at https://xknx.io/sensor and this part:

> value_type controls how the value should be rendered in a human readable representation. The attribut may have may have the values percent, temperature, illuminance, speed_ms or current.

Will try out other values soon! Thanks!