XKNX/xknx #307: Gira climate control => single 1 bit for heating/cooling (no DPT 20.105)

spacegaier, 2020-06-14:
**Description of problem:**
My climate room controllers (product details below) are not completely following the KNX standard. They correctly use DPT 20.102 for the operating modes which works fine in HA. 
However, the element does not support proper controller modes (no DPT 20.105) to e.g. switch between heating and cooling. Instead there is a single 1 bit to switch between heating and cooling mode (see function object 89 on manual page 56).
Reading the current value is even more tricky, since it is a single bit inside a 2 byte DPT 22.101 that needs to be read (see manual page 157).

=> Is there any way to get that to work currently? If not, what would be the best option to try to add support for that into the library? Sub parameters for `controller_status_address` and `controller_status_state_address` to control what DPT to use?

- [ ] using xknx standalone
- [X] using Home-Assistant knx plugin

**Version information:**
- xknx / Home-Assistant release with the issue: 0.11.3 / 0.112.0.dev0
- last working xknx / Home-Assistant release (if known):

**KNX installation:**
I have multiple installed Gira Push Button Sensor 3 Plus 2-gang (DE: KNX Tastsensor 3 Plus 2fach) = Order Number: 5142 00
They are a combined climate control and button/switch unit.

English manual: https://download.gira.de/data3/51423490.pdf

**Problem-relevant `xknx.yaml` or `configuration.yaml` entries:**
```YAML
climate:
  - platform: knx
    name: KNX Office
    temperature_address: '3/1/4'
    target_temperature_address: '3/2/4'
    target_temperature_state_address: '3/6/4'
    operation_mode_address: '3/4/2'
    operation_mode_state_address: '3/4/3'
    # controller_status_address: '3/4/0'
    # controller_status_state_address: '3/4/1'
    min_temp: 7.0
    max_temp: 32.0```

3/4/0 and 3/4/1 are the GAs for the 1 bit mode switch and the 2 byte status reading.

farmio, 2020-06-14:
Can you read the heating/cooling status from reading the 1bit Object 89 when you set a Read-flag on the communication object? This would be easier than parsing the 2byte value. 
You could then configure a switch for this with a state_address.

spacegaier, 2020-06-15:
Yes, I can easily get a 1 bit status value, because in the end the decision whether to cool or heat is currently not made on a room level, but globally for the whole apartment anyway. My initial approach was nonetheless to get the status value from the individual room controllers, but that is overkill.

```YAML
switch:
  - platform: knx
    name: office mode
    address: '3/4/0'
    state_address: '3/4/1'
```

I just tested a KNX switch entity and that works, but it of course does not interact with the Lovelace climate card, since that requires the correct internal entity state to be set. Not really sure how to close that gap. High-level logic would need to be:

- Read status value from 3/4/1
- If true/on: Set the climate entity `state` attribute to "heat" 
- If false/off: Set to "cool"

Any ideas how to solve that? Is that were HA templates come into play?

spacegaier, 2020-06-15:
I also thought about triggering the state of all climate entities via my button by calling a service. However, with that I see two issues:
1. I would need to incorporate the button state into the service call and I don't think that is possible.
2. There is a bug in the xknx handling of the  `climate.set_hvac_mode` service in general (see separate issue).

PR #313 (2020-07-26): add support for binary heating cooling mode
This adds support for binary heating cooling mode

Heating mode defaults to DPT1 value 1
Cooling mode defaults to DPT1 value 0
as used by Gira, Jung and MDT actuators.

closes #307
closed 2020-08-18 by farmio