XKNX/xknx #308: Service calls fail (climate.set_hvac_mode & climate.set_preset_mode)

spacegaier, 2020-06-15:
**Description of problem:**

Runtime error occurs when attempting to call the service `climate.set_hvac_mode` from the HA developer tools.

[image]

The service calls goes through here:
https://github.com/XKNX/xknx/blob/c22e394ca1c7ade54b2cffe8bcedfd782f47bc75/home-assistant-plugin/custom_components/xknx/climate.py#L314

The problem is that `set_operation_mode(self, operation_mode)` is used to update both the hvac operation mode (xknx variable `controller_status_*`) as well as the preset mode (`operation_mode_*`).  

https://github.com/XKNX/xknx/blob/c22e394ca1c7ade54b2cffe8bcedfd782f47bc75/xknx/devices/climate_mode.py#L143

But that is incorrect, because now the method attempts to use my HVAC mode "cool" as an preset, which of course fails.

=> I think the cleanest solution would be to split  `set_operation_mode(self, operation_mode)` into two dedicated methods, one for HVAC mode and one for the presets. 

**BTW**: Call of service `climate.set_preset_mode` works, but only because the corresponding `controller_status_*` parameters are commented out. Otherwise I am quite certain it would fail when attempting to update those values. 

- [ ] using xknx standalone
- [X] using Home-Assistant knx plugin

**Version information:**
- xknx / Home-Assistant release with the issue: 0.11.3 / 0.112.0.dev0
- last working xknx / Home-Assistant release (if known):

**Problem-relevant `xknx.yaml` or `configuration.yaml` entries:**
```YAML
climate:
  - platform: knx
    name: KNX Office
    temperature_address: '3/1/4'
    target_temperature_address: '3/2/4'
    target_temperature_state_address: '3/6/4'
    operation_mode_address: '3/4/2'
    operation_mode_state_address: '3/4/3'
    # controller_status_address: '3/4/0'
    # controller_status_state_address: '3/4/1'
    min_temp: 7.0
    max_temp: 32.0
    operation_modes:
      - "Off"
      - "Heat"
      - "Cool"
      - "Comfort"
      - "Standby"
      - "Night"
      - "Frost Protection"
```

**Traceback (if applicable):**
```Traceback (most recent call last):
  File "/home/pi/git/core/homeassistant/core.py", line 1277, in catch_exceptions
    await coro_or_task
  File "/home/pi/git/core/homeassistant/core.py", line 1296, in _execute_service
    await handler.func(service_call)
  File "/home/pi/git/core/homeassistant/helpers/entity_component.py", line 209, in handle_service
    self._platforms.values(), func, call, required_features
  File "/home/pi/git/core/homeassistant/helpers/service.py", line 454, in entity_service_call
    future.result()  # pop exception if have
  File "/home/pi/git/core/homeassistant/helpers/entity.py", line 605, in async_request_call
    await coro
  File "/home/pi/git/core/homeassistant/helpers/service.py", line 485, in _handle_entity_call
    await result
  File "/home/pi/git/core/homeassistant/components/knx/climate.py", line 314, in async_set_hvac_mode
    await self.device.mode.set_operation_mode(knx_operation_mode)
  File "/home/pi/git/core/venv/lib/python3.7/site-packages/xknx/devices/climate_mode.py", line 150, in set_operation_mode
    DPTArray(DPTHVACMode.to_knx(operation_mode)))
  File "/home/pi/git/core/venv/lib/python3.7/site-packages/xknx/dpt/dpt_hvac_mode.py", line 69, in to_knx
    raise ConversionError("Could not parse HVACOperationMode", value=value)
xknx.exceptions.exception.ConversionError: <ConversionError description="Could not parse HVACOperationMode" value="HVACOperationMode.COOL"/>`

spacegaier, 2020-06-15:
Let me know if someone from the XKNX team wants to tackle that, otherwise I might provide a PR.

farmio, 2020-06-16:
Could you test if #305 fixes this issue? I'd think it would.

spacegaier, 2020-06-17:
Yes, seems to fix the runtime error.

closed 2020-06-17 by spacegaier