XKNX/xknx #310: Cover position query causes Error while syncing device: <CouldNotParseTelegram description="payload invalid"

tobiasneumann42, 2020-06-21:
**Description of problem:**
using somfy KNX 510i current position of cover returns error. ETS shows status group address as fully open with value 100,32 DPT Type 2 byte.
Home assistant reports error 2020-06-21 12:24:44 ERROR (MainThread) [xknx.log] Error while syncing device: <CouldNotParseTelegram description="payload invalid" device_name="OG_shutter_hall" group_address="3/0/52" payload="<DPTArray value="[0x18,0x0]" />"/>

somfy 510i reports, 100 as completely closed and 0 as open bellow configuration for cover including the invert_position parameter set to true. 

cover:
  - platform: knx
    name: "OG_shutter_hall"
    move_long_address: '3/0/22'
    move_short_address: '3/0/23'
    position_address: '3/0/24'
    position_state_address: '3/0/52'
    invert_position: true

- [ ] using xknx standalone
- [ x] using Home-Assistant knx plugin

**Version information:**
- xknx / Home-Assistant release with the issue: Home Assistant 0.111.3
- last working xknx / Home-Assistant release (if known):

**KNX installation:**
 
[image]

**Problem-relevant `xknx.yaml` or `configuration.yaml` entries (fill out even if it seems unimportant):**

**Traceback (if applicable):**

tobiasneumann42, 2020-06-21:
[image]

tobiasneumann42, 2020-06-21:
`cover:
  - platform: knx
    name: "OG_shutter_hall"
    move_long_address: '3/0/22'
    move_short_address: '3/0/23'
    position_address: '3/0/24'
    position_state_address: '3/0/52'
    invert_position: true`

farmio, 2020-06-21:
Postition and it's State is expected to be DPT5. Maybe you can set up your actuator to report in this format.

tobiasneumann42, 2020-06-21:
don't think so (unless I am missing something obvious), the actuator doesn't allow to specify the data type... 

states Datatype: not defined and there is no option where I can set anything there. 

[image]

so position should always be DPT5 independent of the actuator manufacturer ? Was wondering if there would be a way to adjust the cover code to deal with different formats, in case this is not a generally accepted standard for all blinds/covers.

farmio, 2020-06-21:
At least I have never seen this as 2byte Float. What can you select at "Datentyp"? Nothing?

You can for sure adjust the code to deal with this format. Here is a starting point: https://github.com/XKNX/xknx/blob/4a80c75b2448bca0ee98bfb6845fdc00f8ef6e80/xknx/devices/cover.py#L62
You can exchange RemoteValueScaling with eg. RemoteValueSensor

spacegaier, 2020-06-22:
> 
> 
> At least I have never seen this as 2byte Float. What can you select at "Datentyp"? Nothing?

I just checked mine (brand "Warema") and can confirm they are also 1 Byte = `DPT 5.*`. The same holds true for all actuators that a quick search revealed online.

farmio, 2020-06-22:
Also the Common visualizations (Gira G1 etc) expect DPT5. Maybe talk to the manufacturer of your actor about it. 
As a workaround you can use a simple logic to convert 2byte float to DPT5. This is also possible in HA. Define a sensor for DPT9 and expose its value as "percent" to a different GA. Use this GA for the Cover.

closed 2021-03-14 by farmio