XKNX/xknx #314: Rework is_open for covers

crazyfx1, 2020-07-23:
**Description of problem:**
I already created an issue in the HA repository (https://github.com/home-assistant/core/issues/34330), but I realized the problem lies in XKNX. 

Problem shortcut: My garage door can be controlled with knx. It just supports up/down and **cannot** report the current position of the garage door. In the current implementation, I cannot open / close the garage door via HA, because the reported state is not correct in 50% of the time. Same problem occurs with central covers like in my HA issue.
[image]

Even for covers which can report the current position, xknx uses the time in the travelcalculator and not the actual position of the cover.

With boils down to:
**The cover is_open and is_closed is not correct for knx covers with and without position state address.**

My proposal:
- knx cover **with** `group_address_position_state`: Use the actual position comparison with 100% or 0% to determine is_open and is_closed

- knx cover **without** `group_address_position_state`: Return `None` for both is_open and is_closed. The travelcalculator is a nice effort, but it doesn't help if the position was changed outside of XKNX. Therefore I believe it can't be used to determine the state at all.

I can provide the implementation if it has a chance to get it pulled ;)

- [ ] using xknx standalone
- [x] using Home-Assistant knx plugin

**Version information:**
- xknx / Home-Assistant release with the issue: 102

**KNX installation:**

**Problem-relevant `xknx.yaml` or `configuration.yaml` entries (fill out even if it seems unimportant):**

**Traceback (if applicable):**

farmio, 2020-07-23:
We could just start the travel calculator when a telegram on updown is received. In theory it should update the State properly then. 

For covers with position status I think it always updates the travelcalculator to the last received status from the bus.

crazyfx1, 2020-07-23:
You are right, the travelcalculator is updated correctly, my bad.

farmio, 2020-08-18:
Telegrams to move_long_address are now (0.12.0) processed by xknx, so cover without group_address_position_state should be in sync too (if the actuators and xknx movement times match).

PR #322 (2020-08-13): Cover update - up/down from bus; stop and movement direction
this is derived from #259 so 
- in xknx 0% is open and 100% is closed now
- adds is_opening and is_closing
- react to up/down from bus and start travel calculator
- add group_address_stop for actuators that have it (or don't support step)

resolves #314 
fixes #257 (I hope)
fixes #105 (without implementing the DPT1 state address - just listening for the normal group_address so initial status may be off)
closes #25
closed 2020-08-18 by farmio