XKNX/xknx #333: Using KNX Virtual implementation

maarthala, 2020-08-24:
**Description of problem:**
I am trying to learn KNX implementation using KNX virtual software provided knx.org. I am running  KNX virtual software on windows 10 VM and my host is mac.

Using this library, i am able to discover the device but could not able to update Switch. It will be really great If anyone have experience in using KNX virtual and point me to the right direction.

Here is the code i am using

    xknx = XKNX(config='xknx.yaml')
    await xknx.start() 
    switch = Switch(xknx,
                name='ch1',
                group_address='1/1/1')
    xknx.devices.add(switch)
    await xknx.devices['ch1'].set_on()

- [ X] using xknx standalone

**KNX installation:**
KNX Virtual from https://www.knx.org/knx-en/for-professionals/get-started/knx-virtual/index.php
ETS : Group configured on 1/1/1 switch DPT 1001

**Problem-relevant `xknx.yaml` or `configuration.yaml` entries (fill out even if it seems unimportant):**
general:
    own_address: "1.1.255"
connection:
    routing:

farmio, 2020-08-24:
I have never tried knx virtual, but there has been Someone on the discord doing screencasts for it. Maybe it helps you. 

https://discord.com/channels/338619021215924227/338619021215924227/723819544250023958

farmio, 2020-09-11:
seems you have worked it out https://github.com/vapourismo/knx-go/issues/31

closed 2020-09-11 by farmio