XKNX/xknx #337: Add new Device to xknx.devices automatically

farmio, 2020-08-26:
Should we add a new Device to xknx.Devices on init? Currently newly created devices have to be added manually to receive incoming telegrams.
https://github.com/XKNX/xknx/blob/fb622f76f2e8b8de0454f94558b16a88c6655a33/xknx/core/telegram_queue.py#L144

```
xknx = XKNX()
sensor = Sensor(xknx, ...)
xknx.devices.add(sensor)
```
I'd like to remove this boilerplate and call `xknx.devices.add(self)` in `Device.__init__()` - and remove it in `__del__()`

marvin-w, 2020-08-26:
Hmm, we also use this boilerplate in the HA plugin and I honestly like it there, but I get your point.

farmio, 2020-08-26:
I think we would have to just drop this line then.
https://github.com/XKNX/xknx/blob/fb622f76f2e8b8de0454f94558b16a88c6655a33/home-assistant-plugin/custom_components/xknx/__init__.py#L159

marvin-w, 2020-08-26:
I don't think so. There is also one reference in the factory.py when creating the climate. So go ahead if you want :-)

PR #340 (2020-08-28): Adds the devices to the device list automatically after instantiation
Before this change all devices had to be manually added to the xknx.devices List.
This was cumbersome and not needed, thats why it was removed.

Additionally, the device class now has a unique_id property that is used within the HA
integration in order to avoid errors where multiple devices share the same name within
different platforms.

Fixes #339, fixes #337
closed 2020-08-30 by farmio