XKNX/xknx #339: 0.12.0 Duplicate names result in incorrect configuration processing

spacegaier, 2020-08-27:
**Description of problem:**
I adjusted the HA YAML configuration file to the new layout as part of the migration to XKNX 0.12.0. There however seems to be a bug with the configuration parsing. I have two entries with the same name "KNX Office" one being a light, one being a cover which is throwing off the logic since it is now attempting to interpret the objects as the wrong type (see log output below). I assume somewhere it is attempting to look up the content of the configuration file only via the name instead of platform and name which would then be unique.
Once I change one of the names to "KNX Office 2" everything works fine, so my defined values/parameters themselves are correct.

- [ ] using xknx standalone
- [X] using Home-Assistant knx plugin

**Version information:**
- xknx / Home-Assistant release with the issue: 0.12.0 / HA latest dev
- last working xknx / Home-Assistant release (if known):

**KNX installation:**

**Problem-relevant `xknx.yaml` or `configuration.yaml` entries (fill out even if it seems unimportant):**
```YAML
knx:
  tunneling:
    host: "192.168.0.222"
  cover:
    - name: KNX Office
      [...]
  light:
    - name: KNX Office
      address: "1/1/29"
      state_address: "1/4/28"
      brightness_address: "1/5/18"
      brightness_state_address: "1/3/15"
```

**Traceback (if applicable):**
```
2020-08-27 11:35:36 ERROR (MainThread) [homeassistant.components.light] Error while setting up knx platform for light
Traceback (most recent call last):
  File "/home/pi/git/core/homeassistant/helpers/entity_platform.py", line 193, in _async_setup_platform
    await asyncio.shield(task)
  File "/home/pi/git/core/homeassistant/components/knx/light.py", line 28, in async_setup_platform
    async_add_entities_discovery(hass, discovery_info, async_add_entities)
  File "/home/pi/git/core/homeassistant/components/knx/light.py", line 37, in async_add_entities_discovery
    entities.append(KNXLight(device))
  File "/home/pi/git/core/homeassistant/components/knx/light.py", line 48, in __init__
    self._min_kelvin = device.min_kelvin
AttributeError: 'Cover' object has no attribute 'min_kelvin'
```

marvin-w, 2020-08-27:
This error occurs because we resolve the devices by device name. Since the refactoring we solely rely upon discovery. 

This error also exists for xknx.yaml configurations.

spacegaier, 2020-08-29:
This fix solves the issue for me. Both entities with the same name are now created and can be used in HA.
[image]

marvin-w, 2020-08-29:
Thanks for retesting and the PR!

PR #340 (2020-08-28): Adds the devices to the device list automatically after instantiation
Before this change all devices had to be manually added to the xknx.devices List.
This was cumbersome and not needed, thats why it was removed.

Additionally, the device class now has a unique_id property that is used within the HA
integration in order to avoid errors where multiple devices share the same name within
different platforms.

Fixes #339, fixes #337
closed 2020-08-29 by marvin-w