XKNX/xknx #345: TypeError in address.py

marvin-w, 2020-08-29:
**Description of problem:**

- [ ] using xknx standalone
- [x] using Home-Assistant knx plugin

**Version information:**
- xknx / Home-Assistant release with the issue: 0.11.0
- last working xknx / Home-Assistant release (if known): -

**KNX installation:**

**Problem-relevant `xknx.yaml` or `configuration.yaml` entries (fill out even if it seems unimportant):**

Will add later..

**Traceback (if applicable):**

```
2020-08-29 12:53:58 WARNING (MainThread) [xknx.log] Error: KNX bus did not respond in time to GroupValueRead request for: 5/0/2
2020-08-29 12:53:58 WARNING (MainThread) [xknx.log] Could not sync group address '5/0/2' from <Light name="wohnzimmer.schraege" switch="GroupAddress("1/0/22")/GroupAddress("5/0/2")/None/None" brightness="GroupAddress("1/0/40")/None/None/None" />
2020-08-29 12:53:59 WARNING (MainThread) [xknx.log] Error: KNX bus did not respond in time to GroupValueRead request for: 5/0/2
2020-08-29 12:53:59 WARNING (MainThread) [xknx.log] Could not sync group address '5/0/2' from <Light name="wohnzimmer.leiste" switch="GroupAddress("1/0/20")/GroupAddress("5/0/2")/None/None" brightness="GroupAddress("1/0/21")/None/None/None" />
2020-08-29 12:54:00 WARNING (MainThread) [xknx.log] Error: KNX bus did not respond in time to GroupValueRead request for: 5/0/2
2020-08-29 12:54:00 WARNING (MainThread) [xknx.log] Could not sync group address '5/0/2' from <Light name="wohnzimmer.sternenhimmel" switch="GroupAddress("1/0/37")/GroupAddress("5/0/2")/<DPTBinary value="0" />/False" brightness="GroupAddress("1/0/39")/None/<DPTArray value="[0x0]" />/0" />
2020-08-29 12:54:01 WARNING (MainThread) [xknx.log] Error: KNX bus did not respond in time to GroupValueRead request for: 5/0/2
2020-08-29 12:54:01 WARNING (MainThread) [xknx.log] Could not sync group address '5/0/2' from <Light name="esszimmer" switch="GroupAddress("1/0/24")/GroupAddress("5/0/2")/None/None" brightness="GroupAddress("1/0/43")/None/None/None" />
2020-08-29 12:54:02 WARNING (MainThread) [xknx.log] Error: KNX bus did not respond in time to GroupValueRead request for: 5/0/2
2020-08-29 12:54:02 WARNING (MainThread) [xknx.log] Could not sync group address '5/0/2' from <Light name="wc" switch="GroupAddress("1/0/31")/GroupAddress("5/0/2")/<DPTBinary value="0" />/False" />
2020-08-29 12:54:03 WARNING (MainThread) [xknx.log] Error: KNX bus did not respond in time to GroupValueRead request for: 5/0/2
2020-08-29 12:54:03 WARNING (MainThread) [xknx.log] Could not sync group address '5/0/2' from <Light name="buero1" switch="GroupAddress("1/0/32")/GroupAddress("5/0/2")/<DPTBinary value="0" />/False" />
2020-08-29 12:54:04 WARNING (MainThread) [xknx.log] Error: KNX bus did not respond in time to GroupValueRead request for: 5/0/2
2020-08-29 12:54:04 WARNING (MainThread) [xknx.log] Could not sync group address '5/0/2' from <Light name="buero2" switch="GroupAddress("1/0/34")/GroupAddress("5/0/2")/<DPTBinary value="0" />/False" />
2020-08-29 12:54:05 WARNING (MainThread) [xknx.log] Error: KNX bus did not respond in time to GroupValueRead request for: 5/0/2
2020-08-29 12:54:05 WARNING (MainThread) [xknx.log] Could not sync group address '5/0/2' from <Light name="buero3" switch="GroupAddress("1/0/35")/GroupAddress("5/0/2")/<DPTBinary value="0" />/False" />
2020-08-29 12:54:06 WARNING (MainThread) [xknx.log] Error: KNX bus did not respond in time to GroupValueRead request for: 5/0/2
2020-08-29 12:54:06 WARNING (MainThread) [xknx.log] Could not sync group address '5/0/2' from <Light name="abstellraum" switch="GroupAddress("1/0/36")/GroupAddress("5/0/2")/<DPTBinary value="0" />/False" />
2020-08-29 12:54:07 WARNING (MainThread) [xknx.log] Error: KNX bus did not respond in time to GroupValueRead request for: 5/0/2
2020-08-29 12:54:07 WARNING (MainThread) [xknx.log] Could not sync group address '5/0/2' from <Switch name="Kueche.Abzug" switch="GroupAddress("1/0/18")/GroupAddress("5/0/2")/<DPTBinary value="0" />/False" />
2020-08-29 12:54:08 WARNING (MainThread) [xknx.log] Error: KNX bus did not respond in time to GroupValueRead request for: 5/0/2
2020-08-29 12:54:08 WARNING (MainThread) [xknx.log] Could not sync group address '5/0/2' from <Switch name="Wohnzimmer.Stereoanlage" switch="GroupAddress("1/0/28")/GroupAddress("5/0/2")/None/None" />
2020-08-29 12:54:09 WARNING (MainThread) [xknx.log] Error: KNX bus did not respond in time to GroupValueRead request for: 5/0/2
2020-08-29 12:54:09 WARNING (MainThread) [xknx.log] Could not sync group address '5/0/2' from <Switch name="Schlafzimmer.Steckdose.Marvin" switch="GroupAddress("1/0/3")/GroupAddress("5/0/2")/None/None" />
2020-08-29 12:54:10 WARNING (MainThread) [xknx.log] Error: KNX bus did not respond in time to GroupValueRead request for: 5/0/2
2020-08-29 12:54:10 WARNING (MainThread) [xknx.log] Could not sync group address '5/0/2' from <Switch name="Schlafzimmer.Wand.Strom" switch="GroupAddress("1/0/5")/GroupAddress("5/0/2")/None/None" />
2020-08-29 12:54:11 WARNING (MainThread) [xknx.log] Error: KNX bus did not respond in time to GroupValueRead request for: 5/0/2
2020-08-29 12:54:11 WARNING (MainThread) [xknx.log] Could not sync group address '5/0/2' from <Switch name="Balkon2.Steckdose" switch="GroupAddress("1/0/30")/GroupAddress("5/0/2")/<DPTBinary value="0" />/False" />
2020-08-29 13:32:27 WARNING (MainThread) [xknx.log] Ignoring not implemented CEMI: <UnsupportedCEMIMessage description="CEMI too small. Length: 10; CEMI: b')\x00\xb0`\x10\x04\x11m\x00\x80'" />
2020-08-29 13:32:27 WARNING (MainThread) [xknx.log] Ignoring not implemented CEMI: <UnsupportedCEMIMessage description="APCI not supported: 0b0100000000" />
2020-08-29 13:32:27 WARNING (MainThread) [xknx.log] Ignoring not implemented CEMI: <UnsupportedCEMIMessage description="APCI not supported: 0b100001100000000" />
2020-08-29 13:32:27 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/xknx/core/telegram_queue.py", line 68, in run
    await self.process_telegram(telegram)
  File "/usr/local/lib/python3.8/site-packages/xknx/core/telegram_queue.py", line 96, in process_telegram
    await self.process_telegram_incoming(telegram)
  File "/usr/local/lib/python3.8/site-packages/xknx/core/telegram_queue.py", line 119, in process_telegram_incoming
    for device in self.xknx.devices.devices_by_group_address(
  File "/usr/local/lib/python3.8/site-packages/xknx/devices/devices.py", line 32, in devices_by_group_address
    if device.has_group_address(group_address):
  File "/usr/local/lib/python3.8/site-packages/xknx/devices/scene.py", line 42, in has_group_address
    return self.scene_value.has_group_address(group_address)
  File "/usr/local/lib/python3.8/site-packages/xknx/remote_value/remote_value.py", line 56, in has_group_address
    return group_address in [self.group_address, self.group_address_state]
  File "/usr/local/lib/python3.8/site-packages/xknx/telegram/address.py", line 62, in __eq__
    raise TypeError()
TypeError
2020-08-29 13:32:27 WARNING (MainThread) [xknx.log] Ignoring not implemented CEMI: <UnsupportedCEMIMessage description="CEMI too small. Length: 10; CEMI: b')\x00\xb0`\x10\x04\x11m\x00\x81'" />
2020-08-29 13:32:30 WARNING (MainThread) [xknx.log] Ignoring not implemented CEMI: <UnsupportedCEMIMessage description="APCI not supported: 0b0100000000" />
2020-08-29 13:32:33 WARNING (MainThread) [xknx.log] Ignoring not implemented CEMI: <UnsupportedCEMIMessage description="APCI not supported: 0b0100000000" />
2020-08-29 13:38:56 WARNING (MainThread) [xknx.log] Ignoring not implemented CEMI: <UnsupportedCEMIMessage description="CEMI too small. Length: 10; CEMI: b')\x00\xb0`\x10\x04\x11)\x00\x80'" />
[... 122 lines cut ...]
2020-08-29 13:50:38 WARNING (MainThread) [xknx.log] Ignoring not implemented CEMI: <UnsupportedCEMIMessage description="APCI not supported: 0b111101000000000" />
2020-08-29 13:50:38 WARNING (MainThread) [xknx.log] Ignoring not implemented CEMI: <UnsupportedCEMIMessage description="CEMI too small. Length: 10; CEMI: b')\x00\xb0`\x10\x04\x11m\x00\xe2'" />
2020-08-29 13:50:38 WARNING (MainThread) [xknx.log] Ignoring not implemented CEMI: <UnsupportedCEMIMessage description="APCI not supported: 0b111111000000000" />
2020-08-29 13:50:38 WARNING (MainThread) [xknx.log] Ignoring not implemented CEMI: <UnsupportedCEMIMessage description="CEMI too small. Length: 10; CEMI: b')\x00\xb0`\x10\x04\x11m\x00\xe6'" />
[... cut ...]
2020-08-29 13:50:40 WARNING (MainThread) [xknx.log] Ignoring not implemented CEMI: <UnsupportedCEMIMessage description="CEMI too small. Length: 10; CEMI: b')\x00\xb0`\x10\x04\x11m\x00\xc2'" />
2020-08-29 13:50:40 WARNING (MainThread) [xknx.log] Ignoring not implemented CEMI: <UnsupportedCEMIMessage description="APCI not supported: 0b110101110000000" />
2020-08-29 13:50:40 WARNING (MainThread) [xknx.log] Ignoring not implemented CEMI: <UnsupportedCEMIMessage description="CEMI too small. Length: 10; CEMI: b')\x00\xb0`\x10\x04\x11m\x00\x81'" />
2020-08-29 13:54:13 WARNING (MainThread) [xknx.log] Error: KNX bus did not respond in time to GroupValueRead request for: 3/1/5
```

farmio, 2020-08-29:
Where are the unsupported Cemi Frames coming from? Have you been programming a Device?

marvin-w, 2020-08-29:
I did, yes. I had to restart my HA as nothing worked anymore.
I dropped this here so that the logs are not gone.

farmio, 2020-08-29:
Good. Then this will be easy to fix, I think. 
On the other hand it seems that our reconnect logic is not reliable .

marvin-w, 2020-08-29:
No it never worked for me. I'll also open a new feature request for a potential improvement

closed 2020-08-29 by marvin-w

farmio, 2020-08-29:
Did this issue just happen pre 0.12 ?

marvin-w, 2020-08-29:
Yes! It works fine with 0.12.0 - I also get weird messages but there is no issue present.