XKNX/xknx #346: Feature request: Expose Sensor to HA that determines if XKNX is connected to the bus

marvin-w, 2020-08-29:
If possible, we could expose a sensor in HA that determines if xknx.connected = True.

Our reconnect logic doesn't work really well and this would really help to "keep track" of when errors occur.

Potential error causes:

* Internet not working (router disconnected and internal network unreachable)
* reprogramming of a device
* ...

farmio, 2020-08-29:
I think HA would like us to raise homeassistant.exceptions.PlatformNotReady if we loose connection.

marvin-w, 2020-08-29:
Yes, correct. It also doesn't anymore on my setup. Weird.

farmio, 2021-04-14:
I think our reconnection logic is more or less stable now.
Except if IP addresses changed - this would be handled by last point of #349 

Closing this in favor for #670 - setting `available` to False should be the preferred way over a dedicated binary sensor.

closed 2021-04-14 by farmio