XKNX/xknx #376: Add brightness state north for weather entity

DirkMu, 2020-09-05:
As mentioned on [discord](https://discord.com/channels/338619021215924227/338619021215924227/751743171708452865) would like to request the enhancement of the weather entity by a brightness state for north. 

Currently, it support south, east and west:
group_address_temperature: "7/0/0",
group_address_brightness_south: "7/0/1",
group_address_brightness_west: "7/0/2",
group_address_brightness_east: "7/0/3",
group_address_wind_speed: "7/0/4",
group_address_rain_alarm: "7/0/5",
group_address_frost_alarm: "7/0/6",
group_address_wind_alarm: "7/0/7",
group_address_day_night: "7/0/8",
group_address_air_pressure: "7/0/9",
group_address_humidity: "7/0/10",
expose_sensors: True,
sync_state: True,

With regards to sync_state, does it support the new state updater of farmio?

farmio, 2020-09-05:
> does it support the new sync state

Yes, every device does, as it hooks one level lower. The weather device even has an attribute for options "sync_state".

DirkMu, 2020-09-05:
> > does it support the new sync state
> 
> Yes, every device does, as it hooks one level lower. The weather device even has an attribute for options "sync_state".

Great. Thanks. Yes, I saw the attribute but I was not sure whether it support true or false only or the new modes also.

PR #383 (2020-09-13): Implement brightness_north attribute to weather device.
Merges upstream changes in weather device. (name attribute)
Adds all non HA supported attributes as device specific state attributes.

State attributes are now available in HA: 
[image]

Closes #376
closed 2020-09-13 by marvin-w