XKNX/xknx #379: Future of xknx and the custom home assistant component

marvin-w, 2020-09-06:
I'd like to discuss the future of the xknx library in this ticket.

I feel like we should talk about the next steps concerning the library and the integration and also talk about some of the pitfalls that we usually encounter during development. (imho)

# Pitfalls and potential solutions

* When adding additional configurable features it is a burden to maintain them twice - once for the `xknx.yaml` and once for the custom HA component.

Sol.: Since HA is effectively going away from the YAML config we could think about removing _***all***_  platform YAML configurations and completely moving to `xknx.yaml`.

* User experience is not that great when using the knx integration inside HA

Sol.: HA is supporting config flows (see #238). We could make use of them and get rid of all the YAML configuration inside the HA component.

# Next steps

This list is sorted in chronological order but is not (yet?) complete and will probably change in the future.

1. Implement a `version` attribute in `xknx.yaml` and set the default version to 1. We can then start implementing a working draft for a future version of the configuration (version 2) that will then also replace the Home Assistant yaml configuration.

**Why**: The current structure of `xknx.yaml` is not straight forward to understand and doesn't look good. We should align it to HA standards.

2. Start implementing a working draft for version 2 of `xknx.yaml`. @farmio already had a couple of ideas how we can improve the current configuration style. The new configuration layout should add schema validation with voloptous (I don't know how to write that awkward word jesus) (like HA does it).

Potential new layout:

```
light:
    - name: "MyLight"
      on_off:
        address: "1/2/3"
        state_address: "1/2/4"
        state_update: "every 90" # for setting state_updater
      brightness:
        address: "1/3/3"
        state_address: "1/3/4"
        state_readable: True # shall answer to GroupValueRead - mutually exclusive with state_update
        passive_state: ["3/4/5", "3/4/9"]
```

**Why**: We should use versioning here in order to not conflict with current user configurations. Also, by doing this we could effectively use smaller pull requests which are easier to review. The new layout can also cover new features like **passive state addresses** (for listening GAs - used for i.e. central functions) and also for globally using and configuring the new **state updater**.

3. (Optional) Implement a migration tool for users to use in order to convert there `xknx.yaml` into the new version.

Why: Users will probably be annoyed by always moving configs back and forth. We should assist them here. We could even think of integrating this directly in xknx and just do the necessary conversion _on the fly_  when we detect that a user still uses an old version. And then after some time we can drop the old config completely.

4. (Optional) Implement a migration tool for users that use the HA YAML configs.

Why: They need to be migrated to `xknx.yaml` version 2.

5. Drop YAML Support for Home Assistant component.

Why: Users can use `xknx.yaml` then. We have less maintenance effort and we can use the discovery features from home assistant for it. 

6. Implement a config flow

Why: A config flow allows us to use the `GatewayScanner` to search for new devices. Users will then be able to configure the KNX integration in the frontend. Please note that they can **only** configure the connections parameters. Nothing else.

Todos:

- [ ] Check if this approach works with the available discovery options for HA

7. Drink some beer, this is gonna be a shit ton of work. :100:

farmio, 2020-09-13:
I think a more flexible configuration schema would be very nice. I'm also in favor of using the same schema for xknx.yaml and HA then.
If we could drop a website where users can past their config and it converts to new schema, this would be great.

I'd postpone dropping HA yaml support until config flow is ready. I really like that HA can be completely configured from within itself, without any external documentation, schemas, websites, programs etc. For this we'd need to enable device config from config flow too, which shoudn't be impossible. 
What's impossible imho is device discovery (apart from connection discovery with GatewayScanner - but auto mode should work for most installations anyway).

I'm also very much in favor with point 7 😉

marvin-w, 2020-09-14:
> I'd postpone dropping HA yaml support until config flow is ready. I really like that HA can be completely configured from within itself, without any external documentation, schemas, websites, [...]

The whole refactoring of the configs only has sense if we drop support for yaml within HA. If that's not the plan I can save a lot of time and just don't do that then. I know what you mean but maintaining both: xknx.yaml and HA the way we do right now is not a really great and maintainable way IMHO. And I'm pretty sure we can't just drop xknx.yaml as that wouldn't make any sense at all. So, should we just keep both or what is your suggestion here?

I'm also not sure how HA would react to moving to our config version.

ugleatle, 2020-09-18:
What does this mean in practice if you want to do your config outside the webinterface in HA? At first glance it seems like a poor idea to me to remove the YAML-functionality, but I might me misunderstanding the practical consequences.

marvin-w, 2020-09-19:
@ugleatle Well the main goal was to use just _one_ configuration which would eliminate the need to maintain everything twice (HA and xknx). I'll have to think about this more throughly and maybe create a PoC.

kalhimeo, 2020-10-14:
My 2cents is that it wouldn't help much to configure everything from the HA interface.
No discussion about the configuration stage, that would for sure bring immediate confidence to new users that the communication between HA and KNX is working, but for the GA configuration I don't think that someone who cannot edit a yaml file would know better about how his KNX installation is working regarding GAs, flags, etc. :p And for users with a lot of GA to configure being able to copy/paste in a text file is way easier than dealing with hundreds of web input fields.
Anyway, thank you so much for maintaining this KNX integration, it adds so much value to my home automation, you guys have all my support ! Keep up the excellent work 👍

da-anda, 2020-10-14:
@kalhimeo well, there is ETS for professionals and then there is ETS Inside. Inside users might not necessarily be too comfortable with messing around with yaml files and figuring out their GAs since IIRC the GAs are generated automatically in the background by ETS Inside (never used it myself though - only had opened an Iside project in my ETS5 and the GAs were a mess).

kalhimeo, 2020-10-14:
@da-anda I never used ETS Inside either so I cannot really say, but if those users don't have a clue about their GAs, as you say they are generated automatically in the background, they wouldn't be able to configure xknx which is based on the GA attribution anyway in it's actual form (whatever it is via yaml or via web forms within HA).
My understanding of an ETS5 project file is that it won't reflect the components as we want to see them in HA (lights, cover, etc), only group objects linked together via group addresses in a way which is so flexible and different for every manufacturer that it's nearly impossible to deduct what is really behind each address. Even if you would have something which works for 50% of the GA, it would still be much more complicated to cross-check everything to catch errors, than creating everything from scratch with my own logic IMO.
They made a small step in a more intuitive way of presenting things with the "functions" feature in ETS5, and I have read that they want to improve this "object" aspect even more with ETS 6 which is supposed to be more IoT oriented, we will have to see how they develop that, but I am afraid that we would be very depended of how the KNX association goes before being able to make something easy to work with out of the box in HA.
Anyway I am not saying that there cannot be any easier way to configure things, but if the goal is to do a web form with the same content as the yaml file, it wouldn't help IMO.

da-anda, 2020-10-14:
well, you have a topo inside the ETS5 project files you could use to automatically generate rooms and entities and assign GAs to them. That of course would only work if you correctly and thoroughly configured/built your topo inside ETS. From what I have seen of the Inside project I had imported, it also mainly uses this topo and is randomly creating GA groups etc per room/floor/without any clear logic. But what I heard from a friend (who uses Inside) you can also create custom GAs and ofc also see which GA is assigned, but I think there was no tree view and he didn't pay attention to GAs at all until I had pointed him to those (so users do not have to actively mess with them out of the box). If anybody is interested I can ask him for more details.

But I fully agree that no web from would simplify the setup process in any way. Only an import would simplify things, but even there you will have to tweak things since AFAIK you do not know which GA of an entity does what. You might be able to guess by it's DPT, but it will just be a guess and likely be wrong.

closed 2021-04-03 by marvin-w