XKNX/xknx #392: Create ETS compatible telegram logger

farmio, 2020-09-10:
It would be nice if we had a telegram logger that stores received telegrams in a format that can be imported by ETS (xml, I think). This would add a very valuable example to our collection and may even be a nice Project on its own.

marvin-w, 2020-09-10:
@farmio Would it make sense to implement this as an additional feature in the core, that can be turned on and off? I think this might be useful for people using home assistant that are trying to debug issues or just to have some kind of logging available. WDYT?

farmio, 2020-09-10:
I think this would be better off as a separate thing, so no new dependencies have to be added to xknx.

I would imagine this being used in an environment where no ETS and HA is available - eg. a clients installation where a raspberrypi with this script is dropped and picked up a week later.

HA already has several logging functions that can help for HA-specific problems, and when ETS is available it can log for debuging too.

But lets see...

closed 2022-02-23 by github-actions[bot]