XKNX/xknx #396: Missing DPT's for water meters

elupus, 2020-09-12:
**Description of problem:**

Seems we are missing DPT types for water meters.

- [ ] using xknx standalone
- [X] using Home-Assistant knx plugin

**Version information:**
- xknx / Home-Assistant release with the issue:
- last working xknx / Home-Assistant release (if known):

**KNX installation:**

**Problem-relevant `xknx.yaml` or `configuration.yaml` entries (fill out even if it seems unimportant):**

**Traceback (if applicable):**
````
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 213, in _async_setup_component
    result = await task
  File "/usr/src/homeassistant/homeassistant/components/knx/__init__.py", line 141, in async_setup
    create_knx_device(
  File "/usr/src/homeassistant/homeassistant/components/knx/factory.py", line 56, in create_knx_device
    return _create_sensor(knx_module, config)
  File "/usr/src/homeassistant/homeassistant/components/knx/factory.py", line 214, in _create_sensor
    return XknxSensor(
  File "/usr/local/lib/python3.8/site-packages/xknx/devices/sensor.py", line 28, in __init__
    self.sensor_value = RemoteValueSensor(
  File "/usr/local/lib/python3.8/site-packages/xknx/remote_value/remote_value_sensor.py", line 29, in __init__
    raise ConversionError("invalid value type", value_type=value_type, device_name=device_name)
xknx.exceptions.exception.ConversionError: <ConversionError description="invalid value type" device_name="Vattenmätare Räknare" value_type="12.1200"/>
````

farmio, 2020-09-12:
Hi!

I can't find "12.100" in the knx specifications. Do you have any sources of its definition?

You can always use generic "12" or "4byte_unsigned" for now.

marvin-w, 2020-09-13:
He asked for DPT 12.1200 which is in the specs as DPT_VolumeLiquid_Litre

Unit: Litre.

https://www.knx.org/wAssets/docs/downloads/Certification/Interworking-Datapoint-types/03_07_02-Datapoint-Types-v02.01.02-AS.pdf

Page 210.

I can take care of it.

farmio, 2020-09-13:
WTH 😱
Im still using the download version from my.knx.org - which seems to be not updated for literally ages! 
Never knew they published an online version too.

PR #397 (2020-09-15): Adds new sensor types DPT 12.1200 and DPT 12.1201
Closes #396
closed 2020-09-15 by marvin-w