XKNX/xknx #400: Adjust workflow to update version to <RELEASE_VERSION>

marvin-w, 2020-09-16:
I don't really want to bother adjusting the version manually before each release. Instead this should be commited automatically - the resulting tag should reside on that commit then.

farmio, 2020-09-16:
So the version number is generated automatically? Could that distinguish between major and minor version changes?

marvin-w, 2020-09-16:
No, we still set the version manually on release. But maybe there is some kind of interception to bump the version in our code before that.

farmio, 2020-09-16:
Ah. So you want to change the content of `__version__.py` on a release. 
Sounds nice!
We could then even append "beta" or anything after release so the version is marked as working branch version.

closed 2021-11-25 by github-actions[bot]