XKNX/xknx #402: Integration with HA is broken in HA 0.115

jncanches, 2020-09-18:
In 0.115, there are still problems for a climate supporting both operation_modes and on_off :

Setting Cool always returns Heating to HA :(

Another problem is that if you use an HA sensor for the climate temperature_address, the climate does not get the temperature.
Its very annoying because, my sensor are not KNX ones, but are computed by HA and exposed to KNX bus. It was working on previous version.

Property hvac_action in climate.py still not exists in climate.py

EDIT
I have made some debug
Here is what happened when I click on "Cool" in HA. Debug are in climate.py, the function is written on each line and start with TEST JN

`2020-09-18 03:42:58 WARNING (MainThread) [xknx.log] TEST JN async_set_hvac_mode hvac_mode='cool' 2020-09-18 03:42:58 WARNING (MainThread) [xknx.log] TEST JN hvac_mode self.device.mode.operation_mode='HVACOperationMode.COOL' and self.device.mode.operation_mode.value='Cool' 2020-09-18 03:42:58 WARNING (MainThread) [xknx.log] TEST JN hvac_mode self.device.mode.operation_mode='HVACOperationMode.COOL' and self.device.mode.operation_mode.value='Cool' 2020-09-18 03:42:58 WARNING (MainThread) [xknx.log] TEST JN hvac_mode self.device.mode.operation_mode='HVACOperationMode.COOL' and self.device.mode.operation_mode.value='Cool' 2020-09-18 03:42:58 WARNING (MainThread) [xknx.log] TEST JN hvac_mode self.device.mode.operation_mode='HVACOperationMode.COOL' and self.device.mode.operation_mode.value='Cool' 2020-09-18 03:42:58 WARNING (MainThread) [xknx.log] TEST JN hvac_mode self.device.mode.operation_mode='HVACOperationMode.COOL' and self.device.mode.operation_mode.value='Cool' 2020-09-18 03:42:58 WARNING (MainThread) [xknx.log] TEST JN hvac_mode self.device.mode.operation_mode='HVACOperationMode.COOL' and self.device.mode.operation_mode.value='Cool' 2020-09-18 03:43:21 WARNING (MainThread) [xknx.log] TEST JN hvac_mode self.device.mode.operation_mode='HVACOperationMode.STANDBY' and self.device.mode.operation_mode.value='Standby' 2020-09-18 03:43:21 WARNING (MainThread) [xknx.log] TEST JN hvac_mode self.device.mode.operation_mode='HVACOperationMode.STANDBY' and self.device.mode.operation_mode.value='Standby' 2020-09-18 03:43:21 WARNING (MainThread) [xknx.log] TEST JN hvac_mode self.device.mode.operation_mode='HVACOperationMode.STANDBY' and self.device.mode.operation_mode.value='Standby'`

As you can see the set if correct. The right mode is sent. The next "get mode" are okay, but some seconds later, the mode rollback to "StandBy" (instead of OFF). I don't really know what's happening...

- [ ] using xknx standalone
- [X] using Home-Assistant knx plugin

**Version information:**
- xknx / Home-Assistant release with the issue: 0.115
- last working xknx / Home-Assistant release (if known):

**Problem-relevant `xknx.yaml` or `configuration.yaml` entries (fill out even if it seems unimportant):**
My configuration

```
- name: day_pac
  temperature_address: '4/0/21'
  target_temperature_address: '4/0/18'
  target_temperature_state_address: '4/0/20'
  controller_mode_address: '4/0/2'
  controller_mode_state_address: '4/0/3'
  on_off_address: '4/0/0'
  on_off_state_address: '4/0/1'
  min_temp: 16.0
  max_temp: 30.0
```

farmio, 2020-09-18:
Thanks for your issue.

Processing of exposed values in other xknx devices broke by accident when fixing a different issue (you could say previously it worked by accident 🤣) see #394

jncanches, 2020-09-18:
> Thanks for your issue.
> 
> Processing of exposed values in other xknx devices broke by accident when fixing a different issue (#324 - you could say it previously worked by accident 🤣)
> see #394

Thanks for your answer. Is there a way where I can patch this in my existing HA install ? Since my HVAC does not know the current temperature, it will continue to cool (or heat) my house, and consume a large amount of energy :(

farmio, 2020-09-18:
It would be possible to use the xknx custom component and check out the PR locally (not in Hassio I think). You will need some Git and Python knowledge.
But if it is that critical for you I'd just downgrade to 0.114 for now.

marvin-w, 2020-09-18:
Do you use Docker for running HA? Or do you use Hass.io?

jncanches, 2020-09-18:
I have Hass.io installed on a Debian