XKNX/xknx #403: Problem with light color_temperature -> knx device uses wrong DPT

crazyfx1, 2020-09-18:
Hi,

I'm using the ISE smart connect KNX Hue (https://www.ise.de/de/produkte/ise_smart_connect_KNX_Hue) to control my Hue lamps in HA.

The problem is the KNX device uses DPT 9.002 (2 byte relative Kelvin) and not DPT 7.600
[image]

Unfortunately the datatypes are not compatible.

Is there a possibility to convert the types in HA or is there a code change in xknx needed?

farmio, 2020-09-18:
Have you talked to ISE about that? The initial Gira G1/X1 had this too and fixed it with a firmware update. It was a mixup of "Temperature (K)" and "Color Temperature (K)"

You could in the meantime try to use expose with the lights attribute as DPT 9. 
Would you need to read the value too?

I think you could also just use HAs Hue integration.

crazyfx1, 2020-09-19:
I will contact ISE next week but I doubt they will change the datatype. :) We'll see.

g0lumit0, 2021-11-04:
> @crazyfx1 Did you contact ISE? If yes, what was the outcome? We'd prefer not to have modes in the code that basically are workaround for bugs in some manufactures firmware.

Hello,

I have the same problem and I have contacted them, they will not update the module :(

Reply : When we developed the "ise smart connect KNX Hue" gateway, the data point type 7.600 did not yet exist. We therefore took the data point 9.xxx, as it was the closest match.
You are right, this makes little sense at the present time and you have to do a conversion with a home server. This cannot be changed either.  
There is no update in sight at the moment.

crazyfx1, 2021-11-04:
Yes I've contacted them a year ago. They said it's possible to update but they are no plans to do it.

closed 2022-02-09 by github-actions[bot]