XKNX/xknx #404: with significant_bit option of binary sensor removed, force operation of presence detector is not possible anymore

Caibuk, 2020-09-19:
**Description of problem:**
After updating HA to 0.115.1 and the KNX configuration accordingly, I cannot set the forced operation ("Zwangsführung") of my presence detectors (set them forcefully to "on" or "off" or to automatic detection ["normal mode"]).
This is due to the removal of the "significant_bit" option in #303 .
Forced operation is a 2 bit data point with the first bit setting the control mode (1 = forced operation, 0 = passive operation) and the second bit switching the light on (=1) or off (=0).

- 00: = 0, automatic mode without force operation
- 01: = 1, automatic mode without force operation
- 10: = 2, force operation light "off"
- 11: = 3, force operation light "on"

It would be great to get this back to work. Either with the return of "significant_bit" option or through a new non-binary "switch".

- [ ] using xknx standalone
- [x] using Home-Assistant knx plugin

**Version information:**
- xknx / Home-Assistant release with the issue: 0.115.1
- last working xknx / Home-Assistant release (if known): 0.114.4

**KNX installation:**

- presence detector EIBMARKT N000501 (data sheet only in German available: https://www.eibmarkt.com/Files/Datenblatt/uni_melder_N000501_DE.pdf, this one is similar: https://www.mdt.de/download/MDT_TM_Presence_Detector_1_fold_01.pdf)

**Problem-relevant `xknx.yaml` or `configuration.yaml` entries (fill out even if it seems unimportant):**
This is an example of the old working config:

getting actual mode of force operation with a binary_sensor:
```
- name: PM_OG_Bad_Bewegung
  state_address: '5/3/18'
  device_class: motion
- name: PM_OG_Bad_Zwang_1
  state_address: '5/3/16'
  significant_bit: 1
- name: PM_OG_Bad_Zwang_2
  state_address: '5/3/16'
  significant_bit: 2
```

setting force operation with a template switch:
```
- platform: template
  switches:
    pm_og_bad:
      value_template: "{{ is_state('binary_sensor.pm_og_bad_zwang_2', 'off') }}"
      turn_on:
        service: knx.send
        data:
          address: '5/3/16'
          payload: 0
      turn_off:
        service: knx.send
        data:
          address: '5/3/16'
          payload: 2
```

**Traceback (if applicable):**

marvin-w, 2020-09-19:
@farmio I think we should readd it. 

Originally, we were unable to define valid use cases for this option but since you have one we should add it back. Not sure yet in which way it will come back.

farmio, 2020-09-19:
@Caibuk the template switch does still work, does it? You could get the current mode using `knx_event` (see fire_event at https://www.home-assistant.io/integrations/knx/ ) and match the same payloads to update the state of your template switch.

I personally think that it was a hack. If we want this functionality we should implement DPT2 properly and find a way to use it in HA (which is the hard part, as HA doesn't have a concept of non-binary switches imho).

Caibuk, 2020-09-19:
Hi @farmio,
yes, you are right. I can still set the mode, but not update it. My issue was worded too unclear in that respect, sorry.
I read about the workaround with fire_events in the mean time and will give it a try.

I guess this issue can be closed.
Thank you very much for your hard work!

closed 2020-09-19 by Caibuk

farmio, 2020-09-19:
@Caibuk Please let us know if updating from knx_event works for your purpose. We could add it to documentation then.