XKNX/xknx #408: DPT 19.001 causes error / disconnect

josiasmontag, 2020-09-20:
**Description of problem:**

- [ ] using xknx standalone
- [x] using Home-Assistant knx plugin

I do see this error in my logfile exactly every 10 minutes:
```
2020-09-20 18:22:52 WARNING (MainThread) [xknx.log] CEMI not supported: <UnsupportedCEMIMessage description="APCI not supported: 0b1111000000" />
2020-09-20 18:22:52 ERROR (MainThread) [xknx.log] <CouldNotParseKNXIP description="KNXIP data has wrong length" />
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/xknx/io/udp_client.py", line 88, in data_received_callback
    knxipframe.from_knx(raw)
  File "/usr/local/lib/python3.8/site-packages/xknx/knxip/knxip.py", line 81, in from_knx
    raise CouldNotParseKNXIP("KNXIP data has wrong length")
xknx.exceptions.exception.CouldNotParseKNXIP: <CouldNotParseKNXIP description="KNXIP data has wrong length" />
2020-09-20 18:22:53 WARNING (MainThread) [xknx.log] CEMI not supported: <UnsupportedCEMIMessage description="APCI not supported: 0b1111000000" />
2020-09-20 18:22:53 ERROR (MainThread) [xknx.log] <CouldNotParseKNXIP description="KNXIP data has wrong length" />
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/xknx/io/udp_client.py", line 88, in data_received_callback
    knxipframe.from_knx(raw)
  File "/usr/local/lib/python3.8/site-packages/xknx/knxip/knxip.py", line 81, in from_knx
    raise CouldNotParseKNXIP("KNXIP data has wrong length")
xknx.exceptions.exception.CouldNotParseKNXIP: <CouldNotParseKNXIP description="KNXIP data has wrong length" />
2020-09-20 18:22:54 WARNING (MainThread) [xknx.log] Error: KNX bus responded to request of type 'ConnectionState' with error in 'ConnectionStateResponse': ErrorCode.E_CONNECTION_ID
2020-09-20 18:22:56 WARNING (MainThread) [xknx.log] Error: KNX bus did not respond in time to request of type 'Tunnelling'
2020-09-20 18:22:56 WARNING (MainThread) [xknx.log] Sending of telegram failed. Retrying a second time.
2020-09-20 18:22:56 WARNING (MainThread) [xknx.log] Error: KNX bus responded to request of type 'ConnectionState' with error in 'ConnectionStateResponse': ErrorCode.E_CONNECTION_ID
2020-09-20 18:22:56 WARNING (MainThread) [xknx.log] Error: KNX bus responded to request of type 'ConnectionState' with error in 'ConnectionStateResponse': ErrorCode.E_CONNECTION_ID
2020-09-20 18:22:57 WARNING (MainThread) [xknx.log] Error: KNX bus did not respond in time to request of type 'Tunnelling'
2020-09-20 18:22:57 WARNING (MainThread) [xknx.log] Resending telegram failed. Reconnecting to tunnel.
2020-09-20 18:22:57 WARNING (MainThread) [xknx.log] Error: KNX bus responded to request of type 'Disconnect' with error in 'DisconnectResponse': ErrorCode.E_CONNECTION_ID

```

When activating the ETS monitor I noticed that this telegram causes this error:

[image]

It is the current date/time (DPT 19.0001), sent by my MDT IP Interface.

I disabled the time broadcast every 10 minutes and the error / disconnect was gone.

**Version information:**
- xknx / Home-Assistant release with the issue: 0.115.2

**KNX installation:**

IP Interface: MDT SCN-IP000.03 

**Problem-relevant `xknx.yaml` or `configuration.yaml` entries (fill out even if it seems unimportant):**

knx.yaml:
```
tunneling:
  host: '192.168.100.5'
  port: 3671
fire_event: true
fire_event_filter: ["1/*/*", "8/2/*"]
state_updater: true
binary_sensor: !include knx_binary_sensors.yaml
sensor: !include knx_sensors.yaml
cover: !include knx_covers.yaml
climate: !include knx_climate.yaml
light: !include knx_lights.yaml
switch: !include knx_switches.yaml
expose:
```

farmio, 2020-09-21:
I think its not DPT 19 that is causing problems, but the thing sent to 0/1/0. 
A date sould have 3 byte, ETS shows 16... whats that?

josiasmontag, 2020-09-24:
Indeed it seems that the IP Interface is sending an invalid date on 0/1/0. I will check why exactly this is happening.

Nevertheless, I think xknx should just ignore such invalid data and should not disconnect from the tunnel?

farmio, 2020-09-24:
Not according to the knx specifications...

farmio, 2020-10-27:
I have tried to reproduce this with my MDT SCN-IP100.02 Router but it is sending a normal payload as expected. It uses a different application than the 000.03 so this may not mean much. 

Have you had a talk with MDT about this issue?

PR #485 (2020-10-29): Bugfix: Unsupported CEMI Message breaking tunnel connections

## Description

- Handle unsupported CEMI Message
In TunnelingRequest header length was added to total length resulting in KNXIPFrame() raising CouldNotParseKNXIP. This was probably a typo resulting in loosing tunnel connections when such frames were received

additionally I have added some details to debug logs / exceptions messages
- add raw CEMI Frame to cemi exceptions for better debugability
- add individual address to "start routing" log message

Fixes #408 

## Type of change

- [x] Bug fix (non-breaking change which fixes an [... cut]
closed 2020-10-29 by farmio